luc-github/Repetier-Firmware-4-Davinci #13: Printer is not recognised in Windows after install of 0.92 on Davinci 1.0

blackmambazaa, 2015-01-02:
Issue:

- I have uploaded 0.92 on to my Davinci 1.0 and the printer isn't recognised in Windows anymore.
- Before the upload, the device showed up as a Bossa... but now it just says Unknown Device.
- I also verified that the firmware has successfully installed on the printer by cycling through the menus on the printer itself.

Steps taken to try to resolve the communication issue:

- I had installed the Sam-ba and Due drivers but none seem to resolve the "Unknown Device" issue.

Any ideas?

luc-github, 2015-01-02:
Normally it is Arduino Due - drivers is in Arduino IDE 
what version of IDE did you used to compile and flash ?

drbogger, 2015-01-02:
Yup, that happens if you don't use the variant.cpp file that's included.

blackmambazaa, 2015-01-02:
I had used IDE version 1.5.8 and the drivers within that. 
I also used the variant.cpp included in the firmware package.

I guess it won't hurt to try the install again hey. I will post back my findings.

drbogger, 2015-01-02:
I had issues getting the variants.cpp to work with the arduino install package.. try using the .zip version, and see if it works that way.

luc-github, 2015-01-02:
looking at https://github.com/luc-github/Repetier-Firmware-0.92/issues/14
I thinks something is wrong in installation, check if variant.cpp is same as fw one to be sure copy is well done

blackmambazaa, 2015-01-02:
Can you please post your variants.cpp file here. I want to compare it to mine.

luc-github, 2015-01-02:
my variants.cpp is the same as FW
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/AdditionalArduinoFiles/1.5.8/hardware/arduino/sam/variants/arduino_due_x/variant.cpp

blackmambazaa, 2015-01-02:
Scratch that. I re-flashed and uploaded 0.92 to the printer. All is well. The Due drivers are recognised now and I am able to communicate with the printer.

Thanks for the help.

closed 2015-01-02 by blackmambazaa

trix8687, 2015-03-03:
i am having this problem. i have reflashed 3 times now

luc-github, 2015-03-03:
OS ? did you copy variant.cpp / usbcore.cpp ? arduino ide version ? how do you got 0.92 : git-hub or Zip ?

trix8687, 2015-03-03:
Window 7 64, yes i copied both. arduino ide 1.5.8, zip, not 100% sure it is the da vinci 1.0 came with version 1.0.1

luc-github, 2015-03-03:
you have micro SD or normal SD Card reader ?

trix8687, 2015-03-03:
micro

luc-github, 2015-03-03:
so it is 1.0A

trix8687, 2015-03-03:
i assumed as much. i did make that change in configuration.h

luc-github, 2015-03-03:
check you copied properly the variant.cpp in C:\Program Files (x86)\Arduino\hardware\arduino\sam\variants\arduino_due_x

download zip again - it may be corrupted

trix8687, 2015-03-03:
just to confirm, i am copying varriant from the zip file to Arduino\hardware\arduino\sam\variants\arduino_due_x,
yes?

luc-github, 2015-03-03:
yes in zip there is one for 1.5.8 and one for 1.6.0 copy the one for 1.5.8 to C:\Program Files (x86)\Arduino\hardware\arduino\sam\variants\arduino_due_x

trix8687, 2015-03-03:
still no dice. windows recognizes it as USB composite device and cannot install drivers

trix8687, 2015-03-03:
My arduino directory is D:\Program Files (x86)\Arduino. That shouldnt be an issue, should it?

luc-github, 2015-03-03:
no as I wrote earlier :  C:\Program Files (x86)\Arduino\hardware\arduino\sam\variants\arduino_due_x

I have win7 64bits also

reinstall arduino 1.5.8 then copy again the varian.cpp / usbcore.cpp as they are erased when reinstalling

there is no reason the arduino drivers is not installed especially if Arduino IDE see the correct port COM

trix8687, 2015-03-03:
still nothing

trix8687, 2015-03-03:
this is crazy. it recognizes it before i flash it, but after, the firmware is working on the printer, but windows wont connect.

luc-github, 2015-03-03:
ho wait - fw is ok on printer - no black bar ?

trix8687, 2015-03-03:
no, temps showing , printer is ready.... all the menus are correct

luc-github, 2015-03-03:
ok this is different then
it is windows mess - not FW
the drivers are in C:\Program Files (x86)\Arduino\drivers
select the unknow drivers by these ones

trix8687, 2015-03-03:
i dont see the drivers

luc-github, 2015-03-03:
you do not have 
C:\Program Files (x86)\Arduino\drivers\arduino.inf

trix8687, 2015-03-03:
i have that, what do i need to do with it?

luc-github, 2015-03-03:
use it to update the unknown or wrong selected drivers
what your device manager says ?

trix8687, 2015-03-03:
ok, i think i got it, sorry im no good with drivers. i'll let you know how that goes

trix8687, 2015-03-03:
"the best driver for your device is already installed"
 "USB Composite Device"

trix8687, 2015-03-03:
still not working

luc-github, 2015-03-03:
update it manually by selecting "browse my computer for drivers software" then select the C:\Program Files (x86)\Arduino\drivers

trix8687, 2015-03-03:
i did. thats when it says "the best driver for your device is already installed"
"USB Composite Device"

luc-github, 2015-03-03:
is this COM disapear when you unplug the printer ?

trix8687, 2015-03-03:
the COM disappears when i flash the firmware

luc-github, 2015-03-03:
when you plug printer a device appear in device manager
when you unplug printer a device disappear in device manager

trix8687, 2015-03-03:
the "USB Composite device" disappears when i unplug it, yes

luc-github, 2015-03-03:
OK then try to uninstall it - then unplug and replug the printer
when the printer will be detected It may ask for drivers so you can select the arduino one

trix8687, 2015-03-03:
ok, it appears to may have been a problem with that usb port, i swapped it with one my regular printer was plugged in and it is showing up as arduino Due (COM4)

luc-github, 2015-03-03:
I read somewhere using USB 3.0 port is not worling well but no issue with 2.0

luc-github, 2015-03-03:
at least you can use any host now printer port is working - should be ok

trix8687, 2015-03-03:
thanks for all the help anyway

garry572, 2015-03-03:
yes i had trouble with a usb3 port, but work fine on usb2

luc-github, 2015-03-03:
Thanks Garry I forgotten where I read that - was from you ^_^

Weisbrot237, 2015-12-26:
I Got the same Problem. but i dont even have USB 3.0 on my pc. only USB 2.0
i tried every usb port i have. on 2 it shows "unknown device" and on the rest it shows "USB Composite Device"
someone got a idea?

alexkaps, 2016-01-10:
Same for me, but with an accidental board short circuit. Now black lines in lcd , nothing in Ports and this in device manager "Windows has stopped this device because it has reported problems. (Code 43)".  Is there any doctor or there is no hope?

luc-github, 2016-01-11:
@alexkaps  Please have a look at http://voltivo.com/forum/davinci-peersupport/734-1-0a-no-power-to-extruder-heater-fix and other related threads

alexkaps, 2016-01-11:
thank you luc, just before the darkness i had the same extruder decoupled message. and while trying to locate the R271 fuse\resistor i had the "board jump". I 'm not 100% sure of jumping the pins because i was not looking, but everything lost.

luc-github, 2016-01-11:
NP let me know if you still have decoupled issue after the fix

luc-github, 2016-01-11:
Ho just to confirm did you jump JP1 or JP37 according your printer type ? - you may just need a reset / reflash

alexkaps, 2016-01-11:
Yes i jumped J37 and i have to reflash if everything goes well. I just bridge/solder the R271 and add a new fuse at the extruder's cable, right?  Could this fuse be a 5A car fuse?
lol it's the only one i got right now.

alexkaps, 2016-01-11:
I did the above and now have continuity between my 12V supply and the left white cable below R271. 
I still get the usb device not recognized error and nothing on ports.  Should the left white cable have continuity with the ground pin? Because it hasn't if it matters.

luc-github, 2016-01-11:
I am not an expert for board fix, sorry - you may post your issue on soliforum and voltivo forum, there are a lot of expert of Davinci Electronic there -  they will confirm the next steps better than me

alexkaps, 2016-01-11:
I ve already post it but you are the only one answered so far. :)  I was considering  changing my board to ramps 1.4 but information i ve found is still elusive to me, so doing more research. If i fix my problem i ll post it here just in case someone have the same problem. Keep up the good work! Thank you.

luc-github, 2016-01-11:
several people change to mega/ramps but you also need to change endstop and screen/keypad in this case

luc-github, 2016-01-11:
FYI - If you are in US and it is MCU issue - this can be fixed also - some guys proposed to do it

alexkaps, 2016-01-11:
Yes, but changing to ramps is my only hope so far. I think you are referring to Keith from Voltivo, but unfortunately for my printer i'm in Greece.

luc-github, 2016-01-11:
the best for HW is here http://voltivo.com/forum/davinci-hwmods/509-my-arduino-ramps-install?start=20#4558
But depending of your ramps knowledge is may not be enough

[8 more replies]