luc-github/Repetier-Firmware-4-Davinci #14: Davinci 1.0 periodically restarts after installation of 0.92

blackmambazaa, 2015-01-02:
After flashing the EEPROM and uploading 0.92 onto my Davinci 1.0, I am able to see that the firmware has successfully installed on the printer, but the issue that I am experiencing is that the printer restarts every 25 seconds.

So whether the printer is idling or whether I am in a menu, it still restarts.

I have one of the newer boards that had 1.1j as the default firmware on it.

Any ideas?

luc-github, 2015-01-02:
This is the watchdog - looks you have watchdof enabled but somehow it is not reset

blackmambazaa, 2015-01-02:
Is the watchdog like a "Stay awake" type of feature?

luc-github, 2015-01-02:
I suggest to first disable watchdog, it is in variant.cpp, to see what is wrong

luc-github, 2015-01-02:
watchdog is a checker printer is not in dead loop by regular ping - if no ping is done within 15 s it reset printer

blackmambazaa, 2015-01-02:
Okay, good news is that I flash the EEPROM and uploaded 0.92 again. It no longer restarts due to the watchdog.

Turns out that I had uploaded "repetier-firmware-davinci" instead of 0.92-master. Apologies. 

Thanks for the help.

closed 2015-01-02 by blackmambazaa

luc-github, 2015-01-02:
ho you used the 0.91 - so yes make sense for the issue

blackmambazaa, 2015-01-02:
Yip I had 0.91 in the same parent folder as 0.92. Recipe for disaster

luc-github, 2015-01-02:
Well issue is solved - this is the key point - enjoy