luc-github/Repetier-Firmware-4-Davinci #15: Pause SD print moves carriage out of reach

Nepenthy, 2015-01-03:
When pausing a SD print, the carriage moves to the front right which is very difficult to get at. For pause print could the carriage be moved to either x/y home or the clean nozzle position? This would allow for both nozzle cleaning and filament change mid print. Thanks.

luc-github, 2015-01-04:
I think if it is also for nozzle cleaning when pause going to home will work only for 1.0
for 2.0 better to park in left drip box as right drip box will not allow to clean right extruder but only left one

luc-github, 2015-01-05:
What do you think to put position also in EEPROM like for Manual Leveling and Auto leveling ?

Nepenthy, 2015-01-06:
Luc, to be honest, Id be thrilled to just get the pause SD print position away from where it is for the 1.0. I fear Im going to develop a deformity with the way I have to twist my hands in there to change or correct filament. ;) By having in firmware, do you mean having a choice of positions a user could pick for pausing prints? If so, I think that would be great! If you meant a choice of locations in the menu once the print is paused, thats even better!

luc-github, 2015-01-07:
I was thinking more just about the X and Y position to be modifable 
Actually after pause user can move X, Y and Z using manual control in repetier / or position in Printer
you know that ? you already can avoid " to develop a deformity with the way you have to twist our hands" may I know why you do not use it ? did you find any issue ?

Nepenthy, 2015-01-07:
Man.... Are you serious? I hated pausing in repetier because it would just rest on the print and I couldnt do much to resolve issues. I had no idea you could manually move it a better position and it would pick up from a pause. When I ran across this feature while printing off SD I was so excited! Now that I think about it, when first starting I paused and disabled stepper which was a catastrophe. From then on I thought pause and move was not feasible. Ive since learned to quit messing with the stepper altogether. ;)

Well, this request has provided great assistance, opportunity,  and embarrassment at the same time....as auto move to different positions is just lazy if I can do it via position menu and resume. Once again Luc, I very much appreciate it.

closed 2015-01-07 by Nepenthy

luc-github, 2015-01-07:
haha you are welcome 
One warning anyway - if you use it, be sure your settings are ok and do not try to go overlimit like pushing  right when extruder is already on max as seems position are modified when physical one is not so result ,for example, is printer think you are -10 when your extruder is actually 0 so when going to original position the position is shifted of 10 ....

Actually every function is available during pause and that is why I set https://github.com/luc-github/Repetier-Firmware/issues/46

you can see my proposal here:https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/Menu%20Plans.xlsx

[image]
but as you see, based on your feedback, disable stepper during pause is not a good idea actually, I did not think about original position when continue ....