luc-github/Repetier-Firmware-4-Davinci #16: Cannot control the printer via Repetier on 0.92

blackmambazaa, 2015-01-04:
Hi,

I cannot  control the printer via Repetier-host, but I am able to do basic movements on the extruder using Pronterface.

In Repetier-host, I am able to configure everything and select the correct com port, but that's about it.

If I click on Manual Control and thereafter click on the connect button, the status changes from disconnected to idle and it says "8 commands waiting".

Any ideas what this could be related to?

luc-github, 2015-01-04:
yes repetier host is not properly setup

check http://voltivo.com/forum/davinci-peersupport/501-da-vinci-setup-guide-from-installation-to-wireless-printing#3773
and http://voltivo.com/forum/davinci-firmware/148-your-configuration

blackmambazaa, 2015-01-04:
Ah, I didn't save the settings to the EEPROM. I did that now within the settings menu on the printer.

Thanks for the tip.

closed 2015-01-04 by blackmambazaa