luc-github/Repetier-Firmware-4-Davinci #18: Extruder table is not correct for 19 degres for 1.0a

luc-github, 2015-01-06:
Extruder table is not correct for 19 degres for 1.0a it generate a sensor error

luc-github, 2015-01-06:
I cannot test so need feedback for tweaking
I have modified the thermistor table of 1.0a in testing branch 
[image]

[image]

superdavex, 2015-01-06:
I am testing this now. My temps are staying above 0 before the patch so I'll have to wait to see if they come back down.

luc-github, 2015-01-06:
the goal is to have bed temperature and extruder temperature identical or +/-1 degres at room temperature

superdavex, 2015-01-06:
They actually are now showing the same value.    After the patch they are both 21c.  I'll keep a watch on them.

luc-github, 2015-01-06:
Great - yes now need to check in early morning when temperature is cold and printer is cold also
wait for your feedback

superdavex, 2015-01-07:
I can confirm now the patch corrects the temperature for values in the 16 to 19 range.  It is now within one degree of the bed temp.

luc-github, 2015-01-07:
Ok great - l will implement it in master then

Thanks

Doudou14, 2015-01-07:
Is the firmware running on 1.0A? (Except the temperature error)

superdavex, 2015-01-07:
Yes, I am running on 1.0A.  The latest table update fixes the low temp error.

luc-github, 2015-01-07:
blackmambazaa has also a 1.0a and run fw without issue

luc-github, 2015-01-08:
fix implemented in master https://github.com/luc-github/Repetier-Firmware-0.92/commit/4cea0ea717c4fd1011e910314e674c43371a8c3e

closed 2015-01-08 by luc-github