luc-github/Repetier-Firmware-4-Davinci #19: Bed temperaure not controlled

garry572, 2015-01-06:
Just noticed a strange issue with my 1.0 using 0.92. Was doing a print with the bed set at 90C, but over an hour it slowly climbed to 105 and the fail safe disconnected the sensors
Just did a test with the maintenance menu to preheat to table and bed. Ive tried both BangBang and Dead Time. Looking at the Repetier temperature graphs, you can see the extruder temperature is controlled, but the bed temps just slowly keeps rising. The graphs also show, if you look at the option to show power to the extruder and bed, that, no matter what the temperature is, the bed is still being supplied power, even when the temperature is well above the set point ie set temp was 90C , but at 95C the bed was still being powered. Haven't check with previous versions to see if I have the same issue. Maybe Ive accidentally altered something without knowing

luc-github, 2015-01-06:
I did several print at 230/90 and did not have any issue, I printed 7h several times, and also some 1h+ print
I also tried to decoupled by stopping heat when high temperature reached  wait a few seconds then heat again , but parameters are fine and did not decoupled
you may check , adjust your pid settings to avoid overheat and oscillation at target

garry572, 2015-01-06:
I think in the process of me editing files I made a type of some sort of mistake . I just tried your 0.92 version and the issue wasn't here. I will compare my  versions to see what I did :(

garry572, 2015-01-06:
The only changes from my version for yours was the LCD initialization and my own thermistor tables, so it not a big issue

luc-github, 2015-01-06:
can you share your table ?
my understanding is if data are not following a smooth curve you can have some glitch in temperature interpretation which can bring unexpected oscillation due to inaccurate usage of pid
that is why I have doing trend curve instead of keeping measured data, to minimize the effect

garry572, 2015-01-06:
Sure it a blend of the data I recorded and data provided by others to fill in the gaps at the extreme ends of temperature I only made measurements within the operating ranges of the bed and extruder

garry572, 2015-01-08:
will send a copy of my tables tonight, stuck at work today

closed 2015-01-08 by garry572