luc-github/Repetier-Firmware-4-Davinci #23: LCD is scramble after emergency stop since latest patch

luc-github, 2015-01-13:
LCD is scramble after emergency stop since https://github.com/luc-github/Repetier-Firmware-0.92/commit/195076d1621c9e6d4b95288737e7284fdc462908
No issue when starting printer

luc-github, 2015-01-13:
If doing one emergency stop lcd is scramble
if doing a second after lcd is back to normal
if doing a third lcd is scramble
etc...

luc-github, 2015-01-13:
First analyzis:
commenting the pins init and adding rs pin to low as before seems solving the problem:

   //init pins to a known state
```
   /*WRITE(UI_DISPLAY_D0_PIN, HIGH);    

    WRITE(UI_DISPLAY_D1_PIN, HIGH);    

    WRITE(UI_DISPLAY_D2_PIN, HIGH);    

    WRITE(UI_DISPLAY_D3_PIN, HIGH);   

    WRITE(UI_DISPLAY_D4_PIN, HIGH);    

    WRITE(UI_DISPLAY_D5_PIN, HIGH);    

    WRITE(UI_DISPLAY_D6_PIN, HIGH);    

    WRITE(UI_DISPLAY_D7_PIN, HIGH);*/
```
    
#if UI_DISPLAY_RW_PIN>-1
    SET_OUTPUT(UI_DISPLAY_RW_PIN);
#endif
    // Now we pull both RS and R/W low to begin commands
    WRITE(UI_DISPLAY_RS_PIN, LOW);
    HAL::delayMicroseconds(5);
    //move set output late to be sure pin is not affected before
    SET_OUTPUT(UI_DISPLAY_ENABLE_PIN);
    HAL::delayMicroseconds(5);
    WRITE(UI_DISPLAY_ENABLE_PIN, LOW);
    **WRITE(UI_DISPLAY_RS_PIN, LOW);**

luc-github, 2015-01-13:
Adding a delay to previous change
    WRITE(UI_DISPLAY_ENABLE_PIN, LOW);
    **HAL::delayMicroseconds(5);**
    WRITE(UI_DISPLAY_RS_PIN, LOW);
and issue is back

luc-github, 2015-01-13:
this sequence seems avoiding the problem, move rs pin output creation after data pins initialization, and create / write in same time as enable pin

```
 // SEE PAGE 45/46 FOR INITIALIZATION SPECIFICATION!
    // according to HD44780 datasheet, we need at least 40ms after power rises above 2.7V
    // before sending commands. Arduino can turn on way before 4.5V.
    // is this delay long enough for all cases??
    HAL::delayMilliseconds(500);
   
    //init pins to a known state
    WRITE(UI_DISPLAY_D0_PIN, HIGH);
    WRITE(UI_DISPLAY_D1_PIN, HIGH);
    WRITE(UI_DISPLAY_D2_PIN, HIGH);
    WRITE(UI_DISPLAY_D3_PIN, HIGH);
    WRITE(UI_DISPLAY_D4_PIN, HIGH);
    WRITE(UI_DISPLAY_D5_PIN, HIGH);
    WRITE(UI_DISPLAY_D6_PIN, HIGH);
    WRITE(UI_DISPLAY_D7_PIN, HIGH);

    //move set output late to be sure pin is not affected before
    SET_OUTPUT(UI_DISPLAY_RS_PIN);
    SET_OUTPUT(UI_DISPLAY_ENABLE_PIN);
    HAL::delayMicroseconds(5);
    WRITE(UI_DISPLAY_ENABLE_PIN, LOW);
    // Now we pull both RS and R/W low to begin commands
    WRITE(UI_DISPLAY_RS_PIN, LOW);
    HAL::delayMilliseconds(10); // Just to be safe
```

luc-github, 2015-01-13:
pushed to testing  https://github.com/luc-github/Repetier-Firmware-0.92/blob/727f511585ac4500ff284aab5105109897d82918/src/ArduinoDUE/Repetier/ui.cpp

Hope no impact on normal start - need to be tested

garry572, 2015-01-13:
I have the reverse problem normal start is OK, but emergency stop LCD is scrambled. Seems that's there are inconsistencies between various hardware making tracking down the cause more difficult

luc-github, 2015-01-13:
Start is fine too on mine  - just  emergency stop LCD was scrambled

garry572, 2015-01-13:
Even with the move of the write to rs I still have a random failure on emergency stop. Needs more investigation. Think the LCD is very fussy with certain data present on its pins during start up/emergency stop

luc-github, 2015-01-13:
may be need to prepare LCD in HAL::resetHardware() before reset

garry572, 2015-01-13:
Might be a good idea, :)

garry572, 2015-01-13:
Works getting in the way of playing, have to look at it later

luc-github, 2015-01-13:
I put back master fw to reproduce issue today and rate is lower now, need several emergency stop to get issue...

luc-github, 2015-01-13:
in void HAL::resetHardware() 
Add ` WRITE(UI_DISPLAY_ENABLE_PIN, LOW);` before RSTC seems help

garry572, 2015-01-13:
ok will try tonight :)

luc-github, 2015-01-14:
OK found the root cause - not solution
the keypad: if you left buttons pressed when doing emergency stop issue happen, if it is light push no issue
having the key pressed when restarting seems generated the issue in some way

luc-github, 2015-01-14:
I desactivated shortcut of emergency stop and used M281 to restart printer
same behaviour when keeping pressing the buttons and sending the M281
so issue is not emergency command but buttons pressed during restart
but no issue when pressing buttons when doing cold boot

garry572, 2015-01-14:
yes i found the same too

luc-github, 2015-01-14:
Duplicate initialization solve issue
#if UI_DISPLAY_TYPE != NO_DISPLAY
    initializeLCD();
   HAL::delayMilliseconds(10);
    initializeLCD();

Need to dig to see which command may need to duplicate to avoid a full duplicate initialization

luc-github, 2015-01-14:
seems adding a third
    lcdCommand(BIT_INTERFACE | LCD_2LINE | LCD_5X8);//LCD Configuration: Bits, Lines and Font
    HAL::delayMicroseconds(150); //more than 39micro seconds

and looks ok

luc-github, 2015-01-14:
I have tested with emergency stop shortcut and with M281 command and pressing some keys while restarting

Can you confirm it works for you too ?

garry572, 2015-01-14:
Have tried the emergency stop, no failures. No LCD initialization errors with two LCD init

luc-github, 2015-01-14:
Did you try by doing single Initialization but adding a third 

lcdCommand(BIT_INTERFACE | LCD_2LINE | LCD_5X8);//LCD Configuration: Bits, Lines and Font
HAL::delayMicroseconds(150); //more than 39micro seconds

luc-github, 2015-01-14:
I have revert testing branch in same state as master and applied my suggestion  of a third init line
https://github.com/luc-github/Repetier-Firmware-0.92/commit/5817c8120e1ce7510a7ada5320d4e2577fd2eaf3

garry572, 2015-01-15:
MInes been good with two initializations. Funny but mine only ever fails at the magical temperature of 25C (Thats whats shown as being the bad/extruder temp)

luc-github, 2015-01-15:
Yes 2 init is working but - I think just one init and 3 
lcdCommand(BIT_INTERFACE | LCD_2LINE | LCD_5X8);//LCD Configuration: Bits, Lines and Font
HAL::delayMicroseconds(150); //more than 39micro seconds

solve the issue

I do not have 25 degres - too cold here so cannot test

garry572, 2015-01-15:
ok will try tonight,

luc-github, 2015-01-15:
Great - thanks a lot - I hope we are done with LCD after this ^_^

squoyster, 2015-01-15:
Works here on a 1.0 (1A3) with test branch.  It was failing with the master.  Also, fixes extruder temps.  Thanks!

luc-github, 2015-01-15:
Thanks a lot for your feedback

luc-github, 2015-01-16:
Garry -when Ok on your side I will upload fix to master and move 0.92 to beta stage and do and official release, let me know if any issue

garry572, 2015-01-16:
Hi Luc, all good on my side, no issues even at my critical temp of 25C

luc-github, 2015-01-16:
Excellent - thanks a lot - I will push the fix to master and close issue
I wish no more issue with LCD  X_X

luc-github, 2015-01-16:
Implemented https://github.com/luc-github/Repetier-Firmware-0.92/commit/8f7fc926857e569286f6074a7cf3aee5a5dec1c3

closed 2015-01-16 by luc-github