luc-github/Repetier-Firmware-4-Davinci #25: Pause printing commands from SD are not immediate but let buffer to empty

luc-github, 2015-01-14:
Pulled from 0.91 https://github.com/luc-github/Repetier-Firmware/issues/55

DrAnonymous, 2015-01-15:
I will study the firmware and host protocol to see if something makes sense. The stop printing command should be possible, because you don't have to worry about any bookkeeping like you would for a pause print command.

luc-github, 2015-01-15:
Immediate Stop command is done already and implemented in 0.92

tecan, 2015-04-27:
this one might be over my head. all i can think is make another executeFString function with a         if(M == 420){ watcher in gcode.cpp that swaps the buffers out.

luc-github, 2015-04-28:
you mean add a M420 command to use  different buffer ?

tecan, 2015-04-28:
just a paused buffer.. or it could be added to m25 too possibly.

luc-github, 2015-04-28:
Unlike pause printing from host, pause printing from SD should be not too hard because printer has all GCODE file on SD so can go back to any line any time

My Idea was for each last GCODE line executed, save the line position in file, save also temperature and x y z positions when command is done. (not difficult)
So when pause is pressed, purge all buffer like it is done with current stop command, it just let the current command ending. (reuse the stop command)
So when resuming, just go back to last position X Y Z and reset temperatures,  (not difficult)
then just restarting the print by reading the line just after the last executed one ( a seek in file then start print, not hard neither)

 no need big extra buffer : just 3 positions, one line number and 2 or 3 temperatures values

what do you think ?

tecan, 2015-04-28:
i can see why no one wants to fix it :)  the final solution will be one where the sd card is not required, and is still compatable with different host softwares.

luc-github, 2015-04-28:
Well, my other printer (a makerbot clone) has immediate pause/stop function so I do not see why repetier do not support it
I am starting the porting of marlin FW, it will be a good opportunity for feature comparison on same printer

tecan, 2015-04-28:
ok i did some more digging around commandsBuffered needs to be swapped out with a temp buffer static GCode tempBuffered[GCODE_BUFFER_SIZE]; 
and a block needs to be set on readFromSerial(). will keep playing with it till it works unless someone beats me to it heh.

luc-github, 2015-04-29:
no worry we wait for you - I do not think someone will do faster

tecan, 2015-04-29:
came up with a mockup function like this.

void GCode::SwapBuffer()
{
    if  (!bufferLength){
      for (int i=0; i < GCODE_BUFFER_SIZE;i++) 
      { // swap temp buffer back and resume serial communication
        commandsBuffered[i] = tempBuffered[i];
        tempBuffered[i]=null;
      }
              GCode::bufferLength=tempBufferLength;
              GCode::tempBufferLength=0;
    }else{
      for (int i=0; i < GCODE_BUFFER_SIZE;i++) 
      {  //swap command buffer into temp buffer and block serial communication
        tempBuffered[i] = commandsBuffered[i];
        commandsBuffered[i] = null;
      };
      GCode::tempBufferLength= bufferLength;
       GCode::bufferLength=0;
    }
}

luc-github, 2015-04-30:
Hi yes looks like a good start but actually GCODE itself is not the main place of operation as you can see buffer is already low
```#define GCODE_BUFFER_SIZE 2```
the "real" buffer is in lines generated by the GCODE and used by motion.cpp
you can check ```cur = &lines[linesPos];``` and ```queueCartesianMove```

you can also check ```MENU_MODE_STOP_REQUESTED``` to see the effect of the flag I use to purge and even, it seems I miss some cleaning in the purge process so that why I need a second flag to help to purge the place I missed

tecan, 2015-05-03:
im still about a week away from something something working. been glancing at it between farming and moving a house.

luc-github, 2015-05-03:
No worry - Take your time - I know what it is to be busy IRL :smile:

tecan, 2015-05-09:
everything i try results in compile errors doing it in motion.cpp . i wont be fixing this because i don't really know c++ oop properly to begin with (self taught).

tecan, 2015-05-09:
lost some progress, got it back. i should really learn proper oop someday, would make it less confusing heh. bug sevarity should be increased for this, its the last obstacle to first 3d printer autounjammer.

luc-github, 2015-05-09:
``` static void queueCartesianMove(uint8_t check_endstops,uint8_t pathOptimize);```
is a static function, it means there is no need of object like ```PrintLine myprintline;``` to be executed,
it can be called dirrectly ```PrintLine::queueCartesianMove(....);```

I guess you did not defined your ```void PrintLine::SwapBuffer();``` as static so can be used

tecan, 2015-05-09:
awesome! set to static , deleted the () and it compiled! now for the fun stuff.

tecan, 2015-05-22:
i cant seem to focus on it. offering a 200k doge coin reward

luc-github, 2015-05-22:
Ha ! you are rich ^_^

tecan, 2015-05-22:
good news i may have gotten it working. doing a print currently.

tecan, 2015-05-22:
yeah 40 dollars aint much but its a nice incentive. makes me feel better for not being able to fix it hehe. i tried setting the flags and it swaps the buffer without actually waiting for the commands to finish . needs some of that purge magic you were talking about.

luc-github, 2015-05-22:
I know the feeling , I was (and still now) very frustrated to not to find a clean purge of the commands, that  is why I hold the pause to see if I can get better understanding latter , but not lot of time for this unfortunnatly

repetier, 2015-06-13:
There is no easy way for direct pause. The path planner has planned for it's stored moves and simply stopping execution may loose steps and you can not do any move commands cause the move buffer is already full. Tricks with sd card will not work with hosts, so that is no solution.  You could think about modifying planned path to stop moves and store last position and already buffered data somewhere. This will not work with delta printers as they are already out of memory. At it will take a few month of testing to get it going stable, which is why it was never implemented. It is so easy to break path planning and it took so long to get it running stable that I will not touch it for something that is not really necessary. If you need a pause to change filament it will normally no problem to finish stored moves. If your printer goes creasy reset it - print is ruined anyway.

luc-github, 2015-06-13:
Yes I know - it is definitely not easy and even more difficult with host printing - that is why I just keep tracking for SD, even I cannot fix myself today for DAVINCI printer (a cartesian printer based on DUE) .

Currently lack of immediate pause is reported also due to filament detection - because on Davinci the sensor is just on top of direct drive, there not so much pad to pause the print when no filament.

Also Davinci stock FW which is based on repetier also, they currently are able to do the pause immediately, so I am sure there is a way at least for a Cartesian/DUE printer.

I know repetier FW need to deal with several boards/configurations (with an increased number everyday) so  that is why I did not pushed so much, and I also did not see a lot of request for this, so I understand it is not a priority and today it may not worth the workload.

That said, I think keeping issue open give more chance to have someone having the skill jumping into it, and may definitely help to find a solution for SD first, and may be for Host after.

Hecot, 2016-03-17:
I went through the old issue list and found this one interessting! What I really missing is an **emergency stop** when printing from SD-card! It will always take too long to go through all the submenus to the point where you can cancel the print, especially if you just entered the room and see what mess the printer did and is still doing you want a quick way :) Just like pressing the up and down Buttons for 5 seconds would bring you to the cancel print option ... would be a great help!

luc-github, 2016-03-17:
well it is implemented it is on read me
Emergency stop (Left key/Down key/Ok key in same time)

Hecot, 2016-03-18:
... RTFM :))

Thanks!!!

tecan, 2017-03-12:
they've been after my brain, i can no longer code on this for some odd reason.

luc-github, 2018-12-09:
As pause is not immediate it is only for filament change and it should be OK
For emergency -> Emergency stop

So I close issue as no solution during 3 years

closed 2018-12-09 by luc-github