luc-github/Repetier-Firmware-4-Davinci #26: Be able pause host print from printer

luc-github, 2015-01-14:
issue pulled from 0.91 https://github.com/luc-github/Repetier-Firmware/issues/52

luc-github, 2015-03-11:
Need to add menu item when printing that 
1 - detect have some serial communication ongoing (other than temperature request ?) and is only visible when host is sending command
2- send "RequestPause:" to host
3 - Dispear when Pause command is sent, no Continue command exists so far yet

question is where to put this menu item ? 
I plan to change name Stop/Pause/Continue of SD Card to a Printing Menu that is displayed only when printing (SD or Host) so will be transparent - if this Menu is visible SD Card should be Hidden as Print File/Delete File should  not be visible when printing

hugobmf, 2017-06-20:
And amazing how things work when you read the manual, I went through the installation step by step and found it to be fail safe and everything seems to be working fine. kkk
My XZY is free!!!

Thank you Luc!!!

luc-github, 2018-12-09:
Issue is old and I never used host in 3 years always print from SD so may not spend time in it now

closed 2018-12-09 by luc-github