luc-github/Repetier-Firmware-4-Davinci #27: Add Jam sensor support

luc-github, 2015-01-14:
Issue pulled from 0.91 https://github.com/luc-github/Repetier-Firmware/issues/38

Doudou14, 2015-01-14:
Complicated as necessary to manage multitasking

luc-github, 2015-02-15:
Seems repetier started to work on this - will keep an eyes on this

axs-gentoo, 2015-03-03:
If this gets added, please ensure there is a menu option to disable it -- those of us playing with PLA know first-hand how difficult it is to deal with jams, and having our prints stop/react to jams all the time is going to make it very difficult to tune settings to get the filament dialed in...

luc-github, 2015-03-03:
Yes I set a setting to disable any sensor when have one - this one will have too

luc-github, 2015-03-20:
Jam sensor support from repetier 0.92.3 is now present but not activated in configuration.h, need to assign the pin(s) defined in userpins.h to activate it
I will add a support to enable/disable it from menu/EEPROM later
I won't test it as it is minor feature and it involves a lot of testing time - so anyone want to work on it, he/she more than welcome

axs-gentoo, 2015-03-20:
Is there a known mapping already for the pins on the old DaVinci 1.0?  If so I may be able to test;  I've got my jamming issues more or less sorted but I'm sure I can mess things up enough to trigger them again.

luc-github, 2015-03-20:
My Bad I though I have set in userpins.h actually I only set in pinout.xls
[image]
[image]
Here the one I found on my Duo so for 1.0 take the right one
Thanks for your help

Doudou14, 2015-03-20:
Good!
It's been that I was not here. I have time next week for testing, I'll let you know

tecan, 2015-04-01:
my unjam command for pronterface looks like "g91:g1 e-10 f400: g1 e10 f400:g90" 20 works better

luc-github, 2015-04-01:
well I did not tested it - but current Jam Detector stop the print and go to a menu "filament change" to move up/down filament which was not worked well when I tested change filament function, I have modified the change filament function to use the already Load/Unload function we have but did not touched the Jam Wizard so it could have some side effect after detection.

That said first test is to see if it is detecting Jam properly.
Once it is confirmed we may allow user to choose between automatic Unjam (like using your commands or customizable in configuration.h) and manual unjam => go to load/unload filament
What do you think ?
Let me know  if Jam detection is working - I cannot test right now

tecan, 2015-04-11:
http://www.thingiverse.com/download:1196524  double-herringbone-interlock.stl   << this is the model file i use to jam my printer the most.  i think the model was exported with funny artifacts left inside it. none the less almost guaranteed jams. i'll take a look over this next week and post back with a finding unless someone else beats me to it.

luc-github, 2015-04-11:
Ok thanks for the help, waiting for your feedback

tecan, 2015-04-18:
it is detecting jams on pin 70 though it might be a bit sensitive. will try and dig into it tonight, cant wait to fix it :)

luc-github, 2015-04-18:
what block was commented for JAM in extruder.cpp ? be careful I put DAVINCI specific as comment when I do change compare to repetier - so if you are talking about block line 284 - it is not done by me but repetier so should not be used - they do not clean the code time to time 

the part I commented is in ui.cpp for filament change as it is in conflict with what we use today and was not working based on my test as it use encoder
look for FILAMENTCHANGE in ui.cpp

tecan, 2015-04-24:
i feel retarded, had lots of free time keep switching between projects. i cant seem to figure out how to extrude and retract in code was going to put the unjam stuff inside #if JAM_ACTION in void Printer::handleInterruptEvent() { in printer.cpp. sorry for not mentioning it sooner.

tecan, 2015-04-24:
theres roughly 40 mm of filament per revolution of the optical wheel.  and 6 black and 6 silver stripes on a 14mm wheel

tecan, 2015-04-26:
if(com->hasE())
        {
            Printer::currentPositionSteps[E_AXIS] = Printer::convertToMM(com->E) * Printer::axisStepsPerMM[E_AXIS];
        }    looks like this might be part of what we need.

luc-github, 2015-04-26:
can you explain more the issue you meet ? I am not clear reading at your previous comments , what this will solve , sorry

tecan, 2015-04-26:
we just need an automatic unjammer option that does this "g91:g1 e-10 f400: g1 e10 f400:g90" i can tweak the settings to pickup the jams properly afterwards.

luc-github, 2015-04-26:
ok

luc-github, 2015-04-26:
I suggest to add a define in configure.h with a string of unjam command, and use it with executegcode command, and add choice in menu when jam happen, auto unjam or manual unjam

another way would be to try automatic unjam they check if still unjam and propose manual

tecan, 2015-04-26:
cool, seems to be working great. i'll iron out the wrinkles and make a patch.

tecan, 2015-04-27:
#define JAM_STEPS 594   // either debug the output using mm 99 steps pe  
#define JAM_SLOWDOWN_STEPS 750
#define JAM_SLOWDOWN_TO 95  //set to -1 to disable ? reduce/increase flow instead of speed.
#define JAM_ERROR_STEPS 900
#define JAM_MIN_STEPS 20

#elif JAM_ACTION == 3 // pause host/print
Com::printFLN(PSTR("UNJAMMING"));
#define unjam "M601;G91;G1 E-10 F400;G1 E10 F400;G90;M601"
        GCode::executeFString(unjam);
        Extruder::markAllUnjammed();

this is working as it should but its not getting into the buffer fast enough to work properly especially for bottom layers that use one command for a large infill. im not really sure what to do next yet.

luc-github, 2015-04-27:
you mean the pause is not fast enough to pause immediately ? so it also delay the use of the unjam command ?

tecan, 2015-04-27:
yeah it works faster pausing from the host side than pausing from the execute M601. seems like anyway, kinda stumped.

luc-github, 2015-04-27:
Yes this is an old limitation (0.91) - I had discussion with repetier developer asking why pause was not immediate and proposed some way to do a stop, purge or move the buffer, then resume but the answer was : too much work ...
IMHO it should be possible but I also did not spend time on it yet : https://github.com/luc-github/Repetier-Firmware-0.92/issues/25

tecan, 2015-10-11:
upping reward to 600k dogecoin

luc-github, 2015-10-12:
I plan to sync 0.92 from repetier with davinci one when the native usb port issue is implemented, could be an occasion to check repetier implementation for jam sensor

thomastech, 2015-11-05:
I would like to cheer you on in your efforts to add jam detection to the daVinci Repetier firmware. It's a nice feature to have. 

It's great to know that the hardware is already in the daVinci to do it.  That saves us from adding a filament travel sensor.

FWIW, I recently created a filament monitor for my Marlin based printer. It uses a rotary encoder to sense filament travel. When feed problems are detected some custom code executes a filament change command. So prints that would ordinarily be damaged/lost are saved without any fuss. 

- Thomas

luc-github, 2015-11-05:
actually jam sensor is already supported in repetier but the flow of jam sensor bring also the flow of repetier load /unload which was not working well with davinci without encoder, and because davinci has already a load/unload feature was easier to not use repetier one, but for next sync I will do my best to make everything to work together

Hecot, 2016-03-17:
I'm just working on a complete mechanical rework of the extruder on my DaVinci 1.0A as I had a lot of trouble with stuck filament. Currently I'm printing PET what is softer than ABS what cause more trouble. It's a pitty that I didn't read this tread before as this may have been solved just by software. But anyway, I will now mount the sensor plate to the new idler roller to have the benefit of this feature too! Latest post is from Nov.2015 - is this feature now released in 0.92?

luc-github, 2016-03-17:
not released yet in master - It should be part of dev branch 0.92.6 but got delay due to my moving 
so could not verify anything without printer  - I will be back on business next week as printer is arrived, just need setup a location for it and be sure moving from 110V to 220V won't burn my printer :fire:

tecan, 2017-03-24:
offering 100 dollar reward for this working properly.

axs-gentoo, 2017-04-28:
Sorry, I never did have time to test this.

If I am reading correctly, the issue here is that the action taken when a jam is detected is getting inserted at the end of the queue (and hence takes way too long to actually occur) right?  

How is the no-filament action handled, does that not immediately stop the print or does it keep going until the queue empties too?  (i realize there's still filament to be used when the sensor triggers, and so the printer does have a bit of time before it actually runs out of filament, so maybe they never bothered with an immediate action in that case either)

axs-gentoo, 2017-04-28:
Along these lines though, what about simply aborting the print when a jam is detected?  It's not an ideal situation (a de-clog action would be a lot better) but in my use-case I just don't want the thing to keep going for an hour with the extruder gear chewing up filament (and who knows what burning in the nozzle) when the head is jammed.  Does this make any sense for anyone else?

jph290, 2018-04-02:
@luc-github is jam detection available and working in 0.92.10?  I would like to have jam detection on.  Also, there is some sort of encoder on the PCB with the filament sensor?  Does this sensor work translucent filaments?  I just bought some limit switches, but if there is already jam detection hardware I would rather use that.

luc-github, 2018-04-02:
the pin for the jam sensor is known, repetier has a jam sensor support in tbeory, but I never had time to  check if it is working and compatible with davinci one.
the filament sensor is an optical sensor which does not support translucent filament

jph290, 2018-04-05:
@luc-github From what I have been reading online the filament sensor is sending back 2.5V from the photo-transistor.  Is there any reason the input can not handle a limit switch instead?  It should be like a regular input but it would be returning 5V.  I have been trying to find anyone that has replaced the filament sensor, but can only file people that have switched to RAMPS.

luc-github, 2018-04-05:
davinci is a due based which use 3.3v max  not 5v like other arduino

jph290, 2018-04-05:
So is the optical sensor supplied with 5V or 3.3V?  Did they just use resistance to get down to the correct voltage?

jph290, 2018-04-05:
I think there is 3.3V on the power supply connector, I think I will just come off that instead of the 5V that was going to the extruder board.  I have already put inline terminals on the 5V anyway.  Thanks for the information, I did not realize the inputs were 3.3V.

tecan, 2018-06-03:
https://www.thingiverse.com/thing:2934923, it stopped the jamming i was having.

luc-github, 2018-06-03:
Well I never had jam issue, even when I was trying to print PLA - but I used several tricks to avoid that and quality was not as good as my flashforge, so I droped the pla with Davinci
I kept Davinci for ABS because of enclosure which avoid wrapping  and use flashforge for PLA ^_^

luc-github, 2018-12-09:
Jam sensor support is old and actually never really missed so I close issue

closed 2018-12-09 by luc-github