luc-github/Repetier-Firmware-4-Davinci #28: Add layer information in main menu

luc-github, 2015-01-14:
Issue pulled from 0.91 https://github.com/luc-github/Repetier-Firmware/issues/10

luc-github, 2015-01-21:
One easy solution would be a GCODE post processing script - in python ?

limited660, 2015-12-28:
Could a custom code be added for each layer like M456 and have the firmware count the number of instances and then increment another number as each is read?

luc-github, 2015-12-28:
yes but it same as my previous suggestion, you need a post slicer processing action, not all slicer allow it

luc-github, 2015-12-30:
seems layer info is available now but only for SDCard printing
```
            Com::printF(PSTR(",\"currentLayer\":"));
#if SDSUPPORT
            if (sd.sdactive && sd.fileInfo.layerHeight > 0) { // ONLY CAN TELL WHEN SD IS PRINTING
                Com::print((int) (currentPosition[Z_AXIS] / sd.fileInfo.layerHeight));
            } else Com::print('0');
#else
            Com::printF(PSTR("-1"));
#endif
```

luc-github, 2018-12-09:
As major use is SD card print because USB native is not really reliable I think issue can be closed

closed 2018-12-09 by luc-github