luc-github/Repetier-Firmware-4-Davinci #29: Unknown command:Nxxx M107

oscahie, 2015-01-14:
Back to business with my davinci 1.0 loaded with Repetier 0.92 (latest version from master), and I see the following messages in the console when starting a print job from Repetier Host:

23:41:28.613 : X:-33.00 Y:-12.00 Z:0.000 E:0.0000
23:41:28.641 : Unknown command:N153 M107
23:54:11.151 : Unknown command:N187 M107
23:54:11.153 : Printing layer 1 of 60

00:25:31.478 : X:-33.00 Y:-12.00 Z:0.000 E:-3.0000
00:25:31.483 : Unknown command:N6707 M107

00:34:12.212 : X:-33.00 Y:-12.00 Z:0.000 E:0.0000
00:34:12.217 : Unknown command:N70 M107
00:34:21.686 : Unknown command:N100 M107

Don't recall ever seen these before, with 0.91.

luc-github, 2015-01-15:
by default you cannot control extruder fan as it is used to cool extruder when temperature is over 50 degres so commands - M106 - Fan on
- M107 - Fan off won't work
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1 will allow to control fan but this is for PLA extrusion not extruder cooling
So if you want to do this you must add an additional fan for permanently cool extruder and use the fan managed by commands for extrusion

on 0.91 command was ignored if FAN CONTROL was not activated - in 0.92 commands are implemented based on compilation directives
#if FAN_PIN>-1 && FEATURE_FAN_CONTROL
case 106: // M106 Fan On

and by default FEATURE_FAN_CONTROL is set to 0

to avoid error you must uncheck enable cooling in repetier in Trancheur or slicer tab as this feature is not availble if you do not activate it in sources

closed 2015-01-15 by luc-github

oscahie, 2015-01-15:
Ah, that makes sense. I'm running now with REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 while I figure how to properly tap power for the fan.

luc-github, 2015-01-15:
0.92 is more verbose when have unknown command and more strich when doing compilation

BTW : do you still have LCD issue ? we have changed a lot to fix starting issue - one is still open but should be closed today

oscahie, 2015-01-15:
Yes, I've been following the development on the LCD issues. No problems yesterday in the first run but I'll keep an eye on it.