luc-github/Repetier-Firmware-4-Davinci #30: Build fails on Arduino Nightly (1.6.0) Jan 14

squoyster, 2015-01-15:
It looks like the current nightly builds of Arduino have incompatible changes with variants.cpp.  I was able to build by copying the "hardware" directory from 1.5.8.

drbogger, 2015-01-15:
Did you use the variants.cpp included with this firmware?

squoyster, 2015-01-15:
Yes, along with USBHost.cpp.

drbogger, 2015-01-15:
Must be a change in Arduino itself.

luc-github, 2015-01-15:
May I know where you got the 1.6.0 ?
I know only 1.0.6 (incompatible as no due)
and 1.5.X the current branch of arduino on github

squoyster, 2015-01-15:
I got that from arduino.cc.  http://arduino.cc/download.php?f=/arduino-nightly-windows.zip

luc-github, 2015-01-15:
Ho Seems description in github and in revision.txt are different
1.5.x vs ARDUINO 1.6.0rc2

luc-github, 2015-01-15:
Any reason to use nightly build vs the stable version ?

squoyster, 2015-01-15:
I was unable to install 1.5.8 from the installer on my new Windows 8.1 machine.  It appeared to be corrupt.  I tried building 1.5.8 from source but ran into issues with the com libraries.  I decided to try the nightly and it worked after the change I mentioned.  I'm not sure if the installer that's out on arduino.cc is corrupt or if it's an issue with my PC.

luc-github, 2015-01-15:
Windows 8.1 is painful.... never had issue with my win 7, OSX neither linux but my wife is crazy with all compatibility issues she had on her windows 8.1...

squoyster, 2015-01-15:
I know what you mean.  I've been using my old  MacBook with no issues, but wanted to upgrade to a faster processor for Slic3r, etc. :)

luc-github, 2015-01-15:
I plan to buy new one also this year - was waiting for broadwell but I am worry to get windows 8.1...not sure coming update will help....

squoyster, 2015-01-15:
Broadwell would be nice.  I found a Shuttle DS61 small form factor PC with 6 USB + RS232 and parallel.  I'm using that with my DaVinci and have a port available for a CNC I want to buy.  Nice and compact.  Works well with the i7 3770S and 16GB ram.

luc-github, 2015-01-15:
What software you plan to use with your CNC ?

squoyster, 2015-01-15:
I haven't thought through all the details yet.  Something FOSS, most likely, if I get a machine with an Arduino controller.  I want the machine for routing PCBs for the most part.

luc-github, 2015-01-15:
OK - I was asking because I am building a new printer and now thinking may be I should build a CNC first to help to build part for printer but no idea about kind of software for milling

garry572, 2015-01-15:
I already have a CNC mill, I bought the Da Vinci to make some parts for it, I use Mach3 with mine, plus CAD/CAM software

luc-github, 2015-01-15:
I read about MACH3 seems very used - but license looks expensive - does it worth it ?
For CAD I get used to build part with blender and now playing with Solidworks - it is like night and day for precise measurment and ease of use but did not find good way to edit STL files so still use blender.

garry572, 2015-01-15:
Mach3 is kind of expensive think it about $200 but I also bought Dolphin CAD/CAM as I use it to produce income at times.  Never tried blender but have solidworks but haven't ventured far yet I started off thinking I was going to do PCB, ended up doing everything else. Front panels, engraving, labels

garry572, 2015-01-15:
My CNC project has grown over the years, replaced the stepper motors with servos,  upgraded the spindle to a water cooled 800W only trouble now its, it becoming unforgiving of mistakes

luc-github, 2015-01-15:
Interesting  - thanks for sharing ^_^

garry572, 2015-01-15:
sure anytime

luc-github, 2015-01-16:
I left issue in keep watching but won't work on this before official release as it is noght build it is not really tested and can change anytime - actually everynight

Doudou14, 2015-01-16:
I have Arduino 1.5.8 on Windows 8 X64 and it works very well

luc-github, 2015-02-13:
I did a new version of variant.cpp for 1.6.0, it compiles but cannot test yet on printer will do next week

luc-github, 2015-02-16:
Variant.cpp updated by https://github.com/luc-github/Repetier-Firmware-0.92/commit/4d5d66d2eaed2fd8c98f6bebaa836839edf12d71
I did not put USBCore.cpp as now have watchdog - actually it is same as 1.5.8

closed 2015-02-16 by luc-github