luc-github/Repetier-Firmware-4-Davinci #31: Error compiling

lyweilian, 2015-01-15:
I pulled down the firmware this morning and tried to compile in Arduino 1.5.8 and got this error.

Arduino: 1.5.8 (Windows 7), Board: "Arduino Due (Native USB Port)"

HAL.cpp: In static member function 'static void HAL::setupSdEeprom()':
HAL.cpp:69:8: error: 'class SDCard' has no member named 'setupEeprom'
     sd.setupEeprom(sdEepromImage, SDEEPROM_SIZE);
        ^
HAL.cpp: In static member function 'static bool HAL::syncSdEeprom()':
HAL.cpp:78:19: error: 'class SDCard' has no member named 'syncEeprom'
         return sd.syncEeprom();
                   ^
HAL.cpp: In static member function 'static void HAL::resetHardware()':
HAL.cpp:257:21: error: 'playsound' was not declared in this scope
  playsound (1000,400);
                     ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Any clues? I did copy the files usbcore and variants as instructed.

luc-github, 2015-01-16:
I do not have this issue
Do you use Git to synchronize or do you take the ZIP file from GitHub ?
Do you already have some arduino library installed that could overide the FW one like SDCard ?

lyweilian, 2015-01-16:
I did a fresh install of IDE 1.5.8. I did not install any additional plugins and I downloaded via zip.

luc-github, 2015-01-16:
well then your extraction zip may have a problem - I did also a zip download and no issue
what OS do you use ?

lyweilian, 2015-01-16:
I use Win7 pro.

lyweilian, 2015-01-16:
Do I have to build with USB plugged into a reset Da Vinci?

luc-github, 2015-01-16:
No , no need to have printer plugged to build 
do you change anything in files after extraction ?

lyweilian, 2015-01-16:
I copied the variants.cpp and usbcore.cpp as per instructions after fresh install. Should I try to uninstall and reinstall?

luc-github, 2015-01-16:
try to redownload zip - I doubt issue come from arduino itself and variants.cpp / usbcore.cpp
also check Documents\Arduino\libraries to see if it is empty

lyweilian, 2015-01-16:
Luc, I just tried a fresh install without replacing the variants.cpp and usbcore.cpp. It worked. Thank you for following up with me. It it building now. Thanks again! Case closed. :)

luc-github, 2015-01-16:
so was zip corruption
You are welcome

closed 2015-01-16 by luc-github

MoBohnsack, 2016-07-02:
Hey, i also have this problem.

''NVIC_EncodePriority' was not declared in this scope"

But on my own a fresh install without replacing the variants.cpp and usbcore.cpp wont work.

Im using win7 pro 64bit and Arduino 1.6.8 but i also tried it with 1.6.5.

Full error message:

> Arduino: 1.6.8 (Windows 7), Board: "Arduino Duemilanove or Diecimila, ATmega328"

In file included from sketch\Repetier.h:370:0,

                 from sketch\Commands.cpp:22:

sketch\HAL.h: In member function 'void InterruptProtectedBlock::protect()':

HAL.h:164: error: 'NVIC_EncodePriority' was not declared in this scope

         __set_BASEPRI(NVIC_EncodePriority(4, 3, 0));

                                                  ^

HAL.h:164: error: '__set_BASEPRI' was not declared in this scope

         __set_BASEPRI(NVIC_EncodePriority(4, 3, 0));

                                                   ^

sketch\HAL.h: In member function 'void InterruptProtectedBlock::unprotect()':

HAL.h:168: error: '__set_BASEPRI' was not declared in this scope

         __set_BASEPRI(0);

                        ^

sketch\HAL.h: In constructor 'InterruptProtectedBlock::InterruptProtectedBlock(bool)':

HAL.h:173: error: 'NVIC_EncodePriority' was not declared in this scope

         __set_BASEPRI(NVIC_EncodePriority(4, 3, 0));

                                                  ^

HAL.h:173: error: '__set_BASEPRI' was not declared in this scope

         __set_BASEPRI(NVIC_EncodePriority(4, 3, 0));

                                                   ^

sketch\HAL.h: In destructor 'InterruptProtectedBlock::~InterruptProtectedBlock()':

HAL.h:177: error: '__set_BASEPRI' was not declared in this scope

         __set_BASEPRI(0);

                        ^

sketch\HAL.h: In static member function 'static void HAL::hwSetup()':

HAL.h:297: error: 'TimeTick_Configure' was not declared in this scope

         TimeTick_Configure(F_CPU_TRUE);

                                      ^

In file included from sketch\Repetier.h:370:0,

[... 2093 lines cut ...]

sketch\Commands.cpp:2104:53: note: in expansion of macro 'READ'

                             if(EEPROM::busesensor &&READ(FIL_SENSOR2_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);

                                                     ^

HAL.h:145: error: 'PIO_Get' was not declared in this scope

 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)

                                                                                                 ^

sketch\Commands.cpp:2104:53: note: in expansion of macro 'READ'

                             if(EEPROM::busesensor &&READ(FIL_SENSOR2_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);

                                                     ^

exit status 1
'NVIC_EncodePriority' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

MoBohnsack, 2016-07-02:
Just tryt it with Linux Same error message:

http://pastebin.com/qGpTqXY0

luc-github, 2016-07-02:
Yes you did not selected the correct board I see :
```Arduino: 1.6.8 (Linux), Board: "Arduino Duemilanove or Diecimila, ATmega328"````

you need to select Due Native port - be sure you installed Due board
1 - Go to board manager
[image]

2 - Check / Install Due doard
[image]

3 - Select Due board in menu
[image]

4 - Once selected compilation should be fine
[image]

danny1231233, 2019-08-02:
i am also having an issue. i get the following error upon compiling. 

Arduino: 1.5.8 (Windows 8), Board: "Arduino Due (Native USB Port)"

In file included from Repetier.h:572:0,
                 from BedLeveling.cpp:107:
HAL.h: In static member function 'static void HAL::serialSetBaudrate(long int)':
HAL.h:684:14: error: 'class UARTClass' has no member named 'setInterruptPriority'
       Serial.setInterruptPriority(1);
              ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.