luc-github/Repetier-Firmware-4-Davinci #32: G10 and G11 commands

luc-github, 2015-01-18:
these commands are now available in repetier, need to importe them

luc-github, 2015-01-18:
https://github.com/repetier/Repetier-Firmware/commit/54298f82172c04ad39a24c73263982ebd8de1650

Doudou14, 2015-01-20:
If I can, I do it in a few hours

luc-github, 2015-01-20:
Would be great - I am little bit busy this week 
But take your time - there is no hurry so far

luc-github, 2015-02-03:
Sorry very busy - https://github.com/luc-github/Repetier-Firmware-0.92/pull/35
I keep it in tracker

luc-github, 2015-02-26:
M360 done : https://github.com/luc-github/Repetier-Firmware-0.92/commit/d9bbcf3c5b4201d9d19ccc0ff429794e6def0f8e
M355 done: https://github.com/luc-github/Repetier-Firmware-0.92/commit/d46b1e12f31ff44711860a98bce4e3d371d70706

luc-github, 2015-03-20:
All done - M600 is slightly modified as buggy and I think it is more clear to use Load/Unload menu
https://github.com/luc-github/Repetier-Firmware-0.92/commit/78a45d94b1f597cdf2deedb59553d9a3b04077e7

closed 2015-03-20 by luc-github