luc-github/Repetier-Firmware-4-Davinci #34: When heater is decoupled printer do not stop current print

luc-github, 2015-01-20:
If any decoupled issue printer should stop printing if it is printing - currently extruder go to dry run and stop heating but printer continue to move like printing is still ongoing

garry572, 2015-01-20:
yes i had the same happen to me

luc-github, 2015-01-20:
I think if heater is decoupled everything should be stopped - right ?
I am also thinking about audio alert - as I am not close to printer so I relly on printing noise to know process is ongoing

garry572, 2015-01-20:
sounds like a good idea

Doudou14, 2015-01-22:
I do not understand is if we remove the USB cable during printing? (Not on the SD card)?

luc-github, 2015-01-22:
nothing related to USB - it is related to sensor and heating - there is a feature in 0.92 to ensure control of heat
if it is heating but temperature does not raise of some degres during a specific period the fw consider there is an issue heater and sensor are decoupled and stop heating, generate sensor issue

luc-github, 2015-02-26:
done https://github.com/luc-github/Repetier-Firmware-0.92/commit/b501f1a7acd54fc583c9a046b7c4fd54531c285d

closed 2015-02-26 by luc-github