luc-github/Repetier-Firmware-4-Davinci #37: Safeguard to avoid crashing into the Z-home sensor

oscahie, 2015-01-21:
I've had this kind of crash a few times on my 1.0. If you manually home the X and Y axis in the wrong order (first Y, then X), chances are you'll end up with the extruder's fan slamming into the Z-home sensor. Same thing if you try to home Y with the head more or less lined up with the sensor. This is a very unfortunate design overlook, but I guess it could be fixed in the firmware. 

I believe something like this was already in the early versions of the firmware (with the ENDSTOP_Y_BACK_ON_HOME thing), but it had the side-effect of not letting the head home in the right position. What if instead we'd forbid the printer head from moving into negative Y coordinates at any time unless X is in the negative too. So homing Y and then X should result in the head moving only up to Y0, and then X-HOME. Another approach would be to simply forbid any X move while Y is in the negative. Or rather than forbid the movement, perform it but dodging the Z-home sensor.

Does this sound doable or crazy?

luc-github, 2015-01-21:
well forbid printer to go to negative is not an option as in negative position it is park for 1.0 and the autolevel plate for 2.0

I am not sure to understand the problem as 1.0 and 2.0 have little bit different configuration on the back - do you have drawing or picture to show the problem ? it will be more easy to understand the problem 
Sorry to ask

oscahie, 2015-01-21:
Ok, it's going to be embarrassing to show my Paint drawing skills but here we go :)

[image]

As you can see, if the printer head is on Y = -12 (i.e. Y home) and you want to home X then the back of the printer head (basically the fan) will hit the Z endstop. This won't happen at Y = 0 but only when Y is on some negative values.

The same issue is indeed applicable to the autoleveling contact points, but only when Z = 0 (or nearby), so it's more limited.

luc-github, 2015-01-21:
It is better than what I can do - so no worry ;)
Ok I understand
now I have an additional question - why Y can be -12  if out of parking position ? - this is not a "normal" position - just to understand the context

oscahie, 2015-01-21:
During regular printing it should never go to negative position indeed. This is actually only a problem for manual jogging and gcode scripts. Say for example that the printer head is on the middle on the bed (you've been doing some manual work, bed leveling or whatever) then you use RH to hit the 'home Y' button followed by 'home X' button . In that case you'd probably get a crash. This has happened to me quite a few times... :(

luc-github, 2015-01-21:
so as I mentioned only half of solution can be implemented 
if Homing X and Y already homed so In known position : if Y < 0 then move Y to 0 and then homing X
if Homing Y and X already homed so In known position : if X < 0 then move X to 0 and then homing Y
but 
if Homing X and Y not homed => no solution
if Homing Y and X not homed => no solution

oscahie, 2015-01-21:
Yeah, I suppose that any workaround will require both X and Y to be known positions (homed or not should not matter).

Maybe it'd a better idea to fix this at hardware level though, i.e. print a replacement Z-endstop holder that gives the proper clearance...

luc-github, 2015-01-21:
on my duo the X sensor do not allow to go to X negative - so far from Z sensor - it is on left of printer
where is it located on 1.0 ?

oscahie, 2015-01-21:
Your X endstop is on the left side? Or do you actually mean the Y sensor?

luc-github, 2015-01-21:
sorry y sensor, on left ,yes, x sensor is on carriage on right
 it is late....

oscahie, 2015-01-21:
:) Yes, the end stop is far in the back enough to allow the extruder to go negative Y (about 12 mm). Another way to fix this issue would be to move both the Y endstop and the dripbox (or at least the cleaning bits) about 1cm forward..

luc-github, 2015-01-21:
the Y support is in plastic you should be able to print a modified one :smile:

Doudou14, 2015-01-22:
I'll try to add a function if I have time, it happened to me several times in repetier host and it's pretty bad for the belt - pulley - Engine

luc-github, 2015-01-23:
To be honest FW won't be a good solution as need a known position to be effective, as mentioned it is design problem on 1.0, it is not on 2.0 they fixed it.
the best would be to move the Y end stop few mm in front to avoid the head to hit the Z sensor
and may be to move the dripox cleaner blade accordingly.

oscahie, 2015-01-23:
I just realized that since I'm planning to move to an E3Dv6 extruder very soon (already purchased it), this will likely not be an issue anymore, as the E3D has a lot more clearance, I think.

Doudou14, 2015-01-23:
If moves the Y sensor, the cleaning tank will be too far.

If you can do something like this:
If sensor X = 0 (or 1 I do not know)
Y homing
else
X homing
Y homing
endif

luc-github, 2015-01-23:
if X is after sensor and Y is -12 this will also cause trouble 
If Z is 0 you may also hit autolevel plate
If Y is in front of blade and you home X you also hit something on 1.0 the blade by the side

no mention that if you have Y home function is to Home Y only, not X and Y, in this case it is like homing all what ever you choose, this is no more safe area but remove Y homing

As mention earlier - ther is no "simple" function to solve the problem - it is just moving the problem in another place

garry572, 2015-01-23:
Why not move z 10 so it's out of the way then home x and y then home z

luc-github, 2015-01-23:
yes another example : this is based on fact Z is homed to be sure position is not max, 
so do to Y homing alone : move z down , home X and then home X - it become complexe and not what user expect when he wants just home Y. so if home X is as complexe: Z down, Home Y, move Y to avoid sensor then Home X, but . .. Home Y need first to Home X as mentioned before...,   again it is chicken and egg.

Back to original point : when homing Y alone or X alone, how to be sure not to hit Z sensor on 1.0 ?
for me no good solution  and a least no easy one, as always have a drawback.
 
The safe one is move Y sensor to the limit to avoid the posibility to hit
Another solution is to add a second Y sensor on the hit position (Z sensor) to stop the homing and avoid the hit, it won't give a good homing but avoid the colision, and same for X ?

oscahie, 2015-01-30:
I've installed the E3D v6 hotend and, as expected, I no longer has this problem, so maybe we can close this thread if there is really not much that can be done on the software front.

luc-github, 2015-02-03:
I get used to have request to fix hardware issue solved by software,  but unfortunatly cannot always solve the problem, but bring always some complexity in code which become hard to maintain with this kind of "patch". 
Sorry.

closed 2015-02-03 by luc-github