luc-github/Repetier-Firmware-4-Davinci #38: Simplify 3D stop randomly to print then resume after few seconds

luc-github, 2015-01-22:
When this happen on serial monitor I see bunch of "wait" commands and on LCD the buffer decreasing, Can navigate in printer menu when this happen - so printer is not frozen and this does not trigger the watchdog
After few seconds print resume, I did not observed any crash neiter print failure but, seems weird.
This never happened with repetier host so far
I have set same setting for communication in S3D than for repetier
202 cache / Hw flow control /  baud rate 230400

not sure how to address this

luc-github, 2015-01-23:
Move to 63 in cache, the default value, and remove the number ack in FW. as it always showed "ok 0" in simplify when the command expected by S3D is "ok"
 this time the printer stop at heating bed but do not send the  heat extruder in gcode script. got a bench of "wait" and even one "wai"
Because S3D do not use the command control of repetier it may be a usb connection issue, so I increase the retry in usbcore of the usb modification from 10K to 100K
at it is random need to test 
next is to change usb cable to one with ferrite to limit signal polution

oscahie, 2015-01-24:
Maybe try increasing the 'Communication timeout' value from 10 to 20?

luc-github, 2015-01-24:
Well I observed that buffer is empty when waiting - so a timeout of 20 second should not change anything
I did not printed today so cannot see yet if the modification of usb patch changed something - at least I did not see after I did it but need to be in front of printer on good moment to so need more test - I have a 10h+ print under preparation, I should launch it Sunday late afternoon, so I will be able to check if happen again or not

garry572, 2015-01-24:
I had issues with the USB connection, on any of the USB3 ports I always had communication errors, moving back to the USB 2 ports the problems dissappeared

luc-github, 2015-01-25:
Well it is connected to "old" laptop (4 years) not having USB3 but port connector is little bit loose so I guess usb connection is not optimale

luc-github, 2015-02-17:
I do not use direct connection anymore but flashair SDCard and SD Card Printing and no issue
Anyway I removed the USBCore.cpp patch for the new 1.6.0 ide, so it give oportunity to test and see if it is part of the root cause 
Feedbacks are welcome as usual

leghk, 2015-02-26:
I just compiled and installed a couple of days ago with 1.6.0 (after the USBCore.cpp removal) and this pause bug happens to me two or three times per hour of printing. The communication log shows:

__WARNING: Firmware unresponsive. Attempting to force continue.,,__

That is during USB printing from a Mac Mini .

luc-github, 2015-02-26:
1.6.0 has some USB issue https://github.com/luc-github/Repetier-Firmware-0.92/issues/51
better use 1.5.8 right now

luc-github, 2015-03-20:
I read on some forum that increasing the buffer on Simplify helped but I do not think it is the problem and the solution,  because computer is faster than Davinci board, GCODE time generation does not explain why printer can be stuck waiting command for several seconds.

I have recently sync the code from 0.92.3 and set the echo command back to 1 like in repetier in case of S3D use this as a check, and should help to see what command are lost

gamecompiler, 2015-04-07:
Id had have this problem too, id  increased the buffer to 2024.
Works great now.

luc-github, 2015-04-07:
Buffer from S3D right ?

interesting, I have seen other people also increasing buffer saying was better but did not tested 

this is good to know - thanks a lot for sharing
let see if anyone can also feedback this is the magic number, so I can close issue -  appreciate your help

gamecompiler, 2015-04-07:
Yes, and id had have the same problems in Repetier and S3D.
2024 had fix the issue

luc-github, 2015-04-07:
Repetier also ? wow first time I heard about this

pallemar, 2015-05-13:
The problem may also be becourse the slicer put in a wait time, to allow the filament to cool, before continuing the printing, if you see some video about how slicer works, you will see it explained ;)

luc-github, 2015-05-13:
During my tests the issue happen in middle of layer, not at the end when the slic3r may do some delay - also my layer was a big one - so at tha time each layer take more time than minimum time set in slicer - 
I am pretty confident it is  communication issue - reading at the thread I now realize that I forget to test at 115200 baud rate to check if it could solve the problem...

pallemar, 2015-05-13:
You are right the problem should not come in big prints :)

thevisad, 2015-05-28:
I made all the changes here and am seeing the same issue still present on the machine. The printer is pausing, buffer dropping to 0 and failing to proceed. One comment in the robo3d forums indicates that they upped the timeout to 20s which solved the problem. I will give this an attempt later tonight, I can pop out the SD card, but I prefer to print via USB. I am using a 2012 iMac with the software.

luc-github, 2015-05-28:
Ok thanks for your feedback, waiting for your results

efespilsen03, 2015-06-24:
hello Luc, i am boolat :) 
My setting:
baud rate: 250000
serial cache size: 2024
Wait for startup command: enable

no problem for 5 days!
it runs like a Ferrari!
thank you for your help LUC :+1:

luc-github, 2015-06-25:
good to know - thanks for sharing

what version of S3D are you using with these settings  2.2 or 3.0 ?

efespilsen03, 2015-06-25:
use 2.2 simplify3d 
Windows 7

luc-github, 2015-06-25:
Ok thanks a lot so I close issue 
again thanks for your feedback

luc-github, 2015-09-04:
seems there is a fix for this behavior related to native port :
github.com/repetier/Repetier-Firmware/pull/425

Will add it later

luc-github, 2015-11-09:
Release 0.92.6 is done but need to see if fix is implemented or still testing

luc-github, 2015-11-11:
Fix seems in 0.92.6, I created the test branch and will start the sync soon

thomastech, 2015-11-11:
That is great news.
- Thomas

luc-github, 2016-01-02:
0.92.6-sync branch is available for testing

Cannonized, 2016-02-05:
I'm having the same issue. Buffer dropping to zero where is the fix? I've been up all night and I'm cranky I apologize for the attitude at the moment. Grrrr printing with DaVinci 1.0!!!

luc-github, 2016-02-05:
did you tried the 0.92.6-sync branch ?
https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync

Cannonized, 2016-02-05:
Thanks luc.  I did find it. I apologize I'm not at my finest quality right now.

luc-github, 2016-02-05:
additionally if you have a 1.0 it is easy to put and extender or a wifi card to print from wifi
printing from host is always subject to possible errors - that is why XYZ print from SDcard on stock FW

Cannonized, 2016-02-05:
How do I go about loading to the SD card without removing from the back of the printer? Do I just name my file as sample## then unplug and hit print from menu? 

Also I assume I need the new src folder from the sync branch.?

luc-github, 2016-02-05:
Yes to plug in an extender or wifi SDCard you need to remove back - but after no need anymore
Repetier list content of file on SDCard no need to give sample## as name just extension is necessary
You even can launch from repetier host also from SDCard management menu

Yes same as master but from sync branch

luc-github, 2016-02-05:
here an example of extender :
http://www.ebay.com.au/itm/Extension-SD-SDHC-Card-Reader-Cable-Cord-Adapter-for-GPS-DVD-LED-Screen-/111183763834?hash=item19e311997a

and here the wificard: http://www.miniinthebox.com/fr/flash-toshiba-8gb-sans-fil-de-l-air-carte-memoire-sdhc_p1679330.html?currency=EUR&litb_from=paid_adwords_shopping

I only use flash air on all my printers - no usb connection - and wifi copy is very fast

Cannonized, 2016-02-05:
OK so do I have to the variants.cpp exchange again or just upload the new repetier.ino file?
I'll look into the cd card mod later. Try uploading from the host but I think I have to keep my laptop connected to keep it printing maybe that will change with the new fw

http://www.ebay.com.au/itm/Extension-SD-SDHC-Card-Reader-Cable-Cord-Adapter-for-GPS-DVD-LED-Screen-/111183763834?hash=item19e311997a

luc-github, 2016-02-05:
variant.cpp and usbcore.cpp are for your arduino IDE installation - if you still use same not need to update again as previous one are still there - but if you do new Arduino IDE installation you need to apply the variant.cpp and usbcore.cpp

Uploading from host is working but it take ages when using wifi or direct copy to SDCard and then plug SDCard to extender is immediate

Cannonized, 2016-02-05:
OK I have the new sync loaded. Print is readying itself. I will update either way to see if the fix takes. 90% of my prints at this point  for this project are 8+ hrs.

Cannonized, 2016-02-06:
Still having issues with the buffer communication from the host. Managed to get it to print for 3 and a half hours before the buffer issue became a problem and the printer stopped. Uploading to the SD card. Now while the code is running from the sd card can I remove the computer to use it elsewhere or do I need it attached still. Slightly nervous on unplugging it from the printer.

 
  On Fri, Feb 5, 2016 at 1:57 PM, clifford cannon<cliffordcannon101@yahoo.com> wrote:   OK I have the new sync loaded. Print is readying itself. I will update either way to see if the fix takes. 90% of my prints at this point  for this project are 8+ hrs.

luc-github, 2016-02-06:
you do not need computer connect when printing from SDCard
So you mean the new code did not fix the issue ? that is bad

Cannonized, 2016-02-06:
Didn't seem to fix it for me. I tried a different port on my laptop got almost 4 hours out of it this time. Going to pick up a better USB cable when I can see if that makes a difference so far on the SD print the buffer is a solid 32 which is nice to see for a change. At least I've somewhat got it back to stock functionality where I can load the file and let my kids watch movies as we await dad's latest creation.

luc-github, 2016-02-06:
well USB connection need a good cable, no electric appliance around, be sure PC do not go in sleep mode, a lot of factor can bring noise on it and make printing fail

luc-github, 2016-02-06:
I will get back my printer on March (if no issue) so cannot test right now but will check why fix do not work

thevisad, 2016-02-06:
I had an issue on my Mac (and only the mac) with simplify3d and the printer
stopping constantly. As soon as I switched over to a PC I stopped having
the printer issues. The printer on the Mac acted exactly like the comments
in here. I tried the other solutions offered but none worked. The only work
around on the Mac was to print from the SD card.

Cannonized, 2016-02-06:
I'm on pc using slic3r from the host. SD printing works amazing. Was finally able to get the print with only a few strings from badly generated supports but that's another issue. Now to figure out how to program in a colour change part way through. I liked the idea of printing from the host so I could use all the filament pause it when there was about 10cm left switch colours and resume. At least I'm printing again and can finally get the parts stage off the back burner for this project.

luc-github, 2016-02-06:
@Cannonized you can pause when printing from SD

Cannonized, 2016-02-06:
The moment I unplug the computer from the printer it just resets. I can't see the SD file in the print file selection on the printer just from host. Am I doing something wrong?

luc-github, 2016-02-06:
Did you applied the usbcore.cpp from my github ? this prevent printer reset when usb connection is active and stopped not properly

does your file has an extension ? if no extension or extension is bin or dat file is not displayed on lcd

Cannonized, 2016-02-06:
Oh maybe that's what I'm doing wrong is not saving.gcode. I didn't use USB core.cpp uploaded with arduino 1.6.5 r2, deleted the core call in one of the libraries. That's all in another thread though.

Cannonized, 2016-02-06:
None of the .guide files will print direct from the printer. It just shutters the print head and resets. Oh well at least printing with the computer attached and the file loaded on the SD card it works. 

 
  On Sat, Feb 6, 2016 at 1:45 PM, clifford cannon<cliffordcannon101@yahoo.com> wrote:   Oh maybe that's what I'm doing wrong is not saving.gcode. I didn't use USB core.cpp uploaded with arduino 1.6.5 r2, deleted the core call in one of the libraries. That's all in another thread though.

Cannonized, 2016-02-06:
All my .gcode files read 41 for size. Except offline_print.gcode

 
  On Sat, Feb 6, 2016 at 1:54 PM, clifford cannon<cliffordcannon101@yahoo.com> wrote:   None of the .guide files will print direct from the printer. It just shutters the print head and resets. Oh well at least printing with the computer attached and the file loaded on the SD card it works. 

 
  On Sat, Feb 6, 2016 at 1:45 PM, clifford cannon<cliffordcannon101@yahoo.com> wrote:   Oh maybe that's what I'm doing wrong is not saving.gcode. I didn't use USB core.cpp uploaded with arduino 1.6.5 r2, deleted the core call in one of the libraries. That's all in another thread though.

luc-github, 2016-02-06:
how do you copy files to SDCard ? what FW version do you use ?

Cannonized, 2016-02-06:
I'm using the latest one from your sync folder. Repetier host V1.6.0. Save file for SD printing from print preview section then upload from the SD card section. I'm going to buy the Toshiba air card that everyone recommends and give that a try eventually.

Cannonized, 2016-02-06:
If all else fails I've found an arduino idea 1.5.8 beta that I might try and see if any of that makes it all work the way it's supposed to. Replace both the variant and usbcore files. Perhaps it's the workaround that I had to do with arduino 1.6.5r2 that's the problem.

 
  On Sat, Feb 6, 2016 at 2:05 PM, clifford cannon<cliffordcannon101@yahoo.com> wrote:   I'm using the latest one from your sync folder. Repetier host V1.6.0. Save file for SD printing from print preview section then upload from the SD card section. I'm going to buy the Toshiba air card that everyone recommends and give that a try eventually.

luc-github, 2016-02-06:
if fix for due do not work - it is better to use master branch which is stable - sync branch is dev branch and not fully tested by me at least - for example I did not tested copy to SD functions

luc-github, 2016-02-06:
About IDE : I use 1.6.5 with module for DUE 1.6.4 and do not have any compilation issue, no need to to workaround for path issue
So you may need to uninstall all and delete the arduino directory in Appdata to be sure your installation is clean
the arduino files are easy to be mess up with different version, if previous is not properly deleted before new version

Cannonized, 2016-02-06:
OK I'll give the newest master a try. Again I had to do the work around not using the usbcore.cpp with arduino 1.6.5r2 that in itself could be the source of all my issues. Joys of modifications and hacking. Frustrating at times but oh so rewarding when it works right.

luc-github, 2016-06-07:
I may have a solution - but cannot confirm - seems USB pins share a pin with filament sensor
now I isolated the pin properly - before was not 
here the change https://github.com/luc-github/Repetier-Firmware-0.92/commit/d5f899f687414b729a9c05332fbcff764f83c0f6
I only applied to 1.6.8 as it is the one i tested

luc-github, 2016-08-08:
if no update/feedback I close issue

luc-github, 2016-08-08:
I consider issue solved in 0.92.10

closed 2016-08-08 by luc-github

bgauweiler, 2016-09-21:
I'm running into this issue with my freshly flashed Da Vinci 1.0A and Simplify3D if the print job takes more than 10 or 15 minutes to complete.  Printing pauses, after a few seconds, printing resumes.  I'm running the latest versions of your Repetier FW and S3D 3.1.0.

[3 more replies]