luc-github/Repetier-Firmware-4-Davinci #39: Add temperature profil for extruder

luc-github, 2015-01-28:
This would be linked to Preheat bed/extruder and used in bed leveling 
About load filament question is open:  if change filament from ABS to PLA need another entry than unload/load but change filament to use the correct temperature to unload then load and change profile

like ABS>PLA
profile = ABS
1  - unload : temperature = ABS
2 - load : temperature = ABS to purge 
3 - change profile to PLA
if PLA ->ABS
1 - unload : temperature = PLA
2 - load : temperature = ABS
3 - change profile to ABS

Preheat : 
1 extruder : follow profile ->Bed/Extruder
2 extruder : follow profile for extruder, follow hotest bed temperature profile for bed 

Bed leveling / cleaning nozzle : they follow preheat

tecan, 2015-04-25:
for a temprorary fix you could leave the ABS temperature setting at 190-200 and fan cooldown temp at 90 in the firmware. it fixes most of the things you want for bed leveling and changing filament's preheat.  also my feedrates got upped to Home X=75 Home Y=65 Home Z=5 works well.

opensourcefan, 2016-07-17:
I have been modifying my 1.0A to try and get PLA to work consistently and have been curious if the 230 unload and load temp is too high for PLA. I notice I get ooze at around 200 so loading at 230 seams too high risking heat creep. I've noticed I have to load from a cold state to keep the extruder as cool as possible, once it's heat soaked then I will almost always have a clog if I load the PLA after. Is that what you're thinking of changing?

luc-github, 2016-07-17:
currently load, unload, clean nozzle and bed level functions use ABS temperature, if you only use PLA it can be better to set PLA temperature instead
they are in EEPROM so you can change then from host or from printer menu

luc-github, 2018-12-09:
Temperature are now I EEPROM so I guess issue can be closed

closed 2018-12-09 by luc-github