luc-github/Repetier-Firmware-4-Davinci #40: Printer is not responding correctly to commands after a killed print

oscahie, 2015-01-31:
I've seen this bug several times already, as I'm experimenting with the new E3D hot-end and that involves many test prints that often get cancelled. 

I think I can repro it pretty much 100% with the following steps:

- Start a print from RH.
- After a while hit the Kill Print button in RH.
- Keep the extruder turned on to maintain the temp.
- Use the manual extrusion button in RH to extrude some filament.

At this point, rather than extruded filament, the print head starts moving slowly on the Y axis and towards the home position. I haven't tried much but I believe that things get very messy from there, as in some gcode commands no longer do what they should. I remember once that the head ended up moving in the opposite direction than it was commanded.

luc-github, 2015-02-01:
do you have the log of the repetier host ?

luc-github, 2015-02-01:
to cancel immediatly I suggest you to use a macro button in repetier host and set the command M50 - it is more clean and fast

oscahie, 2015-02-01:
I have this from the last time it happened, just minutes ago:

01:11:58.220 : N1779 G1 X73.656 Y96.675 E729.0869 *75
01:12:00.606 : N1780 M50 *8
01:12:00.618 : N1781 M104 T0 S0 *62
01:12:00.657 : N1782 M140 S0 *121
01:12:00.657 : start
01:12:00.658 : N1 G1 X-33 Y-12 F4800 *97
01:12:00.659 : SD print stopped by user.
01:12:00.694 : Printed filament:124.28m Printing time:1 days 19 hours 21 min
01:12:00.748 : Error:expected line 1 got 1782
01:12:00.749 : Resend:1
01:12:00.755 : Resend: N1 G1 X-33 Y-12 F4800 *97
01:12:00.757 : N2 M84 *61
01:12:01.108 : N1 M110 *2
01:12:01.113 : N1 M110 *2
01:12:01.117 : N2 M115 *4
01:12:01.118 : N3 M111 S6 *68
01:12:01.141 : FIRMWARE_NAME:Repetier_0.92 FIRMWARE_URL:https://github.com/luc-github/Repetier-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:DaVinci EXTRUDER_COUNT:1 REPETIER_PROTOCOL:2

Looks like RH is already sending M50 anyway?

luc-github, 2015-02-01:
you print from SD using host ?

oscahie, 2015-02-01:
No, I don't use the SD at all. I always print directly from host.

luc-github, 2015-02-01:
Ok 
and you said host send M50 by himself or after you add the macro M50

oscahie, 2015-02-01:
I only used the Kill Print button, no macros.

luc-github, 2015-02-01:
ho you already add M50 on run on kill script then ?
and temperatures stay up ? they are not set to 0 ?

oscahie, 2015-02-01:
Ahh, indeed, I do have M50 in the run on kill script! I do not remember adding this, must have been long ago. I'll know soon if this was causing the issue...

luc-github, 2015-02-01:
but temperature should be stopped - and you say temperature stay up - did you add other code in script ?

oscahie, 2015-02-01:
I did not say that. I do manually enable the extruder again right after the stop to keep the temp up and ready to print again.

luc-github, 2015-02-01:
Ho ok - I will try to reproduce later - I am printing now and no more connected to computer, so need to wait and will reconnect to repetier to try

oscahie, 2015-02-01:
Removed the M50 from the kill script, the issue still occurs. Here is a new host log:

01:54:44.222 : N1810 G0 X74.654 Y95.677 F3720 *94
01:54:46.562 : N1811 M50 *15
01:54:46.614 : N1812 M104 T0 S0 *59
01:54:46.708 : N1 M140 S0 *68
01:54:46.709 : start
01:54:46.709 : N2 G1 X-33 Y-12 F4800 *98
01:54:46.709 : SD print stopped by user.
01:54:46.737 : Printed filament:125.42m Printing time:1 days 19 hours 40 min
01:54:46.838 : N3 M84 *60
01:54:47.218 : N1 M110 *2
01:54:47.219 : N1 M110 *2
01:54:47.219 : N2 M115 *4
01:54:47.225 : N3 M111 S6 *68
01:54:47.261 : FIRMWARE_NAME:Repetier_0.92 FIRMWARE_URL:https://github.com/luc-github/Repetier-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:DaVinci EXTRUDER_COUNT:1 REPETIER_PROTOCOL:2
01:54:47.262 : Printed filament:125.42m Printing time:1 days 19 hours 40 min
01:54:47.266 : N4 T0 *30
01:54:47.267 : N5 M20 *52
01:54:47.268 : N6 M80 *61
01:54:47.270 : Begin file list
01:54:47.325 : ad_check_0001 2400
01:54:47.327 : eeprom.bin 2048
01:54:47.339 : Machine_Life.dat 10
01:54:47.339 : offline_print.gcode 256315
01:54:47.339 : SAMPLE01.gcode 647127
01:54:47.393 : SAMPLE02.gcode 426855
01:54:47.393 : SAMPLE03.gcode 3829421
01:54:47.393 : ad_check_0002 1400
01:54:47.394 : End file list
01:54:47.395 : N7 M220 S100 *70
01:54:47.399 : N8 M221 S100 *72
01:54:47.400 : N9 M111 S6 *78
01:54:47.404 : N10 T0 *43
01:55:05.377 : N17 G91 *7
01:55:05.378 : N18 G1 E10 F120 *0

After this the print head starts moving towards home Y, at a very slow pace (possibly F120?).
01:55:05.379 : N19 G90 *8

luc-github, 2015-02-01:
can you do a last test ? instead of kill print, just send M50

luc-github, 2015-02-01:
strange I still see M50 command in the log

oscahie, 2015-02-01:
Tried with the M50 macro, same problem. Obviously RH is sending M50 for the kill button too.

luc-github, 2015-02-01:
strange M50 is created by me not repetier, will check nehaviour when I have access to printer

oscahie, 2015-02-01:
Wait, I just double-checked and M50 was still there in the kill script! Then I tried killing a print with the button and this time it all worked well. So the bug is only with your custom M50 command, I guess.

luc-github, 2015-02-01:
ok make more sense, so I know where to look at then,

luc-github, 2015-02-03:
I miss something as I do not reproduce the issue
1 - start a print 
2 - kill print using M50 in kill print script
3- keep extruder hot
at this step Extruder is stick to print object as no movement

4- try to extrude 

oscahie are your doing some action between point 3 and 4 ?
as it will be hard to extrude when nozzle is stuck to print so I guess you move extruder in some place. I need to reproduce your exact action as - I do not see what you have

oscahie, 2015-02-03:
Well, immediately after a kill I always manually move the print head out of the printing area, but that's all. What I did notice is that if you home X and Y after the M50 then the bug does not reproduce anymore.

luc-github, 2015-02-03:
this what I was guessing but even moving extruder on X I did not reproduced
can you give the exact steps you use to move out of the printing area - so I can reproduce same behaviour like I was your shadow ;)

FYI - I already noticed this behaviour before but was never able to reproduce it on purpose- it happened randomly and was not linked to M50 - but if you can give a clear steps for reproduction I should be able to fix it

oscahie, 2015-02-03:
I just move the print head with my hand to the right :) I really do nothing more than what I described... make sure to be printing from host though, not from SD.

luc-github, 2015-02-03:
well what you just said is key you move manually not using commands - so it means steppers are disabled if you can manually move the head
I only used the manual tab sending commands that may explain the difference - I will check in your way

oscahie, 2015-02-03:
Aren't the steppers automatically disabled with M50? Or is RH disabling them?

luc-github, 2015-02-03:
I do not do implicitly - need to check the code what is doing this

Still did not reproduced

1 - start a print from host
2 - kill print using M50 in kill print script
3- keep extruder hot
4 - move manually head on right
5 -( extrude)
How do you extrude ? from printer or from host ?

oscahie, 2015-02-03:
Extrude from host. I rarely ever use the printer buttons.

luc-github, 2015-02-03:
Ok like me
but do not reproduce the problem X-X

luc-github, 2015-02-03:
I think I miss something that you do 
I wait print start , wait printer do several print (10/20 s) then kill print
then push head on right manually 
then click on temp for bed and extruder to keep temp high on manual tab 
then click extrude button (+50) - on manual tab 

all behave like expected

oscahie, 2015-02-03:
That's exactly what I do. Hmm maybe it's hardware dependent... certainly it's very weird that a extrude command ends up moving the Y stepper. I'll try again tonight (CET) anyway.

luc-github, 2015-02-03:
ok thanks - also please check if happen also if not moving by hand but using Jog control - without homing

oscahie, 2015-02-03:
Well, here is some more weirdness: if instead of moving the print head manually right after M50, I use the RH jogging controls to move some amount in either the X or Y axis, what actually happens is that the head moves to the home X,Y position, traveling diagonally (i.e. on both axis simultaneously). Immediately after the M50 I see this error on the console:

00:45:49.516 : N303 M50 *54
00:45:50.151 : N305 G0 X75.019 Y166.714 F3780 *93
00:45:50.158 : start
00:45:50.158 : SD print stopped by user.
00:45:50.159 : N1 M104 T0 S0 *0
00:45:50.193 : Printed filament:168.20m Printing time:2 days 10 hours 16 min
00:45:50.242 : Error:expected line 1 got 305
00:45:50.243 : Resend:1
00:45:50.245 : Resend: N1 M104 T0 S0 *0
00:45:50.246 : N2 M140 S0 *71
00:45:50.248 : N3 G1 X-33 Y-12 F4800 *99
00:45:50.249 : N4 M84 *59
00:45:50.616 : N1 M110 *2
00:45:50.617 : N1 M110 *2
00:45:50.617 : N2 M115 *4
00:45:50.618 : N3 M111 S6 *68
00:45:50.640 : FIRMWARE_NAME:Repetier_0.92 FIRMWARE_URL:https://github.com/luc-github/Repetier-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:DaVinci EXTRUDER_COUNT:1 REPETIER_PROTOCOL:2

Not sure if this has anything to do with the issue but that last G1 command (at 00:45:50.248) is also not being sent by me, is it part of the M50 actions? In any case it is not having any effect, as the print head doesn't move at all after M50 (in fact the steppers become disabled).

luc-github, 2015-02-04:
M50 do not send any GCODE - it may send by your Kill print script ?
the disabled stepper is done by repetier (with and without M50) when doing kill print or just when printing and doing M50
seems need printing to generate this - did not have digged more yet
I think manually moving the head then use move command without homing first is the root cause - but I do not get why I cannot reproduce if you have 100% failure on your side

oscahie, 2015-02-04:
Ah, I think I know what triggers it. Check your printer settings, in Repetier. See if you have the option 'Go to Park Position after Job/Kill' enabled. I only tested twice but couldn't reproduce the issue anymore with that disabled. That seems to be what was sending the G1 X-33 Y-12 command.

luc-github, 2015-02-04:
No mine is not checked - I will test with this checked now

luc-github, 2015-02-04:
Yes that was the missing part - now I can reproduce it 
so I guess I the command is not properly purged and bring this side effect -  I can reproduce so I should be able to handle it
Thanks

luc-github, 2015-02-04:
this should solve the issue : https://github.com/luc-github/Repetier-Firmware-0.92/commit/c78976ed65f985c9042b48762cb31f80a0734395

oscahie, 2015-02-04:
It seems to fix it, indeed. Thank you! :+1:

oscahie, 2015-02-04:
I noticed though that the steppers are no longer disabled after killing the print, unless I do it manually. Is this a side-effect of the fix?

luc-github, 2015-02-05:
Yes because I purge all GCODE coming the M84 is ignored like the others send by host
you means after a stop should be better to disable steppers automaticaly ?

oscahie, 2015-02-05:
Not automatically, but I have RH configured to disable steppers after a job kill and now it doesn't do it, so I have to do it manually. It's not a big deal though.

luc-github, 2015-02-05:
automaticaly, I mean M50 command 
this command stop all immediatly
switch off all heaters (bed/extruders)

it is just one line to add to disable all steppers if it make sense to do 
I just never think about it - just let me know

oscahie, 2015-02-05:
Ideally that should be configurable in the host app, I think. But if not possible or difficult then I think it'd make sense to add it to the firmware. This way if one wants to move the print head after killing the print you may either move it manually or use the host/printer jogging controls, which would engage the steppers again automatically anyway.

luc-github, 2015-02-05:
What about something like M50 S0 will disable stepper and M50 alone will not disable steppers ?

oscahie, 2015-02-05:
Yeah, that'd work too.

luc-github, 2015-02-06:
Implemented : https://github.com/luc-github/Repetier-Firmware-0.92/commit/7df09ea1548faac7b7abb2e778d92b52b7246870

Thanks for your help

closed 2015-02-06 by luc-github

luc-github, 2015-02-26:
Repetier has a command string when sending stop command, that I did not used until now, but I think it is better than use parameters for M50 command, because it affect printer stop menu as well when M50 parameters do not, I also think going back to home position is good when stopping to avoid nozzle to stick on printed model, so I do homing and disable stepper in stop command
I will remove the M50 parameter S0 because it is overiden by the command line that is present in configuration.h
/** The gcodes in this line get executed, when you stop a sd print befor it was ended.
Separate commands by \n */
#define SD_RUN_ON_STOP "G28 XY\nM84"

oscahie, 2015-02-26:
Fine by me :+1:

oscahie, 2015-03-05:
A minor bug: after the new stop command the extruder and steppers are turned off, but the fan keeps running indefinitely, even though the host reflects the fan status as off.

luc-github, 2015-03-06:
which fan ? the extruder's one or the repurposed one ?
if not repurposed fan is not handle by command but by temperature : On if extruder over 50 degres, Off if extruder is under 50 degres, the repetier command never commanded this fan so far

if repurposed I need to check

oscahie, 2015-03-06:
The repurposed fan.

luc-github, 2015-03-06:
The repurposed fan is controled by M107/M106 only - so I am not sure it give a status to host if it is running or not
so adding M107 to
#define SD_RUN_ON_STOP "G28 XY\nM84\nM107" should stop  it