luc-github/Repetier-Firmware-4-Davinci #42: Loading left filament on 2.0

Zyphuris55, 2015-02-09:
When loading new filament into the left extruder while the unit is at home position, the unit stays at the home position. This causes the new filament to dump out between the print bed and the right extruder's dripbox.
Luckily there is a straight opening from the left head to the base of the printer, but it would be more convenient for the old filament to go into the proper location.

luc-github, 2015-02-09:
good idea - I never use the dripboxes - I keep issue open

luc-github, 2015-02-17:
I just reviewed code, there is currently no movement done to move the extruder in any position 
dripbox  1 neither dripbox 2

So it means,
1 -  need to check homing is done, 
2 - do homing if not done
3 - move to  right dripbox if 1.0 or extruder 2 on 2.0, move to left dripbox if 2.0 and extruder 1

Apply this to load and unload as when heating for unloading sometimes filament melt and fall a little

Any comment or suggestion about this ?

Zyphuris55, 2015-02-22:
Yes, I think that sounds ok. I don't think the left drip box location has a sensor like the right (home) one, but full movement to the left (199 mm) should work the same.
Also, the filament chip readers on the printer (v1 and 2v), they are disabled in this firmware, right?

luc-github, 2015-02-22:
There are no max sensor, so just go to max xmin + x length will go to the good position for left drip box
and the EEPROM filament reader pins are not used on repetier as it is useless, so then can be used for some HW modifications, you can add max position sensors on X and Y for example on 2.0

luc-github, 2015-02-26:
implemented : https://github.com/luc-github/Repetier-Firmware-0.92/commit/2d9fef9b5b4aec9ea2d2565ab51188c058adb76c

closed 2015-02-26 by luc-github