luc-github/Repetier-Firmware-4-Davinci #43: decompressing compressed gcode in firmware.

tecan, 2015-02-15:
decompress archived gcode file on sd card.  otherwise sd uploads take forever and everyone would need a transend wifi sd card, or the swap method is required to bypass the slow spi data rates.

luc-github, 2015-02-16:
well for this there is nothing I can do from FW point of view.
host (repetier/S3D) do not send a file to SD Card, it sends all commands (GCODE) using repetier binary protocol one by one, and these commands are then stored one by one in binary mode on SD Card.
so ZIP file cannot be send unless host (repetier/S3D) have a real file protocol transfert
today Flashair or SD Card Extension are the fastest way to copy to SD Card

luc-github, 2015-02-17:
if no more comment 

i close issue

closed 2015-02-17 by luc-github