luc-github/Repetier-Firmware-4-Davinci #46: compile fails but only when EEPROM_MODE 0

TMLutas, 2015-02-17:
When EEPROM_MODE is set to 1, compile works fine (Mac OS X 10.7.5 Arduino 1.5.8) but change the value to zero and the following errors appear

Arduino: 1.5.8 (Mac OS X), Board: "Arduino Due (Native USB Port)"

In file included from /var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Repetier.h:332:0,
                 from /var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:22:
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeX1 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:33:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeX1,"Manual-probe X1")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeY1 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:34:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeY1,"Manual-probe Y1")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeX2 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:35:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeX2,"Manual-probe X2")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeY2 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:36:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeY2,"Manual-probe Y2")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeX3 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:37:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeX3,"Manual-probe X3")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeY3 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:38:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeY3,"Manual-probe Y3")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeX4 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:39:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeX4,"Manual-probe X4")
 ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/HAL.h:71:48: error: 'const char Com::tManualProbeY4 []' is not a static member of 'class Com'
 #define FSTRINGVALUE(var,value) const char var[] PROGMEM = value;
                                                ^
/var/folders/fj/_3y948796_b3ms2wjsk0kzp40000gn/T/build2022692941532461976.tmp/Communication.cpp:40:1: note: in expansion of macro 'FSTRINGVALUE'
 FSTRINGVALUE(Com::tManualProbeY4,"Manual-probe Y4")
 ^
Error compiling.

luc-github, 2015-02-17:
I do not test anything out of normal usage - like no EEPROM usage at all
I will check later

luc-github, 2015-02-17:
I moved the faulty lines to avoid error 
https://github.com/luc-github/Repetier-Firmware-0.92/commit/d4ba5db54206cdd94274afd830228209f6aa1851

closed 2015-02-17 by luc-github