luc-github/Repetier-Firmware-4-Davinci #50: Davinci 1.0 Crashing / Resseting Mid Print

abwatson, 2015-02-21:
I have a Davinci 1.0 that worked 100% fine on Repetier .91. After upgrading to .92 I am having major issues with the printer crashing mid job. The LCD back light turns off then the printer immediately restarts and disconnects from my computer with no possibility of resuming. Does not happen after specific layers but after about 20 minutes to 45 minutes into the print. Cannot get a full successful print now. Ideas?

luc-github, 2015-02-21:
do you have more details ?
arduino version
variant.cpp version
host or sdcard printing
host version
if host any error in log monitor
when did you download the 0.92

it would help to understand where issue come from

abwatson, 2015-02-21:
Yes, my apologies.

Arduino 1.6.0 with the latest variant.cpp posted in download for 1.6.0 (updated to .92 last Sunday 15th). I am host printing with Repetier 1.0.6.
No errors are present in log - Repetier sends coordinates, printer works for a strangely inconsistent amount of time and suddenly the printer restarts and says "printer ready" on LCD. 

10 minutes into a 2 hour print the printer restarts. I was able to resume with the unused gcode and start where i left off after heating back up the components. Roughly an hour later it happened again. Then about 30 minutes later, and then one final time before the print finished about 10 minutes later.
I re flashed 0.92 last night and was able to print a 30 minute, 33 layer print successfully. I am starting a larger print now, will know shortly if problem persists.

luc-github, 2015-02-22:
Ok looks like USB connection issue
with 1.6.0 I did not add the usbcore.cpp, so if usb connection happen, printer freeze then watchdog restart it, usb connection issue come from cable or computer go to sleep or PC is too busy for a while an lock usb connection during a command.

If you did not change anything from cable / computer point of view, let check what can be done from FW point of view.
1 - stay with arduino 1.6.0 but replace the usbcore.cpp with the one in my github for the 1.5.8 they are identical between 1.6.0 and 1.5.8 so patch will work

if still have issue
2 - move back to arduino 1.5.8 and change variant.cpp/usbcore.cpp, Arduino 1.6.0 changed several things for serial/usb connection and it is pretty new - may be more changes are necessary but as I did not experienced any issue, I cannot guess if it is the root cause and where it come from....

if still have issue
3 - try SD Card printing 
this will allow to see if it is definitly USB issue or not

Until now I did several prints with 1.6.0 around 1h30min each and did not got any problem

abwatson, 2015-02-22:
I have now changed the usbcore.cpp for the one in your github folder and re-flashed. I am now starting a new print, I'll reply when something (good or bad) happens. Thanks for your help so far.

luc-github, 2015-02-22:
Ok let me know, I do not use USB connection since a while as no space in room close to printer, so I do not see all problems, thanks for your help

abwatson, 2015-02-22:
No problem, I'm glad i can contribute to the community.
I am currently an hour and a half into a 10 hour print with no issues thus far after adding that usbcore.cpp file to arduiono. It's looking good so far, lets hope that's the fix for it. 
Will update again soon.

abwatson, 2015-02-23:
Okay, **after reverting back to Arduino 1.5.8 and using both update files (variant.cpp/usbcore.cpp) I am no longer having USB or communication problems between printer and computer.** It seems to be functioning just as good as it was on .91 with the added bonuses of .92. 
Thank you very much for your help.
-Adam

closed 2015-02-23 by abwatson

luc-github, 2015-02-23:
thanks for sharing, seems 1.6.0 is not fully compatible with the fw from usb point of view

luc-github, 2015-02-23:
I have created an issue in the tracker to follow up
thanks

abwatson, 2015-02-23:
Great, glad to be of assistance, should I reopen this issue? I assumed "close" meant me personally but did not think about the whole issue being not solved.

luc-github, 2015-02-23:
no no it is correct, your issue it is solved in some way
I created issue https://github.com/luc-github/Repetier-Firmware-0.92/issues/51
for general following on this topic, and keep collecting information, I was just sharing I wanted to keep following issue
thanks for your concern

abwatson, 2015-02-23:
Very well, Good luck in the future, and have a nice day.
-Adam

jsnbrgg, 2015-03-29:
Hello..  I'm having the same issue with the printer restarting mid-print.  I'm running version 0.92 (downloaded again over the weekend to make sure I had the most recent version) and arduino 1.5.8.  I cannot print from USB at all, it restarts every couple minutes even when I'm not printing so long as I have the USB connected.  But when printing from the SD card it also reboots, it just waits much longer until it's in the middle of a 10 hour print.

Any suggestions would be appreciated?

luc-github, 2015-03-29:
Looks like USB/Serial is blocked so this trigger the watchdog and reset printer

did you replaced the USBCore.cpp ?

jsnbrgg, 2015-03-29:
Yes, I used the most recent version that you have in your download.  But it
even does this when no USB is connected and  it's printing from the SD card?

Thanks -J

luc-github, 2015-03-29:
did you do any hardware modification ?
it was working before or never worked ?
You may try to disable watchdog to see what is wrong, to see if it is watchdog which reset or a short on some wires

jsnbrgg, 2015-03-29:
I have an E3Dv6 hot end installed but it's always done this, even before
the hardware mod.  Short prints are OK usually but anything longer than 4
hours will randomly reset.  When it resets I can see the power cycling and
the printer comes back up.  I've checked and double checked all the wires
and can't find any sorts.

Last time I downloaded the firmware was Friday night.  Did you make any
updates since then?  Maybe I need to re-download again?

I turned off watchdog this morning and testing right now.

Thanks -J

luc-github, 2015-03-29:
Yes just fixed an issue with G11.

Sorry you mentioned:
__it restarts every couple minutes even when I'm not printing so long as I have the USB connected.__
It start from power up or any action trigger this ?

jsnbrgg, 2015-03-29:
The USB resetting doesn't start until I connect Repetier Host

luc-github, 2015-03-29:
may  I know the settings you use in repetier host ?
what the log of repetier said ? any waiting command ?
So you never been able to use repetier host to print, it that correct ?
it is the same when you open a serial terminal in arduino IDE ?

jsnbrgg, 2015-03-29:
Sure!  Give me a few minutes, I'm going to download the newest version of
firmware and load it.  Then I'll send some screen shots.

Thanks -J

luc-github, 2015-03-29:
new fw need M502/M500 as it has new EEPROM positions

luc-github, 2015-03-29:
ok nothing obvious, temperature of 250, looks high but not related to usb.
what about my others questions ?

jsnbrgg, 2015-03-29:
I was last using the SD card, so I didn't have the host open.  I'll run it
this time and watch the log.

Correct, I've never been able to print reliably from the host.

I haven't tried it through the arduino IDE serial terminal.

Have the new firmware loaded on the printer now and I'm going to test it
out.  I will let you know what happens.

Thanks -J

jsnbrgg, 2015-03-29:
Here are screen shots of the changes I made to the firmware configuration.h as I'm getting ready to load it now:

[image]

[image]

[image]

Thanks -J

jsnbrgg, 2015-03-29:
The cheap filament I bought on eBay likes the temp at 250, lol

luc-github, 2015-03-29:
to disable watchdog it is in variant cpp, uncomment disablewatch dog
to disable decouple test there is a flag in configuration.h , but decouple test do not reset printer just stop all

jsnbrgg, 2015-03-29:
Ok.  I'm going to try it first with the newest firmware that I just
downloaded to see if that fixed my problem before I disable watch dog.

Thanks -J

jsnbrgg, 2015-03-29:
Just started a print form the SD card.  I have the arduino serial running to see if I can catch the error if it fails.

Thanks -J

jsnbrgg, 2015-03-29:
Well, it ran a couple hours and then restarted right in the middle of the print.  This was printing from a gcode file on the SD card, but this time I was monitoring the serial on the arduino.  I think it restarted on the TargetBed:100 line.  All the reset of the commands where just "wait".  Here's a screen shot:

[image]

Here is the latter part of the file:

[image]

So if I disable watchdog in the variant cpp file that wouldn't help right?  Since I'm not printing for the USB?  Or does it still look for that USB connection?

Thanks -J

jsnbrgg, 2015-03-29:
The variant cpp file wasn't in the repetier sketch folder.  Should it have been?  I copied it over to that folder so it would open in the sketch tabs and I could get to it to edit it.

Thanks -J

jsnbrgg, 2015-03-29:
I'm also going to give my bed a little more time for temp adjustments by changing this value; or at least I think that's what this does?

[image]

jsnbrgg, 2015-03-29:
So, apparently I don't know how to disable watchdog because I'm getting all sorts of compile errors.  Can you please tell me the steps to disable it?

Thanks -J

jsnbrgg, 2015-03-29:
Ok.. I have disabled Watchdog and now I'm starting the same print again except this time I'm running it from the host where I can watch the log.  It is a 5 hour print and the last four times I've tried it's reset about 2.5 to 3 hours into the print.  I've wasted a ton of filament on this, but I'm determined to figure out what the problem is before I start my 48 hour prints!

Thanks -J

luc-github, 2015-03-29:
variant.cpp is in arduino directory in your system directory like usbcore.cpp they already exist, just replace the existing
to debug your issue it is better no to use filament - you will save time ( cleaning) and filament

luc-github, 2015-03-29:
I see you are changing decoupled test values - why ? it is not related to watchdog , do you have decoupled error also ?

jsnbrgg, 2015-03-29:
No, I was just grasping at straws.  Thought maybe the decoupling might be
causing the restarts until you mentioned earlier that the decoupling
doesn't do restarts.

It's been running for a while now from the host on USB with the watchdog
disabled.  Longer really than it's ever ran from USB since I changed the
firmware.  One thing that I'm noticing a lot is that it randomly stops like
it's buffering every few seconds, then continues.

Yes, if this print dies then I'm going to remove the filament until I get
the bugs worked out.

Thanks -J

luc-github, 2015-03-30:
Strange - as others 1.0A seems not having this issue - are you using PC / linux / Mac ?

jsnbrgg, 2015-03-30:
I'm using a Desktop PC with Windows 7 and I also tried my Microsoft Surface
Pro 3 with Windows 8 and had the same issues with both.

Thanks -J

jsnbrgg, 2015-03-30:
Maybe I just got a lemon.  I've even been running it through my UPS to make
sure it's getting good clean power.

luc-github, 2015-03-30:
well Davinci is very fragile, easy to get broken in a part when you fix another one, and easy to get fry

jsnbrgg, 2015-03-30:
It's still running with no problems yet on USB with the watchdog turned
off.  So I'm wondering what are the hazards of leaving watchdog off to keep
my printer running?  Will the decouple still catch a problem while the
watchdog is off?

luc-github, 2015-03-30:
yes decoupled test is working even watchdog is off that is why there a flag also for decoupled test to be disabled if needed

The watchdog is a protection for for out of control issue 
you may try to reenable watch dog but change the timer value to 16s
in HAL.H
// INTERVAL / (32Khz/128)  = seconds
#define WATCHDOG_INTERVAL       1024u  // 8sec  (~16 seconds max)

change to #define WATCHDOG_INTERVAL       2048u  // 16sec

to see if it help

jsnbrgg, 2015-03-30:
Update:  It printed all night and completed the print from USB.  So my issue must be in the watchdog.  I'll try your watchdog settings later today and see how it goes.

Thanks! -J

luc-github, 2015-03-30:
well if watchdog is triggered - it means there is a some problem - your board may have an issue so watchout

jsnbrgg, 2015-03-30:
Thanks!  That is what I was thinking too.  I plan to only run it when I'm
home and can keep an eye on it while I have watchdog disabled.  Then later
I'll order an arduino and ramps to replace it, or maybe a factory Da Vinci
motherboard if I can find one.

-J

jsnbrgg, 2015-03-30:
I've had a ton of problems with this thing.  If I lay a straight edge
across the bed, even the bed is slightly bowed so i always try to print in
the middle and the reason I changed the hotend was to try and solve the
issue with the random restarts.  But turns out I really like the E3D!  It's
less than six months old and had these problems straight out the box.
Didn't want to send it back though because I live in such a remote area
it's a major pain to ship something this large.  Plus I think I voided the
warrenty, lol

jsnbrgg, 2015-04-15:
Just wanted to leave an update.  I've had the Watchdog disabled for about 16 days now and haven't had any more printer malfunctions.  I have not tried changing the Watchdog interval yet because it's been working so well.

Thanks

luc-github, 2015-04-15:
good to read it is fine, happy printing :smile:

jsnbrgg, 2015-04-15:
What did the update you made yesterday fix?  Was it related to this issue?

Thanks

luc-github, 2015-04-15:
nope was a work around for people having issue with home sensor and that may make some shift during print - the issue is hardware but this can help to decrease the problem

nobbynosecomb, 2015-06-17:
Hi I seem to be having a similar problem to jsnbrgg I just recently installed 0.92(about a week ago) and followed all the instruction changing the usb core file and variant ,all up loaded fine to my 1.0a I am using repetier host and slic3r all those settings seem right because what it does print looks great, my problem is the print will just stop randomly, show idle on the lcd and often say steppers disabled, the extruder and bed are still at full temp, sometimes I lose connection with the host as the bed is heating,  but regardless of when it happens there is absolutely no connection between the host and printer. I just tried to print a sample from the sd card option found in the printer lcd menu, it caused the printer to immediately try to start printing with a cold extruder and bed. Should I disable watchdog? im really not sure how to do that any help would much appreciated.

luc-github, 2015-06-17:
you must not print sample file from XYZ using repetier it is not good as temperature are not in file that is why it start to print with cold extruder

if you loose connection when heating it means your repetier host settings are not corrects  (baud rate / ping pong...) or your usb cable/connector have issue.

you can disable watchdog but I think it won't help as watchdog reset the printer, and what you describe is not a reset of printer, but usb communication issue because if printer reset, temperatures will  be off.

As mentioned earlier in this  issue to disable watchdog you must edit the variant.cpp previously modified and   comment ```WDT_Disable(WDT);``` like this   ```//WDT_Disable(WDT);```

nobbynosecomb, 2015-06-17:
Ok thank you for your quick reply I will double check those things and see how it goes ill leave watchdog alone

nobbynosecomb, 2015-06-17:
ok I left my printer turned on and connected to the host with all manual commands working fine, and no printing activated.

 when I returned after 30 mins it had lost connection and I couldn't get it back untill i pulled the usb cable out ,and plugged it back in  I have the following settings cache 202 baud rate is 230400 on printer and in host ping pong is ticked

luc-github, 2015-06-17:
what about others settings of repetier host ?

nobbynosecomb, 2015-06-17:
[image]

nobbynosecomb, 2015-06-17:
let me know if you need any other settings I appreciate your help

luc-github, 2015-06-17:
I see nothing wrong in setting so should be something else

is computer is going to standby ? this will shutdown connection and host do not like

nobbynosecomb, 2015-06-17:
I installed repetier host and arduino on my laptop then copied all the same settings out of repetier and slicer,  and it printed with out any problems, maybe its a fluke, will try another print tomorrow,

 but if this continues to work from my laptop I guess there's a problem with my desktop considering there was little to no difference in the software and settings perhaps its a hardware or some windows settings on my desktop, any ideas?

[34 more replies]