luc-github/Repetier-Firmware-4-Davinci #52: Hotend Cleaning before printing.

tecan, 2015-02-27:
it does not know where the cleaning fins are on my davinci 1.0 it goes around them before printing. really annoying when trying to print PLA.

luc-github, 2015-02-27:
Hmm can you elaborate the issue ? I do not  understand

tecan, 2015-02-27:
a quick wipe of the tip across the rubber foam thing or one of the metal fins on the waste basket even if you bend it down to just under the tip of the hotend so nothings hanging down before a print, sometimes it helps to have the trail to start othertimes it makes a mess.  also would the cartridge io P0 and P1 be able to be repurposed into pwm fans for cooling or a extra thermistor for the cooling block ?

luc-github, 2015-02-27:
do you mean M100 command is not working properly ?

tecan, 2015-02-28:
its working my mistake whats the best way to issue that before each print. cura or printrun ?

closed 2015-02-28 by tecan

luc-github, 2015-02-28:
what ever slicer you use, it has a start GCODE customization setting, so need to be there, and better is after heating just before first printing