luc-github/Repetier-Firmware-4-Davinci #54: 2/26 Dav1.0 power cycles every 18-19 seconds.

Nepenthy, 2015-02-28:
Sorry I cant be more specific. I was @ 2/20 and today went to 2/26 on a Dav1.0. After successful flash machine power cycles every 18-19 seconds. Please let me know if theres anything I can do to help narrow something down.

luc-github, 2015-02-28:
no idea what is wrong as am doing a test print right now and already printing  4 hour
be sure no file is corrupted and reflash
if still issue go back to previous commit download the corresponding set and flash it , and do it until no more issue, normally the previous one you used
another test is to disable watchdog in variant.cpp and configuration.h to avoid restart and see what is wrong

luc-github, 2015-02-28:
just checked last commit was to fix an issue for 1.0 and was validated on 1.0 so should be ok, unless you do some change in configuration.h that I am not aware

luc-github, 2015-02-28:
not sure you know how to select a commit , here the way
to go to previous commit on github.com
click latest commit
[image]
then click on parent id
[image]

then browse files
[image]

then you can see the commit id and can download the corresponding sources by clicking on download zip

[image]

you can navigate from commit to commit like this

luc-github, 2015-02-28:
sorry it is almost 11pm here now so will check your feedback tomorrow

Nepenthy, 2015-02-28:
This was my stupid mistake so im going to remove it. I had noticed that I still saw the heatbed init in the config, after reading it was removed, but didnt pay much attention. Going back I see I had edited and flashed the wrong version:
#define VERSION_MINOR_YEAR "14"
#define VERSION_MINOR_MONTH "12"
#define VERSION_MINOR_DAY "12"
#define VERSION_BUILD "1"

#define WARMUP_BED_ON_INIT 0 

My apologies for false alarm.

luc-github, 2015-02-28:
ok good news so, it is scary issue

closed 2015-02-28 by luc-github