luc-github/Repetier-Firmware-4-Davinci #55: Filament Sensor do not pause printer if printing from host

luc-github, 2015-03-05:
MIssing command to pause the host if out of filament

luc-github, 2015-03-05:
Curently only Pause if printing from SD but todaygot command to pause it printing from Host
if send RequestPause: host will pause, it is in FW but never saw, it is not available in menu just in code,  and it is not documented...

Doudou14, 2015-03-20:
I noticed this bug, it detects the lack of wire only if we put the printer manually paused

luc-github, 2015-03-20:
Actually detection is fine - but pause has no effect when printing from host
should be fixed by https://github.com/luc-github/Repetier-Firmware-0.92/commit/eef0b6355fa602a43a917a14afb03d9737bf7b54

closed 2015-03-20 by luc-github

Doudou14, 2016-05-15:
hi luc
I restarted my DaVinci and I noticed that if there is no filament during printing, the printer asks to load a new filament but bug. The bed down and then up but does not charge the filament and printing failed.

luc-github, 2016-05-15:
I did not change anything for ages on this parrt

what is your fw version? what branch?
are you printing from host or from SDCard?
if host what version?

Doudou14, 2016-05-15:
My FW version is 0.92.6 (09-12-15)
Printing from repetier host V1.6.1

luc-github, 2016-05-15:
Will have a look - 
But normally if from host it send 
RequestPause: No Filament - which is supposed to pause host , then wait for  no movement then bed down
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/ui.cpp#L4440-L4445

May be new host do not care the pause command anymore - I saw in latest fw (0.92.9) there is a new way to stop host if error or need action

luc-github, 2016-05-15:
the 0.92.9 (originaly 0.92.6) is still in alpha I move to new issue to track and not confuse with 0.92.3 (stable version)

luc-github, 2016-05-15:
moved here https://github.com/luc-github/Repetier-Firmware-0.92/issues/157