luc-github/Repetier-Firmware-4-Davinci #62: Current Branch dont work for me

gamecompiler, 2015-03-27:
Hi, the f09229b7d002e9fee80b418cb810fd958bb2810e Branch dont work for me.

Id succesfully installed the firmware on my DaVinci 1.0 using Arduino IDE 1.6.1.
And id restored the EEPROM to Failsafe.

When i start a print, the extruder head moves only to the first point of the first path.
And then the printer just do nothing.... He only heats the extruder and bed like he should do.
Id tryed to print a gcode file over the SD-card and over PC using Repetier-Host 1.0.6 for Windows.

Also another part of my questions: Did you succesfully print PLA ? If yes how did you do that ?

Thanks in advance.
Your firmware makes my Printer more awesome as before Oo !

gamecompiler, 2015-03-27:
Addition information:
Id saw that your firmware is based on bgm370´s firmware.
With his firmware i dont have those problems while printing.

luc-github, 2015-03-28:
Ok it means you have 1.0 if you can use BGM FW.

Do not use 1.6.1 ide there are some issues since 1.6.0 : https://github.com/luc-github/Repetier-Firmware-0.92/issues/51.
Please use 1.5.8 arduino IDE to build FW (http://arduino.cc/en/Main/OldSoftwareReleases) , you need to update the variant.cpp and usbcore.cpp.

About PLA http://voltivo.com/forum/davinci-peersupport/399-da-vinci-and-pla, I just tried do not really use as I have another printer which print PLA 1000 times better, but not good at ABS, so I use DaVinci for ABS only.

gamecompiler, 2015-03-28:
Thanks for your reply.Id used IDE 1.5.8 now, but ive got the same Problem as id have before. Could you send me a bin or a hex file please ? Then i could use bossac.Von meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [Repetier-Firmware-0.92] Current Branch dont work for me  (#62)Von: Luc An: "luc-github/Repetier-Firmware-0.92" Cc: Jason Ok it means you have 1.0 if you can use BGM FW.

Do not use 1.6.1 ide there are some issues since 1.6.0 : #51.
Please use 1.5.8 arduino IDE to build FW (http://arduino.cc/en/Main/OldSoftwareReleases) , you need to update the variant.cpp and usbcore.cpp.

About PLA http://voltivo.com/forum/davinci-peersupport/399-da-vinci-and-pla, I just tried do not really use as I have another printer which print PLA 1000 times better, but not good at ABS, so I use DaVinci for ABS only.

luc-github, 2015-03-28:
it is strange: it works on my side
I have some questions : 
So you mean when temperatures are reached, extruder move to bed then it does not move ?
Is menu on printer still working ?
Any log from repetier host when this happen ?
Do you use any custom start code ?
Did you change anything in FW before compiling ?

I do not have PC to compile right now - but you may want to try the 0.91, binaries are available
https://github.com/luc-github/Repetier-Firmware/wiki, RC is the latest
it is in between BGM and 0.92 for the features

gamecompiler, 2015-03-28:
Yes when the temperatures are reached, the extruder only moves to bed then it does not move anymore. The Menu still works, but crashes if i select Stop print while printing over SD-Card.
I dont use costum start code, i use the default start code provided by the current Repetier-Host.
I did not change the firmware, id used your current branch without any modifications.

Here is the first part of my gcode id used printing over PC and also over SD-card.
http://pastebin.com/w4ih3th5
This code works great with bcm´s firmware.

In Repetier appears no error, but i will try to find some....(Sorry for my english).
I will try 0.91 on tummorow, could be interesting.
I will make a video how i interact with my printer with 0.92 firmware, maybe it help.

Thank you very much for your help !

luc-github, 2015-03-28:
I never used G10/G11 it may worth to see if you still have the issue when you remove them, it is midnight here so won t check now, but could be a root cause

luc-github, 2015-03-29:
OK I can reproduce the issue - seems linked to G10/G11 just imported commands from 0.92.3, I change the title to reflect the problem
I will see what is wrong as I just imported the function and never used them

Thanks for sharing

luc-github, 2015-03-29:
I found the root cause - it is an EEPROM issue that erase the speed of unretract - so everything is stopped because speed is 0, you can use the latest commit https://github.com/luc-github/Repetier-Firmware-0.92/commit/03e840c20bf76a48ac9fb2c2fd71936ff98b2899, issue is fixed

I will report to Repetier as issue come from original FW, thanks for your time and your report

closed 2015-03-29 by luc-github