luc-github/Repetier-Firmware-4-Davinci #63: DaVinci 1.0 restarting after connecting to Repetier

DerianF, 2015-03-29:
Hello i have the daVinci 1.0 printer that came stock with firmware 1.1G. i have been able to use the command prompt to change the firmware before but now it doesn't upload anymore(it says it does)i took it off the Repetier before because it was disconnecting mid print but it was just a bad USB cable. I am able to upload Repetier using the Arduino 1.5.8 IDE but when i try to connect it to the printer the lights go dim then the printer resets. If try to connect again it cant find the comm port and will oly find it if i turn the printer off and on again.

luc-github, 2015-03-29:
what FW did you used ?

DerianF, 2015-03-30:
I went back and forth with .92 and stock da vinci FW
and i cant cupload neither anymore through command prompt.
i tried using usb 2 and 3

luc-github, 2015-03-30:
USB 3.0 are not working well with arduino do not use them
did you reset using the jumper ?

DerianF, 2015-03-30:
Yea, mine isn't one one of the new printers so it flashes with jumper1.

luc-github, 2015-03-30:
well if flashing is not working - means the motherboard has issue 
http://voltivo.com/forum/davinci-hwmods/29-i-bricked-my-da-vinci-1-0-3d-printer-upgrading-firmware?start=20#6383

DerianF, 2015-03-30:
Yea i had a feeling, i have some experience with electronics, do you think i will be able to replace the board and some parts and use the frame for a DIY kit?
here is a link to the one i want to buy.
http://www.amazon.com/SainSmart-Mega2560-LCD12864-Printer-Arduino/dp/B00L12W4SC/ref=sr_1_1?ie=UTF8&qid=1427724728&sr=8-1&keywords=3d+printer+kit

luc-github, 2015-03-30:
I do not think you need the full package
but mega/ramps/drivers (and EndStops?) screen also as buzzer and SDCard are on Davinci board 
but no need pulley/stepper/belts IMHO

DerianF, 2015-03-30:
They have this package aswell.
http://www.amazon.com/SainSmart-Mega2560-Endstop-Printer-Arduino/dp/B00F895XO6/ref=sr_1_16?ie=UTF8&qid=1427726551&sr=8-16&keywords=sainsmart+3d+printer
In your honest opinion, should i try to use the da vinci for a frame and build a new printer or should i just use the full package and get a reprap frame?

luc-github, 2015-03-30:
Well is really up to you, you can start using frame of Davinci it is not so bad and have enclosure - which is very good for printing ABS
That said next step will be to change the extuder and the carriage for sure.

I builded a i3Prusa in Kit was nice but ugly, mechanical parts are not so clean as Davinci for some point but better from another part (bearing) and depending of the kit you can also desapointed by the quality.

I really think even you do some modification later the Davinci frame is good base

luc-github, 2015-03-30:
Also be careful with electronic, bt I am sure you now that, - I bought 2 Mega on local market and after 4 flash the board are not allowing any flash, they were very cheap as they were clone I guess, but at the end useless

DerianF, 2015-03-30:
I will go ahead and buy the full kit just in case and keep what i don't need for parts. do you have any recommendations on the extruder to use or anywhere i can buy a 3d printed mount for the one that comes with the kit?

luc-github, 2015-03-30:
it depend what you plan to print actually - everyone says E3D V6 is very good but I did not tried yet , and bowden or direct drive is also question of choice, for flex filament direct drive is the best but for others both can be used

luc-github, 2015-03-31:
I close the bug as it is hardware issue and cannot do anything

closed 2015-03-31 by luc-github

DerianF, 2015-03-31:
Thank you very much for the help.