luc-github/Repetier-Firmware-4-Davinci #64: Second fan support for 1.0 printers with added hardware

kiyoukan, 2015-03-31:
It is possible to add the hardware (npn transistor, 1k resistor, 0/jump resistor, cap) to enable use of a second fan on the 1.0 printers. But in the firmware there is no clear way on how to enable both fans.
I was able to test it via pin change and enabling the fan control to make sure the additional hardware added to the board works. It did.
But i see no way to tell the printer to turn on the stock fan and allow the newly added fan to be software controlled. i attempted to set it as a 2.0 printer but then the system gets and extruder error as it cant detect the second head.
And i dont want to add a second psu to force turn on the stock fan if i can just let the printer do it.

luc-github, 2015-03-31:
If you want to control 2 fans you need 2 pins, if you have a free pin you
can use it  or you use one fan for 2 extruders - my duo has only 1 fan for
2 extruder

you said you do a 2.0 with a 1.0 ? if you populate all missing components :
resistors / mosfets/capacitors/connectors/etc... you should have the pin
for the fan on connector

you also can connect fan directly to 5v/or 12V on board everyone using the
repurpose function to have 2 fan for extruder 1 did this

I am not sure to get your question/problem

kiyoukan, 2015-03-31:
I have a 1.0 printer and populated the missing fan control hardware.
But i see no way to set the firmware to tell it that the hardware has been added and to cool the print when the printer config is set to 1.0
if the printer config is set to 2.0 it works with both fans one set to turn on and off the other to keep the feed tube cooled but after heating up for a bit the system throws an error that the second extruder is not detected. 

So i was wondering if there is a way to take the fan control options only and not the second extruder from the 2.0 config.

Sry if this is not clear.

kiyoukan, 2015-03-31:
i wanted to avoid having to add a second power supply or pull from the 5v rail as then the fan will always run when my printer is on.
I was hoping that this mod would allow my 1.0 to be able to fully control both fans so i dont have to wrry about turning manually a fan on and off

luc-github, 2015-03-31:
you need to modify in configuration.H
```
    #if REPURPOSE_FAN_TO_COOL_EXTRUSIONS==1
     #define FEATURE_FAN_CONTROL 1
     #define FAN_PIN ORIG_FAN_PIN
       #if DAVINCI>=2
          #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
          #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
        #else //DaVinci 1.0
            #define EXT0_EXTRUDER_COOLER_PIN -1 //Warning need to add a permanent fan with  power supply to cool extruder
         #endif
     #else
       #define FAN_PIN -1
      #define FEATURE_FAN_CONTROL 0
    #endif
```

to something like that

```
    #if REPURPOSE_FAN_TO_COOL_EXTRUSIONS==1
     #define FEATURE_FAN_CONTROL 1
     #define FAN_PIN ORIG_FAN_PIN //first fan
       #if DAVINCI>=2
          #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
          #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
        #else //DaVinci 1.0
            #define EXT0_EXTRUDER_COOLER_PINORIG_FAN2_PIN //second fan
         #endif
     #else
       #define FAN_PIN -1
      #define FEATURE_FAN_CONTROL 0
    #endif
```

kiyoukan, 2015-03-31:
Awesome i was going to post a guide in the hardware mods on voltivo for how to add the parts to enable 1.0 users to use both fans but im not the best with software.
I will test this out tonight once i get home and do a detailed write up with part numbers.

luc-github, 2015-04-02:
Ok so I close the item if it is OK

closed 2015-04-02 by luc-github

kiyoukan, 2015-04-02:
Thanks so much for your help.
Your code was almost perfect i just needed to change the fan numbers as it was reversed. 
Seems the mods on voltivo wont allow me to post a thread on how to add the parts to the board. not to hard didnt use a single smd part just 1/8w 1k resistor a jump wire for the 0 ohm resistor and npn 10-92 transistor. Have not added any caps yet but so far things are fine.