luc-github/Repetier-Firmware-4-Davinci #66: | X Axis is not correct | EEPROM Settings | Repetier Settings |

gamecompiler, 2015-04-05:
Maybe its my fault, im pretty sure.
I think the X Axis has been defined a little bit too long (Sorry for my English).
If i print things my objects are too long in the x axis, is it possible to change the steps per mm for the X- Axis in firmware ?

Did you changed some things in your EEPROM after restoring the EEPROM to Fail-Safe ?
If yes what did you change ?

What did you enter as settings in Repetier, like Retractionspeed e.t.c ?
Do you use Sli3er ?
Please share your settings !

Thanks in advance, after the G10/11 fix the firmware works great now.
(I really like the Emergency Stop feature !)

luc-github, 2015-04-05:
Hi X axis is supposed to be fine for 1.0/2.0 but never had confirmation for 1.0A/2.0A/AiO
What printer model do you have ?

you can change step and size in EEPROM using repetier host to edit it 
I use S3D not slic3r. but there is one thread for this : http://voltivo.com/forum/davinci-firmware/148-your-configuration

if you have 1.0A/2.0A/AiO and find better dimensions for Xmax, Ymax or Zmax feell free to share, so  I can add then in FW  and they will be by default

gamecompiler, 2015-04-05:
Thanks for your reply, i have a DaVinci 1.0. I will share my settings, when id found better values.

luc-github, 2015-04-10:
Any update ? as look like slicer issue / EEPROM settings - should I close issue ?

Thanks

gamecompiler, 2015-04-10:
Yes it was a sli3er issue, i do not have those problems If i use Cura.Please close this issue if i cant (If i can, i will close it tummorow, im writing with my Handy now)Von meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [Repetier-Firmware-0.92] | X Axis is not correct | EEPROM Settings | Repetier Settings | (#66)Von: Luc An: "luc-github/Repetier-Firmware-0.92" Cc: Jason Any update ? as look like slicer issue / EEPROM settings - should I close issue ?

Thanks

closed 2015-04-11 by gamecompiler