luc-github/Repetier-Firmware-4-Davinci #67: Command Waiting

xMistdemonx, 2015-04-06:
always getting Command Waiting and have to hit OK button 9000 times

Da Vinci 1.0a J37

Arduino 1.5.8

Port: COM3
Baud Rate: 230400
Transfer Protocol: Repetier Protocol
Reset on Connect: Due native USB Port
Reset on Emergency: Send emergency command and reconnect
Receive Cach Size: 127
Use Ping-Pong: Unchecked

have tryed most other settings and swaped USB ports / cords

luc-github, 2015-04-06:
Use Ping-Pong: __checked__

if not solving the problem try another PC - it may be port com drivers issue (PC side )
check here:
http://voltivo.com/forum/davinci-software/841-repetier-command-waiting-on-davinci-1-0a

xMistdemonx, 2015-04-06:
checking Use Ping-Pong dose the same, and don't have another PC to use

luc-github, 2015-04-07:
well as said it is windows issue not fw side as it was fixed by using another PC - so I can do nothing

gamecompiler, 2015-04-07:
Did you update the EEPROM to Fail-Safe ?
Did you update the Arduino files (AdditionalArduinoFiles) ?

Did you firewall block Repetier sometimes ? Please try to print from SD-Card, please say me if it had work.

What driver you are using to communicate to the printer (.sys files e.t.c, is shown in the device-manager) ?

I hope i can help you.

xMistdemonx, 2015-04-07:
Did you update the EEPROM to Fail-Safe ?
Yes

Did you update the Arduino files (AdditionalArduinoFiles) ?
Yes

Did your firewall block Repetier sometimes ? Please try to print from SD-Card, please say me if it had work(I had have some problems with my firewall).
Didn't think the firewall would block USB, but anyway i disabled it, no joy
I did a cached print and i was able to send it to the SD card for printing but i still had to mash OK 9000 times (thank god for macro mouse)

What operating system do you use ?
Windows 7 Ultimate 

What driver you are using to communicate to the printer (.sys files e.t.c, is shown in the device-manager if you use windows) ?
Arduino Due (COM10) - drivers that Arduino installed

thanks for the help :)
your English is fine

luc-github, 2015-04-08:
you can try to change the port com to be used - in device manager - select the com 10, go to port settings/advanced and choose another available port
[image]

luc-github, 2015-04-08:
Another thing to try is to lower baud rate on printer and repetier host to 115200 or even 19200 to see if the speed is an issue

xMistdemonx, 2015-04-08:
again thanks for all the help, but sadly non of them fixes work :(

i did how ever portion out some of my PC to Ubuntu and it worked first go heh

luc-github, 2015-04-08:
Yes, as I said this is windows drivers issue - if you ever find the root cause / solution in windows feel free to share.
Happy printing :smile: 
I close issue

closed 2015-04-08 by luc-github

xMistdemonx, 2015-04-14:
not sure if anyone will see this but i think i found the fix to my Windows 7 Communication Problem 

all i did was update to the new Arduino 1.6.3  all seems well for now

i did use 1.5.8 to install the firmware, but today i updated JUST Arduino, all seems well

luc-github, 2015-04-15:
```
xMistdemonx)
not sure if anyone will see this but i think i found the fix to my Windows 7 Communication Problem

all i did was update to the new Arduino 1.6.3 all seems well for now

i did use 1.5.8 to install the firmware, but today i updated JUST Arduino, all seems well
```

I did not check but they may updated the driver