luc-github/Repetier-Firmware-4-Davinci #68: Speed multiply cause complete damaged print.

gamecompiler, 2015-04-07:
Hi, if i change the Speed multiply from 100% to 300% during printing, the complete print will fail because the z Axis dont move correctly, the Z Axis will change not enough down than normal.
That result that the printing head will,....other word for drilling through, but vertical.
In plain text, the object is dismembered.

Id tryed to change the Speed multiply from 100% to 300% during printing three times, every time the same issue, if needed i can upload pictures of the damaged prints.

luc-github, 2015-04-07:
Hi I just updated with one fix from  repetier but I do not think it match your problem
I know by experience that if speed is too high the axis won't turn - I do not see anything in latest changes that can solve the issue. so need to investigate

Just to check what are your max Z travel/printing speed in EEPROM ?

gamecompiler, 2015-04-07:
Give me 10 minutes please, im makeing a test print at this moment for the X Axis Issue.

luc-github, 2015-04-07:
I cannot test right now but you may try to do a simple GCODE file 
starting by a G28 then using your first layer change as a model
and copy the line several times and modify the z position by increasing 0.2 by 0.2
and do it until you reach 10cm, you may use excel to generate it easily

then you can see what happen if z is turning right at 100% 200% 300%

luc-github, 2015-04-07:
can you share the GCODE line your slicer generate for Z position ? just to check the travel speed and reproduce with your settings

gamecompiler, 2015-04-07:
Im not sure which EEPROM settings you want to have, but here are following things.

Z-axis steps per mm 2560.0000
Z-Axis max. feedrate [mm/s} 5.0000
Z-axis homing feedrate [mm/s] 4.000

I will make a test gcode tumorrow.

Thank you for your help.
(And sorry for my English).

luc-github, 2015-04-07:
no problem - my english is basic and I understand you very well :smile:

luc-github, 2015-04-10:
Any update ? I was not able to reproduce issue, can you share the Z changing position line from your GCODE ?

gamecompiler, 2015-04-11:
ID found the Error, its a Repetier Sli3er issue, with Cura (Standalone), and S3D i dont have those problems. I need to find out why this happen. But my vacations are almost over. I will close this thread (If i can). I dont have any time more to solve this.Von meinem Huawei-Mobiltelefon gesendet

luc-github, 2015-04-11:
Ok thanks - I close the issue

closed 2015-04-11 by luc-github