luc-github/Repetier-Firmware-4-Davinci #69: Thermocouple?

jsnbrgg, 2015-04-15:
I bought a new MK8 Extruder / Hotend that came with a Thermocouple installed instead of a Thermistor.  Is there anything special I need to do to get the Thermocouple to work on the Da Vincia 1.0a, or should I plan on switching the Thermocouple out for a Thermistor?  I normally wouldn't ask here but I couldn't find an answer elsewhere on the internet.

Thanks!

luc-github, 2015-04-15:
well thermocouple table is not same as thermistor -  so you may have to tune it - I know there was some issue with some thermocouple with repetier you can have a look here : github.com/repetier/Repetier-Firmware/issues/381

luc-github, 2015-04-15:
I guess you have the specifications or the table of your thermocouple ?

jsnbrgg, 2015-04-15:
I don't yet..  I should probably email the company for that if I plan to try it.

jsnbrgg, 2015-04-15:
After taking a quick glance at the link you provided it appears that folks are trying to use a breakout board for connecting a thermocouple.  I don't have a breakout board yet, so if that's a requirement then I might just swap out to a thermistor.  I have a couple new 104GT-2's

luc-github, 2015-04-28:
Ok so I close issue if no more following

closed 2015-04-28 by luc-github