luc-github/Repetier-Firmware-4-Davinci #70: At what temp does PTFE begin to melt?  (Temp Tables?)

jsnbrgg, 2015-04-29:
While trying different settings to eliminate this massive de-lamination problem I'm having, I've noticed that any time I push my MK8 hotend past 230c the small PTFE tube inside the hotend begins to soften and causes major jams.  I have to replace to PTFE tube to fix it.  So according to Google PTFE doens't melt until over 300c.  So does that mean my temp tables are way off?  I've tried both #5 and #8

luc-github, 2015-04-30:
sorry do you use stock thermistor ? if you use another one and trying other table I guess you are not
if not, using  tables are not accurate because  they are for Ramps not for Davinci board so temperature s can be out of range very easily

jsnbrgg, 2015-04-30:
I'm using the Semitec 104GT-2 Thermistor.

Here is a link to the exact ones that I have:
http://www.amazon.com/gp/product/B00GWA2TZ6/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

luc-github, 2015-04-30:
as said table in repetier FW are not for davinci board and are not valid because for RAMPS - so it is sure the temperatures you see on LCD are not corrects - only the custom table used with stock FW are working with DV board - if you use another sensor you have to redo the table - the custom table has been done drilling a hole in extruder close to sensor and using a thermocouple to build a table : ADC value + measured values

jsnbrgg, 2015-04-30:
Ok..  cool.. that gives me an idea.  Thanks!

luc-github, 2015-05-09:
If nothing more on it I close it

closed 2015-05-09 by luc-github

jsnbrgg, 2015-05-11:
Yes, that's ok.. Thanks

luc-github, 2015-09-06:
adding link for reference
http://voltivo.com/forum/davinci-firmware/438-repetier-91-e3d-v6-extruder#3631