luc-github/Repetier-Firmware-4-Davinci #73: Printer reboot during printing

Doudou14, 2015-05-08:
hi !

By printing a piece of 6H, the printer is stopped and reboot after 1:30 hour. On the log repetier I type commands from the layer 31: Unknown command: Nxxxxx M106
The printer reboot on the layer 43.

Is it due to the watchdog, the power from the printer or other?

Thanks !

luc-github, 2015-05-08:
M106 is fan command that is not implemented unless repurpose fan define is set, it does not generate a reboot
reboot is may done by watchdog but the root cause can be several possibilities

I already printed several times more than 8h without issue

Doudou14, 2015-05-08:
How do I know if this is the watchdog that caused to reboot?

luc-github, 2015-05-08:
you can disable it, uncomment disablewd in variant.cpp

Doudou14, 2015-05-08:
I will try, thank you

luc-github, 2015-05-09:
If nothing more on it I close it

closed 2015-05-09 by luc-github