luc-github/Repetier-Firmware-4-Davinci #74: Printer restarts constantly after 0.92 flash.

mi7ch311, 2015-05-12:
I have flashed my 1.0a fw1.0.6 with repetier v0.92. the flash process works fine. when the printer turns on with the firmware installed it works for about 5 seconds before restarting and will constantly do this. i have tried many times to make sure that i have not missed any steps, i have also followed many different tutorials and none have worked. Any help in troubleshooting this would be greatly appreciated. 

Thanks Mitch

luc-github, 2015-05-12:
last time this happen it was because people downloaded 0.91 and 0.92 and actually flashed 0.91 not 0.92
so be sure of what you flash

reset is supposed to be after 8seconds is it shorter? what does the screen display when starting,?

ultimatly you can disable watchdog by removing comment in variant.cpp for WDT_Disable(WDT); it will help to check  what is wrong

mi7ch311, 2015-05-13:
I am pretty sure that i have only download 0.92 (how can i be sure of this). the display comes up with printer ready then restarts. i have tried disabling watchdog and this seems to have worked, what does this mean? does something need to be changed or can i leave it disabled?

luc-github, 2015-05-13:
When you start printer it display 0.92mod in splash before printer ready ?

mi7ch311, 2015-05-13:
yes this is correct.

luc-github, 2015-05-13:
can you comment again the disable Watch dog and reflash to confirm ?

mi7ch311, 2015-05-13:
I re-commented the watchdog and re-flashed, and again it is now working. thanks this is great. but can you please explain why this might be?

luc-github, 2015-05-13:
I suspect the timer is predefined very short by default and for some reason the previous flash did not enabled properly the time - so reset time is not enough because too short - but it is just a guess - I have also saw some issue disapear after a reflash because first one was not correct but did not raised any issue....

Unfortunately Arduino is not a full proof development platform, it is for teachers/students with several drawback...

Please keep me updated if issue come back - I will think how to reinforce the first timer setting

In all case - disable watchdog is also a way, it remove a safety feature but if you apply the usbcore.cpp patch it should not be a problem, FYI in 0.91 the feature was not present and no issue so far - but it was with previous board generation, 1.0A is a more fragile board as they do a lot of cost down -  so watch out

Hope everything will be fine, happy printing :smile:

pallemar, 2015-05-13:
I solved this by removing the SD card and the chip filament, i guess the real problem was the chip filament, but not sure.

luc-github, 2015-05-13:
unless there is a short the FW do not care the chip filament - it even do not check the pin related to this.
But if you removed the SDCard you have no more EEPROM settings - is that convenient ?

pallemar, 2015-05-13:
It is some time since I had this problem, I think I removed something on the SD card, and then plug it in again, and everything was okay again, some of the old data from the DaVinci firmware. maybe it is the problem?

luc-github, 2015-05-13:
should not, as FW do not care others files than EEPROM.bin which generate by itself.

it is 1.0A or 1.0 ?

pallemar, 2015-05-13:
okay thanks :)

it is a 1.0A

pallemar, 2015-05-13:
BTW, do you have any idea how to add a sensor for Auto Leveling on the Davinci board?
I have got this, for detecting a Aluminium bed http://www.ebay.com/itm/221457637744

luc-github, 2015-05-13:
current one it using probe/plate like end stop to says bed is reached - it is just a simple contact system.
The Davinci pins do not accept more than 3.3V, so you will have to adjust the signal of your sensor to avoid to damage the board as I see it need at least 6V so you will have to power 12V- unless 5V is enough but still too high for Davinci pins

pallemar, 2015-05-13:
Thanks for the help, heh, I should have brought a orange tip sensor. according to this guide, not a blue tip. http://www.instructables.com/id/Enable-Auto-Leveling-for-your-3D-Printer-Marlin-Fi/
but if i follow toms guide here it should work with a blue tip.
https://www.youtube.com/watch?v=EcGFLwj0pnA

Shall we move this auto bed leveling to a new thread?
and thanks again for your help :)

luc-github, 2015-05-13:
Yes please - it is better to keep thread dedicated to one question/issue - thanks

luc-github, 2015-06-04:
I have added a ping to watchdog after watchdog started - not sure it help but worth a try
https://github.com/luc-github/Repetier-Firmware-0.92/commit/d2c10acd1ed245b4d76d378f07aaa79dc935130b

luc-github, 2015-07-10:
seems not working per feedback so I have modified watchdog functions to use the one planned for Due 1.6.5 (github.com/arduino/Arduino/commit/4688a7a23f4665e7ef609cf75be4b7679f5a83f9) , I did not tried with IDE 1.5.8, I just tried with IDE 1.6.5 with module 1.6.4 for DUE but as it use same functions as variant.cpp it should be Ok

no need to edit variant.cpp to disable watchdog, this is now linked to configure.h setting but still need to remove function from variant.cpp like it is already done

I hope using "standard" function from arduino instead of the custom one from repetier will help to solve this random issue.

I just can guess as I never got the issue which seems linked to new generation

implemented here : https://github.com/luc-github/Repetier-Firmware-0.92/commit/5d090030a5fd2230caac2ffded421d8b6e7356f3

luc-github, 2015-08-04:
Did not got any new feedback about this so I guess last update was the good one
I close issue - reopen if needed

closed 2015-08-04 by luc-github

KMC16, 2020-04-13:
Hola tengo una DAVINCI AIO, acabo de hacer la instalacion, pero la impresora se reinicia cada 24s que puede ser he echo todo lo de los comentarios y nada me funciona