luc-github/Repetier-Firmware-4-Davinci #75: Autobed leveling

pallemar, 2015-05-13:
This have not been test by me yeat, but o far I have brought a blue tip sensor, it should have been a orange tip, to make it all easier, but according to Tom's guide, https://www.youtube.com/watch?v=EcGFLwj0pnA I just add some resistors at the end to reduce the voltage output. but before you add it to the board please measure it first to be sure it is below 3.3V, by testing it.
here is the one i got, 
http://www.ebay.com/itm/221457637744
here is the one I shoulde have got.
http://www.ebay.com/itm/LJ12A3-4-Z-BY-Inductive-Proximity-Sensor-Detection-Switch-PNP-DC6-36V-/140907575134?pt=LH_DefaultDomain_0&hash=item20cebef75e

according to this guide. http://www.instructables.com/id/Enable-Auto-Leveling-for-your-3D-Printer-Marlin-Fi/

Comment from luc-github: current one it using probe/plate like end stop to says bed is reached - it is just a simple contact system.
The Davinci pins do not accept more than 3.3V, so you will have to adjust the signal of your sensor to avoid to damage the board as I see it need at least 6V so you will have to power 12V- unless 5V is enough but still too high for Davinci pins

Thanks to luc-github for helping me :) hope all goes well tomorrow, then I test it.

luc-github, 2015-05-22:
Any progress on this ?

luc-github, 2015-06-05:
If no update I close issue - just reopen it if needed

closed 2015-06-05 by luc-github