luc-github/Repetier-Firmware-4-Davinci #76: stepper current manage by fw

luc-github, 2015-05-16:
I see several printer boards manage stepper current by fw, as Davinci stepper drivers do not have potentiometer to tune current, is there a chance it id done by FW ? that would explain why dome people get them stepper hot and others not.
anyone has information on this is welcome.

I will check repetier current implementation to so how it works, but help is welcome

linpp, 2015-05-19:
Not sure if this will help, as you've probably already looked at the data sheet for the A4988. In case you haven't, on page 12 is the schematic for a reference board layout.  The current limiter is managed by voltage on pin 17, as described on page 9, Itripmax = Vref / (8 x Rs).  On the schematic, pin 17 is wired to a simple voltage divider, Vref = (R2 / (R2 + R3)) * Vdd.  Driver boards with a potentiometer in place of R3 would simply adjust Vref by turning the pot.

As you know, on Da Vinci there is no pot, but there are two resistors off pin 17 in a fixed voltage divider similar to the reference schematic.  If, as in the reference, it is tied to Vdd, then there is likely no way to control it from FW.

It would be possible to design a board that takes PWM into a capacitive divider on pin 17.  Vref would then vary based on the duty of the PWM, which could be controlled by FW.  I don't believe that is the circuit we have on the Da Vinci though.

luc-github, 2015-05-19:
Yes - I have checked it - looking for clue where the components should be  - I have also parsed the board looking for a dac or a digipot but I did not find any - at least on 2.0 old generation - I cannot confirm for new board generation, but I do not see any reason to introduce it on new generation if was not used on old one, as new generation is a cost down version.

So seems no way to workaround by FW for people who have motor getting hot but only by decreasing speed and accelleration

luc-github, 2015-05-22:
Ok I think need to close this topic

closed 2015-05-22 by luc-github