luc-github/Repetier-Firmware-4-Davinci #77: Printer goes "idle"

SlickPLead, 2015-05-24:
I have been using .92 for two days now and the first couple of prints I had worked fine. 

 * Recently I have been getting printer idle message in the middle of a print.
                 So, I moved the file to my SD card I I am getting the same issue.

I'm not sure what is causing this but I would like to help get this fixed. Anything I can supply you with just let me know.

SlickPLead, 2015-05-24:
thought that I should leave a little more detail...

It's the new version of XYZPrinter with J37 or serial that starts with 3F1...

* Also I dont know how to pull the event log for this printer. Can you give me a breakdown so I can post it?

luc-github, 2015-05-24:
it happen with all print or they are new model you never print before ?

what repetier log display  in terminal ?
[image]

SlickPLead, 2015-05-24:
I can try one that completed 100% again when I get home but the last few prints have been going idle before they complete then I can't control the temp with the app unless I reboot the printer but I can control the temp in the LCD panel. Another side note, I can move the axis with the app at when the machine goes idle.

* Any ideas? 

Ha, I didn't even notice the log...I'll print something in the morning and see what it does.

luc-github, 2015-05-24:
you means you are not in front when it goes idle ?
so the printer may have been reset by the watchdog

SlickPLead, 2015-05-24:
Well I don't like sitting at the computer desk all day lol. So I am not around when the idle error happens.
 So it's possible watchdog is resting it but I thought by printing from the SD card would eliviate that as a possible issue. 

 * how is watchdog implemented? \embedded in the Firmware, the application or both ? 

Also if it's in the firmware what steps do you recommend ?

luc-github, 2015-05-24:
watchdog reset printer when printer is in dead loop to avoid temperature being out of control

I suggest to 
1 - try a print with previous model that was ok to see - if issue come from a slicing issue because model is not manifold and command generated bug the FW
so need to pass the model to a corrector like : https://modelrepair.azurewebsites.net/

2- still issue that means something is wrong but I do not know what - you can disable watchdog in FW there is a flag in configure.h change it and reflash - it may help you to see the real problem but becareful as one safety feature will be desactivated

SlickPLead, 2015-05-24:
Thanks for the feedback. 

I'll try one of the prints that went through without any issues then i'll pass the print through a corrector.( I really don't want to disable watchdog but the temp have always ran fine.

 * If it is watchdog would a additional heat-sink help ?

SlickPLead, 2015-05-24:
I totally read that wrong.... what could couase a dead loop? Even when printing from the SD card ?

luc-github, 2015-05-24:
sorry it is morning here, I mixed my explaination fix.
there is 2 safety : 1 for temperature control, the flag in configure.h for decoupling error that can be hidden by a idle message
2 watchdog to avoid deadloop, to disable it need to edit variant.cpp

the dead loop can happen when printer is put off control due to bug or unexpected conditions, like some commands generate a buffer overflow so printer freeze

I do not know which of these safety is occuring unless you are in front of printer to see there is a restart or a decoupling error, I guess would be a feature to add when printer start or has decloupling error to log it on sd , but today there not

SlickPLead, 2015-05-24:
So I won't be able to tell which issue is occurring by the log? How would I determine which one is the throwing the flag?

luc-github, 2015-05-24:
the log may show there is a communication issue or what command is stuck, so can lead to a direction to investigate, but right now no it will not tell you straitforward what it the problem

luc-github, 2015-05-24:
that is why I suggest to print a previous model that do not have issue to check issue come from model or printer

SlickPLead, 2015-05-24:
Ol, I'll keep you posted and copy the log in a couple hrs.

SlickPLead, 2015-05-24:
I was able to print a module this morning that worked previously. It might be an issue with the slic3r...?

 I am going to do a couple more prints and see what happens...Thanks for the support and everything you have done thus far. 

I'll keep you posted.

SlickPLead, 2015-05-24:
The one thing that concerned me is the buffer, it keeps going to 0 then buffers back up. Can I increase the buffer size or the refresh rate?

luc-github, 2015-05-25:
you can increase the buffer in repetier host to see if it help, this was suggested and experienced working with another software :s3d,

about slicing issue yes can be a root cause is model is not manifold or has other "bug"

SlickPLead, 2015-05-26:
I'm still not sure what's going on....I just had another print fail on me when it was looking superb. :( ...
But I was able to pull the log for this print......Can you take a look at this code?  Also can I force the print to continue without starting over? 

20:44:14.246 : N56767 G92 E0 *98
20:44:14.261 : ok 56767
20:44:14.261 : N56768 G1 X90.837 Y138.604 F6600 *88
20:44:14.261 : ok 56768
20:44:14.261 : N56769 G1 E1 F1200 *50
20:44:14.339 : ok 56769
20:44:14.339 : N56770 M204 S700 *66
20:44:14.371 : ok 56770
20:44:14.371 : N56771 G1 X91.101 Y138.868 E1.03045 F5100 *21
20:44:14.386 : ok 56771
20:44:14.386 : N56772 G1 X90.837 Y141.69 E1.26178 *70
20:44:14.480 : ok 56772
20:44:14.480 : N56773 G1 X94.852 Y145.704 E1.72528 *120
20:44:14.589 : ok 56773
20:44:14.589 : N56774 G1 X91.766 Y145.704 E1.97715 *117
20:44:14.685 : ok 56774
20:44:14.685 : N56775 G1 X90.837 Y144.775 E2.08446 *121
20:44:14.747 : ok 56775
20:44:14.747 : N56776 M204 S700 *68
20:44:14.810 : ok 56776
20:44:14.810 : N56777 G1 E1.08446 F1200 *45
20:44:14.810 : ok 56777
20:44:14.810 : N56778 G92 E0 *108
20:44:14.810 : ok 56778
20:44:14.810 : N56779 G1 X91.481 Y105.309 F6600 *94
20:44:14.810 : ok 56779
20:44:14.810 : N56780 G1 E1 F1200 *53
20:44:14.904 : ok 56780
20:44:14.904 : N56781 M204 S700 *76
20:44:14.997 : ok 56781
20:44:14.997 : N56782 G1 X90.837 Y104.666 E1.07433 F5100 *26
20:44:14.997 : ok 56782
20:44:14.997 : N56783 G1 X92.632 Y103.376 E1.2548 *75
20:44:15.091 : ok 56783
20:44:15.091 : N56784 G1 X93.307 Y104.05 E1.33269 *72
20:44:15.091 : ok 56784
20:44:15.091 : N56785 G1 X97.288 Y104.946 E1.66578 *114
20:44:15.122 : ok 56785
20:44:15.122 : N56786 G1 X95.718 Y103.376 E1.84706 *118
20:44:15.122 : ok 56786
20:44:15.122 : N56787 M204 S700 *74
20:44:15.186 : ok 56787
20:44:15.186 : N56788 G1 E0.84706 F1200 *47
20:44:15.202 : ok 56788
20:44:15.202 : N56789 G92 E0 *98
20:44:15.249 : ok 56789
20:44:15.249 : N56790 G1 X104.237 Y105.724 F6600 *100
20:44:15.249 : ok 56790
20:44:15.249 : N56791 G1 E1 F1200 *53
20:44:15.327 : ok 56791
20:44:15.327 : N56792 M204 S700 *78
20:44:15.358 : ok 56792
20:44:15.358 : N56793 G1 X103.837 Y105.324 E1.0462 F5100 *16
20:44:15.358 : ok 56793
20:44:15.358 : N56794 G1 X104.974 Y103.376 E1.23036 *65
20:44:15.421 : ok 56794
20:44:15.421 : N56795 G1 X105.814 Y104.216 E1.32738 *79
20:44:15.452 : ok 56795
20:44:15.452 : N56796 M204 S700 *74
20:44:15.561 : ok 56796
20:44:15.561 : N56797 G1 E0.32738 F1200 *33
20:44:15.561 : ok 56797
20:44:15.561 : N56798 G92 E0 *98
20:44:15.828 : ok 56798
20:44:15.828 : N56799 G1 X108.059 Y103.376 F6600 *110
20:44:15.828 : ok 56799
20:44:15.828 : N56800 G1 E1 F1200 *50
[... 1908 lines cut ...]
20:50:22.232 : ok
20:50:22.232 : N98 M105 *54
20:50:25.920 : ok
20:50:25.920 : N99 M105 *55
20:50:29.124 : ok
20:50:29.124 : N100 M105 *6
20:50:31.577 : ok
20:50:31.577 : N101 M105 *7
20:50:35.312 : ok
20:50:35.312 : N102 M105 *4
20:50:38.032 : ok
20:50:38.032 : N103 M105 *5
20:50:39.939 : ok
20:50:40.126 : N104 M105 *2
20:50:40.470 : ok
20:50:40.673 : ok
20:50:40.829 : ok
20:50:41.001 : ok
20:50:41.189 : ok
20:50:41.345 : ok
20:50:41.533 : ok
20:50:41.720 : ok
20:50:41.908 : ok
20:50:42.079 : ok
20:50:43.189 : N105 M105 *3

luc-github, 2015-05-26:
20:44:45.110 : start
20:44:45.110 : Host Print stopped by user.
20:44:45.126 : N1 M104 T0 S0 *0 
means a stop command has been send
3 possibilies:
1- menu stop has been selected in printer when printing from host
2- M50 has been sent from printer when printing from host but I do not see it log so should not be
3 - decoupling error 

As I guess you did not stopped the print or anyone did not when you are not there, you may have decouple issue, first test is to disable the decouple test in configure.h to confirm same message do not occurs again

SlickPLead, 2015-05-26:
ok, I will test it out tomorrow but i'm not the best at coding. Do I just change this to 0?

#define FEATURE_DECOUPLE_TEST """"1""""
//this will hide on sd card no extension files and bin/hex/dat files to make navigation and selection easier

luc-github, 2015-05-26:
yes this is the flag just change to  0

SlickPLead, 2015-05-26:
I updated the FW and printed one of the prints that failed with no issue. Maybe I should add an extra fan to the extruder?

luc-github, 2015-05-26:
no - it is not fan problem
decouple errors can be due to several problems
can be the connector of thermistor than can be loose 
can be the connector of the heater that can be loose

if you search on voltivo forum and on soliforum you will see it is a common problem
the solution usually is to change the connector or to sold the wires
but doing this may bring more issues as Davinci is very fragile and if you are not really good in soldering.

Personnaly I do have this time to time and just unplug replug firmly the thermistor and add some tape on connector/wire to avoid any movement

also this feature is present in repetier but not in stock so you can letf it disable or try to adjust the decouple parameters for your printer to give more "pad" if glitch happen.

if variant.cpp has the watchdog enabled I think it is ok to have decouple test disabled

SlickPLead, 2015-05-26:
That's fine. I have the E3d V6 on it's way so I can wait until then. I just wanted to print out the parts to mount it at least.

* Also I have been soldering for over 5 years at my previous job, I'm a pro. 

Thanks for the help

luc-github, 2015-05-26:
Good so we close the issue ?

SlickPLead, 2015-05-27:
I am still going to look up which sensor is causing the issue and see if I can get it fixed. Since I will be using the same sensor it may cause issues with the new E3D v6. 

So I'll leave that up to you....?

luc-github, 2015-05-27:
Well I suggest to close the issue as now the root cause of your problem is known : decouple error

I suggest to open a new issue related to decouple error if issue happen again with new data

SlickPLead, 2015-05-27:
Sounds like a good plan. Thanks for the help again!!

closed 2015-05-27 by SlickPLead

thevisad, 2015-05-27:
I am also having this issue, printer has stopped printing multiple times. I use Simplify3d for my printer and completed a 6 hour job last night. I have had two print jobs fail with a pause today. First thing I noticed is that the data that is being returned from the printer is not wait/ok. The data being returned is garbled http://pastebin.com/5LysYwaM

luc-github, 2015-05-27:
this looks more this issue https://github.com/luc-github/Repetier-Firmware-0.92/issues/38

change your S3D buffer as suggested and decrease baud rate on printer and S3D to 115200