luc-github/Repetier-Firmware-4-Davinci #78: Adding Multiple Extruders

lyweilian, 2015-05-28:
Have you ever thought about extending the firmware to connect to a slave board like RAMPS 1.4 over UART to add more print heads or to enable the possibility of multi-material extrusion (IE: E3D Cyclops, Chimera, Kraken)? It seems to make sense to use a slave board since we will not be able to populate the existing 1.0A board with components to convert to 2,0A without the possibility of letting the smoke out.

Just curious.

Thanks in advance!

luc-github, 2015-05-29:
IMHO changing the board will be easiest and reliable way

luc-github, 2015-05-29:
But if you have an existing solution please share

RAMPS 1.4 alone is not using UART, need a board for the RAMPS as far I know, I am not electronician but that is my understanding, a slave board connect by UART will need an MCU (like the ESP8266), if no MCU the slave board should then be connected to some existing PINS of master board

thevisad, 2015-05-29:
You are correct,  ramps itself is only a method for providing the pins
required to the individual components on it.

lyweilian, 2015-05-29:
Sorry I misspoke. I was thinking another Arduino Mega with RAMPS 1.4. The Arduino with RAMPS Shield would be the slave by connecting RX, TX, and SCL to the Davinci 1.0A. Thoughts? Possible?

luc-github, 2015-05-29:
I do not see the advantage to that - replacing DAVINCI with Arduino Mega with RAMPS 1.4 is more easy and result will be same no ?

lyweilian, 2015-05-29:
True, however having a slave board can allow the addition of more than two that is the limit of RAMP1.4 alone without a slave board. Perhaps the work isn't worth the pay off. Maybe using a Duet may be better. Thanks for clearing the air of the idea. Gonna close this issue.

closed 2015-05-29 by lyweilian

luc-github, 2015-05-29:
well having 4 extruders on davinci is more than an easy challenge = change the board, change the extruders, change the carriage and the bed glass
you must consider that the 4 heads must be in exact same plan, if any of the 4 nozzle is not in same position of others it will ruin the print, this must be done also with a very flat bed - which is not what davinci is.

Having a dual extruder, I already have to play with the bed leveling to get my second extruder not ruining my print, so I cannot imagine the difficulty with 4 nozzles.

For davinci with 4 colors I would more go for a mixing extruder than 4 separate extruders - but this is my personnal though