luc-github/Repetier-Firmware-4-Davinci #79: Z Offset doesn't apply when calibrating

laptopdude90, 2015-06-03:
I'm trying to use a standard Repetier heatbed and glass with my Davinci 1.0a. It's about 5mm thick, so I'm using an offset of 5mm. However, when I calibrate, the hotend runs into the bed. Any way to fix?

luc-github, 2015-06-03:
you need to do a proper bed leveling at Zoffset=0 using the screws 
Unlike stock fw, repetier fw allows you to do anything if you have 5mm offset instead of bed leveling properly and you do a Z homing alone with the noozle on middle of the bed you will break your glass.

Z offset is for small adjustment, fine tuning, not such big gap, for big gaps you need to use the screws under the bed and  do a proper bed leveling

laptopdude90, 2015-06-04:
The screws don't go far enough, unfortunately.

Can I somehow level it with gcode?

luc-github, 2015-06-04:
well there is a way but it won't prevent any accident if you do Z homing alone and I do not know if it is reliable :
in configuration.h line 844 ```#define ENDSTOP_Z_BACK_ON_HOME 0```
try ```#define ENDSTOP_Z_BACK_ON_HOME 5``` or what ever the value you need it should drop bed to 5mm after homing and set this new position to Z = 0

Another solution, hardware this time, print a longer flag (+5mm or what ever you need) for the Z end stop, people who mod their DAVINCI with EDV6 extruder without changing carriage use this mod as  
extruder nozzle is longer than stock one so they are in same situation as you.

luc-github, 2015-06-04:
here the flag : http://www.thingiverse.com/thing:621440

laptopdude90, 2015-06-04:
Oh, I like the hardware solution. Going to have to try that out.

luc-github, 2015-06-04:
IMHO this is the safer one

laptopdude90, 2015-06-04:
Any idea on a temporary one I can use? Obviously, I can't print one off.

luc-github, 2015-06-04:
I suggest you to test the software solution it should allow you to print the flag - just do not use Z home alone but Home all

laptopdude90, 2015-06-04:
Hmm, I'll try that, but first I'm going to try a DIY hardware thingy, like maybe a piece of wood, or some rigid stick notes.

luc-github, 2015-06-04:
flag need to be black -  I cannot recommand anything as I do not know what 1.0A end stop looks like, XYZ change their hardware and have at least 2 kinds

laptopdude90, 2015-06-05:
Confirmed working, stuck a sticky note on the metal bit.

luc-github, 2015-06-05:
good - if you do a new flag design please share it - would be helpful

laptopdude90, 2015-06-05:
I plan on designing one eventually.

luc-github, 2015-06-05:
Ok Happy printing :smile:

luc-github, 2015-06-06:
I close issue then

closed 2015-06-06 by luc-github