luc-github/Repetier-Firmware-4-Davinci #80: Optimized settings for AiO (no scanner)

luc-github, 2015-06-04:
Add better default setting for AiO, bed size, plate positions, homing position, etc

zeus10, 2015-06-04:
bonsoir avant tous jtiens a remercié tous ceux qui ont participé a ce projet.  bien vu vous avez raison j'ai mesurer et je suis a 6.8mm 
 #define ENDSTOP_Y_BACK_ON_HOME 0 end #define ENDSTOP_Y_BACK_ON_HOME 7   
 pour l'instant ce paramètre est a 0 
mais cela veux dire que je doit recommencé et remplacer FW   je ne suis pas chaud pour le faire a votre avis pourrais ton a la place de FW modififer eprom sur repetier hot je sais que c'est une aio mais sur les autre modèle davin.. il ont aussi sur le lit chauffant c'est quatre coin en métal donc pas que moi qui a ce problème  autre question quand je lance une impression j'ai un bip qui me dis que le couvercle au-dessus de l'imprimante est ouvert alors qu'il n'es pas ouvert j'ai lu sur le forum Voltivo problème avec interrupteur qu'il doivent le changer par un mécanique mais je ne sais pas car traducteur goo loll 
je vous remercie d'avance pour votre soutien

luc-github, 2015-06-05:
ok ca fait pas mal de questions ca :smile: :
1  - reflasher n'est pas difficile si l'ordinateur que vous avez utiliser la premiere fois est toujours avec arduino , il suffit de : brancher l'ordinateur sur l'imprimante, faire la modification dans arduino ide et appuye sur compiler/upload (le bouton avec la petite fleche), pas besoin du jumper ni d'autres modifications. de plus vous ne pourrez pas endomager la carte, donc vous pouvez le faire autant de fois que vous voulez.
2- le ENDSTOP_Y_BACK_ON_HOME n'est pas dans l'EEPROM donc pour l'appliquer, il faut reflasher, pour les autres parametres de position en revanche ils doivent etre dans l'EEPROM.
3-l'AiO fonctionne mais personne n'a vraiment repondu ce qui marchait et ne marchait pas, je pense que le senseur du couvercle n'est pas reconnu par le Firmware, il marche mais ne communique pas correctement, il suffit de le desactiver dans le menu parametres/settings, au fait le menu est en francais si vous voulez, Edouard a fait pas mal de traduction pour le FW, si vous reflasher le fw :smile: 

Sinon sur AiO il y a une lamelle de metal pour nettoyer la buse ? Je me rappelle avoir lu que les DripBox avaient ete enleves pour laisse la place au scanner, du coup comment l'imprimante nettoie la buse de l'extrudeur?

zeus10, 2015-06-05:
bonsoir merci encore.. pour le nettoyage de l'extruder il étais dans le menu sur l'imprimante mais comme elle étais nouvelle je ne l'ai jamais employé il n'y a pas longtemps que je suis sur les imprimantes 3d quand je l'ai acheter j'ai étudié blender et ensuite l'imprimante 3d  je cherche a comprendre tous et c'est pas facile mais j'y arrive pour l'instant j'étudie configuration.h Repetier
je suis technicien Electronique donc moteur stepper ect.. je connais après je ferais des essais
et je reviendrais dire ce que j'ai observé  petite question pourquoi je ne sais plus entré sur le forum voltivo mot de passe ne passe pas même après réinitialisation bizard

luc-github, 2015-06-05:
pas de soucis

pour voltivo je ne sais pas desole, je peux me connecter donc ce n'est pas le forum

zeus10, 2015-06-08:
Bonjour, problème 1 : avec le scanner j'ai dans le bas de l'imprimante un plateau tournant et en faisant des essais d'impression, qui ont donné d’ailleurs de bon résultats le lit chauffant a la fin de l'impression c'est abaisser mais j'ai du intervenir car il descendait beaucoup trop bas il a touché le plateau tournant, j'ai vérifié dans le configuration.H mais beaucoup de paramètre et comme je ne tien pas a craché mon imprimante je ne sais ou modifier pour que le lit chauffant ne descende pas si bas  .
(épaisseur du plateau tournant qui se trouve dans le fond 30mm).
2eme question : Quand j 'envoie mon fichier G CODE sur ma carte sd si je le lance de slirc 3 alors sa fonctionne mais quand je vais sur le tableau de commande de l'imprimante et que je veux avoir accès pour lancer du tableau sa n'a pas l'air de fonctionner.
3eme question : les deux pièces que j'ai réaliser ne se trouvent pas au centre du plateau, mais plus vers la gauche et trop vers la porte de l’imprimante voila donc je sais que sa fait beaucoup de question mais dans l’état actuel des choses ma connaissance actuel ne me permet pas d'effectuer moi même c'est connexion 
ps: mes résultats d impression son magnifique 
d'avance merci

luc-github, 2015-06-08:
1 - les distances MAX sont modifiables directement depuis l'EEPROM qui est editable depuis repetier host (menu configuration->EEPROM)
[image]

Pour Z c'est Z Max Length qu'il faut diminuer, a vous de definir combien - je vous conseille commencer a 150 pour voir puis augmenter petit a petit jusqu'a la bonne valeur.

2 - qu'est ce qui ne fonctionne pas ? le fichier n'est pas liste ? ou il est visible dans le menu de l'imprimante mais ne fait rien ?

3 - cela depend 
   1-  des valeurs Xhome pos et Yhome pos defini dans l'eeprom, pour que l'imprimante sache ou est l'origine de l'impression, elle sont pour une 1.0 donc pas faorcement adaptes pour une AiO (ces valeurs sont modifiables dans l'EEPROM) 

   et 2 de la taille de la zone d'impression (le lit) definit dans le slicer, car si la taille n'est pas correcte forcement ca ne sera pas centre

zeus10, 2015-06-09:
bonjour merci pour tous  pour la question 2  je vais sur main menu ,sd card,imprim fichier,et la j'ai toujours retour  mes fichiers ne sont pas la, il sont pourtant sur la carte sd 
d'avance merci 
j'ai réussi a corriger la auteur du plateau à 180 grâce a vous

luc-github, 2015-06-09:
il n'y a aucun fichierqui apparait ? ou il manque juste les votre?
 comment transferez vous les fichiers sur la carte SD ?
quel nom donnez vous aux fichiers

zeus10, 2015-06-10:
bonjour et encore merci bien vus car je n'avais pas mis l'extension .gcode c'est pour cela que je ne pouvais pas les voire. slic3 les avais envoyé en binaire sur la carte SD

luc-github, 2015-06-10:
oui j'ai mis cette securite pour eviter que l'efface depuis l'imprimante un fichier comme l'EEPROM virtuelle ou les fichiers du stock FW 
toutes les extensions sont affichees sauf les .hex, .bin, .dat et les fichiers sans extension

luc-github, 2015-06-25:
Je viens au nouvelles
A part le Zmax a 180 et le ENDSTOP_Y_BACK_ON_HOME 7
il y d'autres choses a modifier pour l'AiO ?

luc-github, 2015-07-11:
Anyone has optimized setting for AiO ?
currently feedback is limited to : Zmax to180 and ENDSTOP_Y_BACK_ON_HOME to 7

kryts, 2015-08-02:
I am using this firmware on my AIO I am not sure what information you need but I certainly am willing to help.

I do know that the top sensor doesn't work correctly in this firmware on the aio it is always indicating that it is open even when it is closed.

The AIO doesn't have the cleaning box like the others do so I added this gcode to make it clean the tip.

G1 Z3 F5000 ; lift nozzle
G1 Y50  F500
G1 X0 F500
G1 X-33 F500
G1 X0 F500
G1 X-33 F500

I have my x home as x -33 y as -12  z 0
max x length is 237 max y length is 217 max z as 202
I had to change the z probe locations
the z probe  x1 is -7 y1 is -10   x2 is -7  y2 is 200 and the x3 is 179 y3 200.

I have since then put in the e3d-v6 and the z probe doesn't ever work with it. it now just thinks it touches as soon as you start the probe. so I have the auto level turned off.

I also now have the filament sensor off since the board isn't guiding the filament.

luc-github, 2015-08-02:
Hi thanks a lot for your feedback - really appreciated
since you already moved to e3d-v6 do you remember:
1 -  if  z = 0 when homing the extruder hit the zprobe plate ? is that correct ?
2 - previous feedback said Zmax is 180 not 202 as defined due to Scanner plate support,  from another hand xyz specifications say 190, but on your side you seems not having this issue - can you confirm ?

thanks a lot

kryts, 2015-08-03:
The z did = 0 when it hit the probe plate when I had the original print head. I did have to change the position where it probed since originally it was  hitting the screw not the plate. Once I adjusted the x,y position it worked fine for the z probe with the  stock hot end.

Instead of the cleaning box there is a wiper that is on the side of the bed. I can take a picture of it if you want. I think with the scanner the cleaning box would interfere with the scanner. so they just have the wiper that is on the side of the bed. the wiper that is mounted to the bed isn't to high off the bed  its a 5mm but the height shifts some the low is about 3mm and the high is 6.

I created that code when I had the stock hot end.

My code moves  the hot end to the wiper and makes move back and forth over it. which cleans off the excess plastic from the nozzle. You can speed up the moves if you wanted and don't have to go fully back to home but I did that speed locations since I knew that would work. 

The other reason I did that is the first print I did after flashing and using the repetier software with the stock head had the print head move over the heated bed hold down screws banging them across it . I was using slicer.

I can bring it fully to 202 but it is almost bottomed out I might be able to go slightly more but I didn't dare to do so. Perhaps the connecter that joins my z axis screw is slightly different in position. then who you got the original feed back from?
If 
I am still using the carriage that came with the printer  so the x and y is still the same position ( at least I think it is it certainly seems that way from my tests.
 
My z may have shifted a small amount from stock but its well within the build plate adjustment. so my z is still 0 from the end stop.  I obviously can't do the z probing any more.  if I knew what wires I needed to hook up in the current setting to make it work I could do that and try it if you want.

Currently if I try it, the probe thinks the hot end has touched the plate no z movement and it just moves to the next probe point. The z-probe currently always reads high.

luc-github, 2015-08-03:
Yes a picture would be great
 I will check if when contatct Z-probe read H or L when back home and let you know the expected value as I do not remember - now, it could help for correct wiring check

Thanks again for your feedback

kryts, 2015-08-03:
Here is a picture.

[image]

kryts, 2015-08-03:
The top sensor did work correctly when I had the stock firmware. But current firmware always reads it as open. If you need any other photos or want me to test something I will be happy to do so.

luc-github, 2015-08-03:
ok thanks so I will add your code to M100 clean nozzle command when will add formal AiO support this week

so about z max what is your recommendation ? follow specs=190 ?
also about ENDSTOP_Y_BACK_ON_HOME value?

about top sensor looks like pin is not correct for AiO so need to find out which one is it

luc-github, 2015-08-03:
I will build a test sketch to help to find out the right pin

kryts, 2015-08-03:
After this print I am doing right now I will lower the bed to the 202 and take a picture of the clearance for you and post it up here. 202 is really close if I remember right so the 190 is a safer bet. the print has 32 min to go. I figured it wasn't the right pin but I am not sure how to find out what one it is. I can trace it back from the sensor to the mother board if that will tell you anything.

luc-github, 2015-08-03:
thanks a lot, or I can provide you a test fw to read different pins values when you open and close top to identify which one is reacting if you have time

kryts, 2015-08-03:
Sure I will be happy to.

luc-github, 2015-08-03:
thanks a lot will send it tomorrow

kryts, 2015-08-03:
ok, and not a problem. I still haven't received the email notification from github and I am not sure if that is publically listed any way. so let me know if you need to send it via email and I can give you my email address

luc-github, 2015-08-03:
yes notifications are sometimes slow, no worry they always find a way ^_^

luc-github, 2015-08-03:
yes all is public, the only private messenging in github is gitter if you target one person for the chat

kryts, 2015-08-03:
One other thing to note is the glitter badge does not light up on the AIO either I certainly am not worried about that but if you wanted me to do any of the test surrounding that I can do that as well.

luc-github, 2015-08-03:
sorry what is " glitter badge " on AiO?

kryts, 2015-08-03:
Its the light badge you can put a picture in and then it lights up.

[image]
[image]

Here is a picture of it. Maybe they 1.0 and 2.0 don't have this?

luc-github, 2015-08-03:
nope they do not have, will add this in test code to find out the pin :)
thanks a lot

luc-github, 2015-08-03:
what is the behavior of this badge ?is just follow light behavior?

kryts, 2015-08-03:
Here is the picture at z 202
[image]

as you can see it really is tight. If your cables don't drop just right they can bind up in there. I always watch it when I go down that far to make sure it doesn't. but for unattended I would stick to 190. 
in the stock firmware it was on when the printer was on and there was a option to turn it on or off.
it no longer is on at all. I thought the other printers had it to but it appears not to be that way?

However it certainly not something I am worried about.

luc-github, 2015-08-03:
ok actually it is very important for a coming feature, a command to low down the bed to prepare bed or when print is done

luc-github, 2015-08-03:
ok will set 190 and may be adjust later

kryts, 2015-08-03:
For comparison here is 190[image]

luc-github, 2015-08-03:
yes 190 is safer you right and seems no need 180

thanks a lot

kryts, 2015-08-03:
Not a problem. if you need any other info that pertains to a AIO or any tests for it I will be happy to help. I can even put the stock hot end back in if needed.

luc-github, 2015-08-03:
thanks a lot I do not want to bother you with stock extruder, should be ok with your current one
I am still out will add a function to read pin for sensor, and give you a pin list and process for badge, and will check behavior for probe

luc-github, 2015-08-03:
Reading at the code seems current FW has all tools to test pin just using GCODE and the log of repetier
if want to test Pin 78 (LED light)
```M42 P78 S0``` will set pin to LOW
```M42 P78 S255``` will set pin to HIGH 

so useful to find the badge

to Read a pin like 5 (Zprobe)
```M42 P5``` result will appear in repetier host log

useful to find the change of state of a pin for the top sensor

this will allow to use all pins but the ones for extruders sensors/heaters/steppers 
you can use the pinout https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/pinout.xlsx
to check what pins are availables for AiO

I will prepare a GCODE file you can use to test pins tomorrow as now it is too late for me

kryts, 2015-08-03:
On the Xlsx sheet is the pin number I am polling or setting the arduino number?

luc-github, 2015-08-04:
This is the list of pins with their known usage or availability - I will provide you today a list of possible pins and a GCODE script to be used in repetier host for testing using M42 command
I just want to test on my printer first to be sure it is ok and safe

luc-github, 2015-08-04:
Basic support for AiO is now added for positions - use DAVINCI 4

https://github.com/luc-github/Repetier-Firmware-0.92/commit/04f9bab18fd6add9256de80941c4a839be0e8938

I will deliver the test GCODE quickly for the missing pins

luc-github, 2015-08-04:
I have updated the pinout to add AIO tab and list missing pins
I have added a script to check pins used for sensor readinput.gcode - you need to manually check if any state changed and so identify the related pin in repetier.log

I have added a script to check pins that state can be changed -  as this may affect the printer and evn damage the printer if wrong state is set to a pin for long time, it is safer to do little number of pins then restart printer, end of the script contain a M112 to reset printer to help on this, when reset done you need to manually change pin numbers for a new batch of test- you can use readinput.gcode as reference or use directly pinout.xls

good luck and thanks for helping the FW getting better

kryts, 2015-08-04:
Top Door is P64 1 is closed 0 is open

kryts, 2015-08-04:
If you want the entire log I can post it

kryts, 2015-08-04:
M42 P6 S255  turns the light badge on 
M42 P6 S0 turns the light badge off

kryts, 2015-08-04:
the filament sensor is the same as in the 1.0a models and so is the z probe. Its just now that I have the E3D in there I cant use the filament sensor since I can't run the filament through the board. 
The zprobe sensor isn't valid when I have the E3D either It did work fine before I put in the E3D after changing the z probe xy locations 

I am not sure on the jam sensor. I don't know how to test that.

kryts, 2015-08-04:
When I put filament through the filament sensor on the daughter board M42 P24 switches from 1 to 0 so  p24 is still the filament sensor on the AIO.

kryts, 2015-08-04:
If its easier on the light badge for it to follow the internal lights I think that would be fine. I don't see why you would need to have it on if you are turning off the other lights.

luc-github, 2015-08-05:
wow you did a lot ! ^_^ 
ok will add quickly  support for light and door sensor as you confirmed others sensor, no worry for Jam sensor, it is not fully implemented yet.

about light badge, can you describe stock behavior for my information ? any behavior is easy to do so just let me know which one is best, if follow internal light is best I will do ^_^

kryts, 2015-08-05:
Stock you could turn it on or off in the menu. 
I don't know if it turned off with the power save option as I didn't have that on. I like having the lights on so I can watch it.

kryts, 2015-08-05:
Either having an option to turn it on and off and still have the power save turn it off, or have it follow the internal lights is fine.

The advantage to having of and on separately is that it is really bright so you may not want it on.

luc-github, 2015-08-05:
Ok so it follow light behaviour, and can additionally be off independently

luc-github, 2015-08-05:
should be done today - thanks a lot for your help

kryts, 2015-08-05:
no problem if you need other tests for the aio just let me know and I'll try and do them

luc-github, 2015-08-05:
thanks a lot - I will release a new FW later today to support sensor and badge - if you can check it is ok it would be great - I can simulate badge for pre test but better real test for light

luc-github, 2015-08-05:
ok support for AIO should completed now : https://github.com/luc-github/Repetier-Firmware-0.92/commit/5763e174df09daf0aeff30f34473d24e4bc50fb6
Let me know if something is missing / not working as I do not have a AiO only the old 2.0 Single fan ;)

Thanks a lot for making this FW better

kryts, 2015-08-06:
I will download the updated fw and flash it again after I get these series of prints done. and let you know how it goes. I am printing some stuff for some one who is leaving work so don't want to change it while it is working since that is time sensitive. but I should have the results this week for you.

luc-github, 2015-08-06:
sure no  problem - take your time

[339 more replies]

closed 2016-05-11 by luc-github