luc-github/Repetier-Firmware-4-Davinci #89: Arduino 1.6.5 support

luc-github, 2015-07-07:
Currently using 1.6.5 IDE  - with same variant.cpp / USBcore used for 1.6.3
Printing during several batch of 8h+ without problem using SDCard - no test done yet using USB as no PC close to the printer

Any feedback is welcome

g-goessel, 2015-07-22:
Used 1.6.5 with same variant.cpp / USBcore used for 1.6.3 and Arduino SAM board 1.6.4 from Boards Manager on Ubuntu to flash
It works fine with direct USB and Windows repetier-host

luc-github, 2015-07-22:
that is good news - thanks a lot

luc-github, 2015-07-22:
I have move 1.6.3 to 1.6.5 directory - thanks again for confirmation
https://github.com/luc-github/Repetier-Firmware-0.92/commit/dd8940063daba8ef81d7e8dddc959501db95d17f

luc-github, 2015-10-16:
as seems ok - I close issue

closed 2015-10-16 by luc-github