luc-github/Repetier-Firmware-4-Davinci #91: Resetting when pressing print file, 0.92

tayiou, 2015-07-11:
Davinci starts and i can use everything but when i press print file it freezes. 

I used 1.5.8 beata to program
settings/Load_eeprom
it also will also freeze when connected to repetir host software and not do anything.
the screen just turns dim and stays frozen. 
any suggestions? 

It worked with 0.91 but i have having problems with 0.92

luc-github, 2015-07-11:
what freeze ? printer or host ?
did you used the variant.cpp/usbcore.cpp provided with 0.92 ?  this will enable watchdog so printer won't never freeze but in worst case reset

if printer is working standalone - it is communication issue with host
check baud rate is same on both sides and repetier host has proper settings

tayiou, 2015-07-11:
The printer was not working with standalone or host, disabled watchdog and reset again. Works form host now but still freezes when i try to print from the printer as standalone

luc-github, 2015-07-11:
may I know why you disabled watchdog ?
also does heat extruder / heat bed / move axis / sound  / light are working ?

tayiou, 2015-07-12:
Everything else works. 

I disabled it because i seen it in another post.

luc-github, 2015-07-12:
well watchdog is a safety feature and works perfectly on old generation (the one you and me have), new board seems having to init the timer at first flash but I just did  new fix
In your case disable it is not good as printer freeze for some unknown reason and this is dangerous if you are heating.

If load-failsafe setting do not solve the problem I suggest to delete the EEPROM.bin on SDCard 

About serial communication be sure WIFI feature is not set when compiling or disable it in Settings (Wifi Off)

once done change printer baud rate to 115200 and then connect using arduino serial monitor to see what printer display

luc-github, 2015-07-12:
also can you share the GCODE which freeze printer ?

luc-github, 2015-07-14:
did you solved your issue ?

luc-github, 2015-07-18:
any update ? should I close issue ?

tayiou, 2015-07-18:
Sorry I have not been able to use it. It works fine with the computer but does not work in standalone but this is fine. I am currently working on a dual extruder for it with one stepper motor and a servo. Thanks for the help.

luc-github, 2015-07-25:
Ok so I close issue

closed 2015-07-25 by luc-github