luc-github/Repetier-Firmware-4-Davinci #93: Davinci 1.0A

celevra, 2015-07-21:
Hi,

thank you for your work, you asked for feedback everything worked perfect. 
Installation -> no issues
Calibration -> no issues
first print -> no issues

if there is anything i should test, tell me.
thank you again

regards

celevra

luc-github, 2015-07-21:
Thanks a lot for your feedback, really appreciated
can you confirm you flashed with watchdog enabled ? this is the random issue on 1.0A - first flash make printer to restart, but with latest version I changed WD initialization command

Thanks again

celevra, 2015-07-21:
i'm not sure what you mean with watchdog, i've done exact every step in the step by step Installation guide here on github. 
i've first resettet the printer with the jumper j37, turned the printer on for about 10 seconds.
I'm on OSX and downloaded arduino 1.5.8 Beta
copied the files, changed the configuration to 1 for the A version, changed the port and pressed upload.
i'm not sure if the printer restartet automaticly but there was a really loud "beep"
i have written the settings to eeprom directly on the printer not via repetier host.

hope there is the information you needed, if not guide me in the right direction and i will try to answer ;-)

luc-github, 2015-07-21:
it is perfect if you followed the installation steps it means you did not disabled the WD 
So this is good -yeah I have added the loud beep to show FW is working in case of issue with screen (at the begining screen was randomly scrambled , but it is fixed now and I left the beep)  

Thanks again for your feedback and happy printing :smile:

celevra, 2015-07-21:
sounds great, for clarification, is watchdog a protection mechanism from xyzware or something?

luc-github, 2015-07-21:
it is a safety feature in case of printer loose control for any reason and go into a dead loop without controling heat, if printer is out of control for 8s it will reset and so stop all heaters to avoid damage

luc-github, 2015-07-21:
xyz does not have this feature as far I know

celevra, 2015-07-21:
thanks for clarification
have a good one

uladz, 2015-07-25:
Running 0.92 firmware on Divinci 1.0A for a week already, no issues so far. Noticed two possible problems though:
1. When doing auto calibrating at 2nd point the head seems to be trying to move beyond max Y position. Meaning it moves Y axis all the way to the front and hits hard the closes to the door polley on the right with a loud noise and possible skipping a step. May break the pulley or gears at some point?
2. When printing is finished the head is moving to position x=0, y=0 without changing z position, which is the far right corner of the bed instead of moving to to -33,-12 parking position. I had an issue when once I've cancelled a print when it was still heating up a bed and the bed was at its highest position z=0, the head has moved to x=0,y=0 hitting the bed edge and possibly scratching it.

Otherwise prints are so much better then original XYZ firmware/software, can't compare.
Thanks a lot for all the work!

luc-github, 2015-07-25:
Hi thanks for your feedback
about point 1 : you may need to adjust the Y2 position in EEPROM, currently it is set to : Z_PROBE_Y2 203 and was given by 1.0 owner but 1.0A may be lower, please have a try to decrease it and let me know

about point 2: this is done by your end Gcode of your slicer not by printer itself, you may need to check it, also depending of host you are using there are some predefined macro that may do this behavior and you may need to edit them

Thanks again for the feedback

uladz, 2015-07-25:
Hi Luc,

Thanks a lot for your quick reply, much appreciated. I'll play around with Z_PROBE_Y2 and see if a lower value would work better for 1.0A.

For G-Code I'm using Repetier-Host + Slic3r. Could not find any way to move the hear the far right position yet, it always gets back to 0,0. Slic3r g-code does not have G1 x0, y0 code, so it must be Repetier-Host I think...

Thanks!!!

luc-github, 2015-07-25:
In repetier hosh check this:
[image]

go to park position should be unchecked or park position should have correct values not 0,0
better unchecked as if you kill a job at 50% go to Z=0 is very bad idea as model may hit carriage rods

uladz, 2015-07-26:
Thanks a lot, will try it out now!

uladz, 2015-07-27:
BTW, y=203 for the probe seem to be correct value, it stops right before the pulley. For some reason when you do auto calibration the head smashes into pulley though... Wait a minute, in the EEPROM settings I see that probe Y position is set to 205 instead of 203! That's the problem. Also max X length should be 237 and mx Y length = 215. These values seems to work fine.

On the repetier config, setting X=-33, and Y=-12 resolves the issue with the head going into position 0 upon job completion or job cancel. I've put Zmin to like 10, so the head is always above the bed. Repetier would only lower the bed if its Z position is < Zmin, but would never rise it, so no problem with hitting hit carriage rods.

uladz, 2015-07-27:
Gee, I need to be careful, twice hit a bed with a print head while moving it around to get correct max coordinates. Unfortunately there is no "blocking" protection in the motors, they continue to pull all the way even if a head is not moving any more and belts are skipping. Fortunately belt tensioners are saving mechanisms from fatal failure. Well...

luc-github, 2015-07-27:
Ok so you mean what should be changed for 1.0A :
```
#define Z_PROBE_Y2 203
#define Z_PROBE_Y3 203
```
and 

```#define Y_MAX_LENGTH 215 - ENDSTOP_Y_BACK_ON_HOME```

as X is already correct : ```#define X_MAX_LENGTH 237 - ENDSTOP_X_BACK_ON_HOME```

uladz, 2015-07-27:
yep. but i thing 1.0 and 1.0A mechanically are the same, only the board is different + minor cosmetic changes. so this might be true for 1.0 too.

uladz, 2015-07-27:
btw, off topic question, do you know if XYZ sells spare parts? looks like I've cracked one of the bearing holders :(. can not find any spares on XYZ website, only replacement bed, head, and tool.

luc-github, 2015-07-27:
1.0 values come from 1.0 owner, so I won't change them I think -  I will apply them only on 1.0A

about spare parts check voltivo forum, some people did some metal bearing holder, you may ask them, or there are printed version available on thingiverse also

uladz, 2015-07-27:
super thanks!

celevra, 2015-07-27:
recognized that behavior also, to park the extruder i use the gcode M100
works if its not allready parked, if it is parked you will hear the loud noise.
where do i have to change these values, i think in arduino IDE and flashing again, am i right?

luc-github, 2015-07-27:
all these values are in EEPROM no need to flash

and to park (XY) the command is G28 XY
This home XY but not Z

luc-github, 2015-07-27:
update done https://github.com/luc-github/Repetier-Firmware-0.92/commit/29d84876b834f990eae1b4474ea615e332a216b3
Thanks for your feedback

uladz, 2015-07-27:
BTW, doing auto leveling - does it actually affect anything? If I store the values in EEPROM I do not thing Repetier takes them into account anyway.

luc-github, 2015-07-27:
need to be sure autolevel is set to 1
when doing autolevel it generate a positions matrix used by FW, it is tranparent for host as FW will adjust the z position according the positions matrix when printing.
When using autolevel you will see z moving up and down permanently during printing

That said - I never use autolevel, only to validate code, I only use manual level, once done properly, I do not need to redo for several prints and when you get used to it, IMHO it is really faster than autolevel.

uladz, 2015-07-27:
Aha... I see. BTW, the Z_PROBE_HEIGHT is wrong for 1.0A, it's set to .28 now but I think the correct value is around .54. I'm comparing autolevel vs manual with paper, will let you know once I know the correct value (if I do not brake my printer before :)). Also one question, if I do manual leveling with paper, what's the Z-offset in slicer to use? Keep 0?

luc-github, 2015-07-27:
yes 0

uladz, 2015-07-27:
Okay, Z_PROBE_HEIGHT=0.56 seems to give me the most closest to paper-leveling numbers. I'll keep an eye on the level over time to make sure the this is the right value for 1.0A.

thevisad, 2015-07-28:
I just pulled out my feeler gauges, dropped the hot end to the plate and measured .38mm between the bed and the end of the Z probe  .56 is too much and .28 is too little.

uladz, 2015-07-28:
how did you measure?

thevisad, 2015-07-28:
I pulled out my feeler gauges, https://en.wikipedia.org/wiki/Feeler_gauge

uladz, 2015-07-28:
I see. You need to take into account the metal brackets that are in all 4 corners of the bed and that are used as contacts when auto level is measuring distance in 4 corners. This will and approx. 0.2mm and bring your number right to mine :).

I did manual calibration with paper sheet, then I have run through auto calibration and pulled the measurements right from XYZ Davinci itself.

Agree?

thevisad, 2015-07-28:
The metal brackets in the corner are .58mm high, found by dropping the Z probe to the metal bracket until it makes contact. Then with the tip of the hot end over the bed itself, using the feeler gauge to identify the distance between the hotend and the bed. The setting indicates that this is the distance between the very tip of the hot end and the very tip of the z probe. This follows the same setting in Marlin and should not be offset by the height of the metal brackets. 

/** The height is the difference between activated probe position and nozzle height. */

uladz, 2015-07-28:
.58 sounds good

thevisad, 2015-07-28:
I corrected the above actually, I forgot to add the two numbers together to get the actual thickness of the bracket. I verified this by pulling out the caliper and throwing it on the bracket. So the offset value does not include the bracket thickness.

uladz, 2015-07-28:
You've lost me here a little bit. Shouldn't this be way around, 0.92 is the metal bracket, 0.38 is the diff between probe and tip. Then the probe height would be 0.92-0.38=0.54 in your case as this will be the distance between tip and bed when probe touches the metal bracket. Or I do not understand something...

thevisad, 2015-07-28:
Yup, you are right, I am a bit tired and was adding the two. I at least have the correct offset for mine now. I would best it's pretty close for everyone elses (give and take a little due to the bracket bending).

uladz, 2015-07-28:
@thevisad, we are in agreement, awesome! :+1: 
@luc-github, I guess you can update the Z_PROBE_HEIGHT too for 1.0A.

uladz, 2015-07-28:
@luc-github, is it possible to add a menu item for lowering bed to the bottom? Would be extremely useful for prepping bed for a new print.

luc-github, 2015-07-28:
0.28 is value provided originaly by BGM and never checked it - when going home I will see if this value works for my 2.0

About menu - I use Z position there is a 100mm set, but can imagine a menu item to position to max Z

luc-github, 2015-07-28:
I have created 2 trackers https://github.com/luc-github/Repetier-Firmware-0.92/issues/98 and https://github.com/luc-github/Repetier-Firmware-0.92/issues/99 - will process them later - thank you

klHaribo, 2015-09-02:
the new firmware works perfect.
Changed to german language, but only some lines seems to be translated. 
I'll try to find some time to work on it.
Had some issues with heater decoupled, but it seems this is caused by opening the door and cold air flows into the printer.
Upload to SD_Card via Host would be great ;)

luc-github, 2015-09-02:
Hi thanks a lot for your feedback
Yes sorry - I only maintain English (too lazy to do others languages, even French and Spanish....) all menu need to be reviewed in others languages as they are for 20x04 screen when Davinci's screen is 16x04 one

For heater decoupled you may adjust the setting for your printer - it may be too sensitive 
```
// If the temp. is on hold target, it may not sway more then this degrees celsius, or we mark
// sensor as defect.
#define DECOUPLING_TEST_MAX_HOLD_VARIANCE 15
// Minimum temp. rise we expect after the set duration of full heating is over.
// Always keep a good safety margin to get no false positives. If your period is e.g. 10 seconds
// because at startup you already need 7 seconds until heater starts to rise temp. for sensor
// then you have 3 seconds of increased heating to reach 1°„C.
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
```
and

```
/** Time in ms between a heater action and test of success. Must be more then time between turning heater on and first temp. rise! */
#define EXT0_DECOUPLE_TEST_PERIOD 30000
```

the upload to SDCard from host already exists and works, but it uses repetier protocol which is very slow compare to wifi SDCard transfert

klHaribo, 2015-09-03:
due to my poor school english I don't really understand this text ;)
I found these settings already. changed DECOUPLING_TEST_MIN_TEMP_RISE to 3, autotuned PID's and it seems to work for now.
Printer only stopped when I was doing something in another room, seems like he wanted to fool me ;)

luc-github, 2015-09-03:
Hmm ```DECOUPLING_TEST_MIN_TEMP_RISE  3``` will increase risk of decouple error for my understanding, the best is to increase extruder instead
```EXT0_DECOUPLE_TEST_PERIOD 60000```
also check your connectors they also can be the root cause in some position

The-Experimentalist, 2015-10-15:
One note is that the display on the printer for the 'Light AutoOff' prompts for a value in ms but is actually really minutes. By comparing the values set on the printer and those stored in the EEPROM I can see that the value is configured on the printer in minutes and stored on the EEPROM in ms

Not a big issue but thought I would mention it for any later revisions. Thanks again.

bagwan, 2015-10-19:
Hi!
I have flashed repetier on my 1.0A before a Week, and printed some Meters of Filament without any Problem. Yesterday I would start a Print, and it stopped after 2 minutes with "Heater Decoupled" Okay, checked the Heater and the Cabels. No defects. So I startet a Job and measured the Pinoutput on the Board for the Heater. There are no Power on the Pins. 
Do you think I have a Problem with the Firmware/Software?, or is it a real Defect on the Board? In a Forum, which I´ve asked for a circuit Diagramm, theres another Person, which have exact the Same Problem. Flashed with Repetier, printed a few Meters, Then "Heater Decoupled"
Thanks for your Help

luc-github, 2015-10-19:
Can have several factor for decoupled error:
1- the connectors (sensor/heater are loose) this is common problem on Davinci - several people changed their connector or soldered cable directly and ensure cable are not moving
2- the door is open or a fan is blowing on extruder and so do perturbation on air flow and heating
3- hardware issue - no heat

none is FW/SW
 
first you need to eliminate point 2 by disabling additional fan if any and close door
if not point 2 to debug problem you can flash FW disabling decouple test in configuration.h then heat
if heating it is not point 3 but point 1 - check also all 3 led are working on main board if one of them is off it you got a hardware problem

you also can observe the heating chart in repetier to monitor temperature changes according extruder position to see if there is no short

Hope it help

bagwan, 2015-10-19:
First, thanks for your Quick reply.

The Cable and the Heater is definitv ok. The 3 LEDs on the Board are working. When the Heater should heat up, then should the Connector on the Board have 12V. But my Multimeters shows no Volts. on the Pins. So i Think, there is defective on the Board.
Is it meanwhile possible to go back to XYZ Firmware? Because i have actually 4 Weeks Rest Guarantee...

luc-github, 2015-10-19:
No for 1.0A there is no way to revert , XYZ have implemented a lock for this and no tools is able to flash back but the one from support 
 I suggest you to fix the problem here a thread for the problem:
http://voltivo.com/forum/davinci-peersupport/734-1-0a-no-power-to-extruder-heater-fix

or use the jumper to erase FW and contact XYZ and tell them you power on your printer and there is 2 black bar and never mention repetier FW they will send you a new board or ask to send printer back for fixing

bagwan, 2015-10-19:
Thanks 1000Times. That was my Problem. Fixed the Heater Problem.
Made my Day

thomastech, 2015-11-04:
Flashed my 1.0A (3F10A hardware version) from XYZ V1.0.3 to Repetier 0.92 a couple days ago. First, a big Thank You to the developers / contributors.

Some comments / feedback:

1. I used Arduino 1.6.5. Because of my confusion on the directory location of the two AdditionalArduinoFiles I fought the 2-black bar (brick) issue for awhile. After getting the files into the correct directories (under /users) things went much better. None of this would have occurred if I had used Arduino 1.5.8 since the two files go in more obvious directories ( under /programs files).

2. On my 1.0A the top cover open Alert did not work. I changed the TOP_SENSOR_PIN from 6 to 64 and that solved the problem for me.

3. I experienced a lot of host "wait" command delays during long prints. Increasing Simplify3D's buffer cache size to 2024 has reduced these events, but the wait delays still randomly occur. I understand that the issue is related to the TIMER1_COMPA_VECTOR() interrupt. If there is a patched version of HALL.cpp then I would definitely like to try it out.

Overall I really like the 0.92 upgrade versus the stock XYZ firmware. The stock feature I really miss is the ability to do standalone USB prints (send file, then shut off PC). So now I am looking into octoprint as a workaround. 

- Thomas

The-Experimentalist, 2015-11-04:
Not tried it yet but I believe you can do USB prints with Repetier Host but it is slow transfer due to using Repetier protocol to transfer the files from what I remember reading, I am sure Luc cac confirm or deny this ability. For simple cheap file transfer I just have a micro SD card extension to bring the SD into the main chamber. I have a DaVinci AIO and mounted mine on the redundant scanner module. I just pop out the SD and transfer the files using a Micro SD USB adapter on the PC and then put it back in. It only takes a couple of minutes and is cheap, see pictures. One other mod I made was to break out the firmware erase jumper to a button which I mounted on some prototype PCB and printed a little holder for. I can also access this from the main chamber so no need to take the back off again, came in handy when I was first messing around and will make any future firmware updates a 2 minute job.

thomastech, 2015-11-04:
// Not tried it yet but I believe you can do USB prints with Repetier Host 
// but it is slow transfer due to using Repetier protocol to transfer the files 
// from what I remember reading.

I have a Prusa i3 with Marlin firmware that can do this, but it is much too slow. So SD card printing is the workaround. But the Prusa printer is not my main go-to machine, partially because of this inconvenience.

// For simple cheap file transfer I just have a micro SD card extension 
// to bring the SD into the main chamber.

Same here, long ago I also moved the SD card socket to a better place (the front of the printer). 

[image]

But my davinci printer is in a remote room on the first floor (so I don't have to smell ABS and hear the noise). My office is on the second floor, so running back and forth was inconvenient. So my final solution was to plug the stock davinci into a wireless USB printer hub. This allowed me to send a print file to the davinci printer (host PC not needed after the file xfr). For my situation, this has been a fantastic way to print. So I miss being able to do print file xfr's after the 0.92 upgrade.

[image]

Maybe Luc can advise on good way to achieve what I need. But I have a feeling the raspberryPi octoprint solution is going to be needed.

// One other mod I made was to break out the firmware erase jumper to a button
// which I mounted on some prototype PCB and printed a little holder for.

Ditto. I added a pushbutton reset switch during my frustrating attempts to get Arduino 1.6.5 to build working firmware. Much better than shorting the PCB's pads with tweezers. 

- Thomas

The-Experimentalist, 2015-11-04:
If you go the Octoprint route you may want to read the article by Jake also recommended by Luc:

http://voltivo.com/forum/davinci-peersupport/501-da-vinci-setup-guide-from-installation-to-wireless-printing

There is a section on "Using OctoPrint for Wireless Printing" and he recommends the BeagleBone Black over the Raspberry Pi.

Might be worth a read?

thomastech, 2015-11-04:
@The-Experimentalist: Thanks for the OctoPrint link. Too bad the RPI is not recommended since I already have a RPI2 model B and a RPI model A+. But it seems that I will need to buy a BeagleBone if I build the OctoPrint project.

In the meantime, fingers are crossed that there's a magic 0.92 upgrade that has buffered USB printing. :)

- Thomas

The-Experimentalist, 2015-11-04:
Keep reading. I think if you already have a RPI then you can use it. Jake is recommending BBB if you don't already have hardware I believe. I think if you search you will find info on how to use your RPI. Good luck.

luc-github, 2015-11-04:
@thomastech thanks for sharing I will add the new pin for top sensor, about the patch - repetier is still working on it and I am waiting an official release as I do not use USB at all I prefer to be sure it has been heavily tested -  it seems more than a simple patch as I was suggested initially that is why repetier is testing it, this is my trigger to do a new sync between official repetier and davinci one.
About 

To print from SDCard, I recommend Wifi SDCard (toshiba one) copy is really fast
then as you already make your Davinci wireless, launch remotely the SDCard print using repetier host

or you can add an ESP module and my FW https://github.com/luc-github/ESP8266
so you can use repetier host to launch remotely the SDCard print or use the web interface to launch it but you still need a wifi SDCard for faster file transfer

luc-github, 2015-11-04:
@The-Experimentalist / @thomastech  Octoprint is good solution if you want all in one (slicer +host) and follow by webcam, it works on all Raspberry so no worry - but it use USB, and in my opinion SDCard print is more reliable - I used for a while on Pi and worked well

thomastech, 2015-11-05:
@all: I'll have to think about how to settle my quest for recreating / emulating the standalone USB printing method I've grown to enjoy using. My USB print buffer hub worked brilliantly for me on the stock davinci firmware (100% reliability), so it is unfortunate it is not a viable approach for buffered printing on my 0.92 upgraded printer.

At the moment I think I will back off from OctoPrint. Now I'm leaning towards ordering the Toshiba 8GB WiFi SD card.

@luc: In regards to the upcoming patch for fixing the host-command wait delays, if you need someone to test your patched davinci version then just holler.

New Question: On my Prusa i3 Marlin printer I added some custom hardware/software features. I'd also like to do some customization on my Repetier daVinci. There are several unpopulated connectors on the 1.0A (3F10A) board. I have the fantastic pins list that Luc posted -- Is there by chance a published list that shows which CPU pins are found on the various unpopulated PCB connector pads? 

- Thomas

luc-github, 2015-11-05:
sure, will do a test branch for this, will do annoucement when ready, thanks

I am not aware of this document, as so many boards now ....

[41 more replies]

closed 2018-06-02 by luc-github