luc-github/Repetier-Firmware-4-Davinci #94: DaVinci 2.0A Feedback

g-goessel, 2015-07-22:
Yesterday I installed repetier on my DaVinci 2.0A (jumper J37, `#define DAVINCI 3` and `#define MODEL 1`).
I did a manual bed calibration and it seems to work fine.

But when I tried to upload the Arduino sketch I had an issue (with both 1.5.8 and 1.6.5 on Ubuntu), the compiler said that String.h was not found ( https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Communication.cpp#L23 )
` Communication.cpp:23:20: fatal error: String.h: No such file or directory `

After googling I found that I had to replace the line with `#include <string.h>` and it flashed perfectly.

A feature that I am missing from the stock firmware is the fact that I could move my bed way lower. When the printing was done it would go almost all the way down but with repetier I can't have it lower than 150mm (when I try to move the bed with repetier-host it doesn't go any further). Is there any setting to change ?

And a disturbing thing is that the extruders name are quite inconsistent : on the screen they are called "left" and "right" (easy to understand), on the stock firmware they are called 1 and 2, with repetier host they are also called 1 and 2 but the are reversed (the 1 from stock is the right one and is called 2 by repetier). I think you should stick to the stock firmware naming.

When I try to pause/cancel/emergency stop the printing with repetier-host it is far from being immediate. I have to wait about 1 minute for the printer to obey my order. This is probaly linked with issue #25 .

When I am finished adding a new filament (this works fine) there is still a little bit of plastic (I only have ABS right now) that flows out. There is about 3cm of fine plastic which is quite a deal. I've never had this issue with the stock firmware.

Anyway I'm really happy with this firmware, I am finally free from the crappy XYZware, if you need any other feedback I would be happy to help you.

luc-github, 2015-07-22:
Good catch for string.h I am on windows and did not compiled for a while on mac and did not saw the case sensitive issue - I will fix it today

About bed lower this is something you need to add in your custom end code to as bed to lower when print is done, if repetier host does not allow you to go more than 150mm it means you put wrong distance for Z = print area height

About 1 and 2 reversed compare to stock this is the result of several discussion - by defaut 1 (left) is selected when homing and can be used directly - if select 2 (right) then extruder is moved to left to take position of extruder 1 (left one) this is done by repetier itself which is consistant because homing position is right.
if we name left extruder 2 and right extruder 1 when homing - because default movement is right to left the extruder won't move as right extruder is already in maximum position because homing is in right, so to be consistent with repetier host need to modify FW to change movement which make more difference with repetier fw and add extra maintenance. Also be noted this happen only with repetier host - S3D does not have this issue as naming can be configured to match printer

about delay for response yes this is a repetier issue - you can add M50 command in script for stop - it stop faster than normal stop

About new filament - that is correct - it may need a retractation command when done but I did not took time to work on this yet 

Thanks a lot for you feedback, it is very valuable

luc-github, 2015-07-22:
I have added issue https://github.com/luc-github/Repetier-Firmware-0.92/issues/96 as I am not at home for a while and far away from printer to work on this

adrianbear, 2015-07-27:
I also installed repetier on my DaVinci 2.0A (jumper J37,  #define DAVINCI 3  and  #define MODEL 1 ).
I did a manual bed calibration and it seems to work fine, but I immediately tried to print one of the sample files installed with the new firmware and it jammed the extruder into the print bed (sideways).

I have done most of my printing with Simplify3d (previously with the xyz firmware).
The things I have noticed so far
 - Extruders are reversed (for my details below extruder 1 is the right extruder, as seen from the front of the device, and as labelled as 1 inside)
 - Extruders are printing with weird offsets 
    -- extruder 2 prints about 35 mm to the left of where it should
    -- extruder 1 cuts about 35 mm off the left hand side of the print (everything that crosses the line is printed down the line)
 - I have to ensure I manually home the printer before starting a print, otherwise it just starts extruding into mid air
 - At start of print...
   -- extruder 2 attempts to dump into extruder 1's bin (missing and dropping on the floor and around the sides of the bed
   -- extruder 1 attempts to dump into extruder 2's bin and forces a large amount of filament into a spot on the bed causing a large clump to form and stick under the extruder
 - The print head just stops at the end of the print and does not leave its last position, melting a hole in the print job (I have worked past this with some gCode snippets in Simplify3d).
- I cam home from work the other night, after NOT printing anything, and found at least one of the extruders was still hot. It obviously had been for over a day without printing anything. This may have been my fault, but is certainly a safety concern.
 - Prints that use the two extruders are impossible as the offsets are wrong and the output does not line up (about 35mm out).

I'll add more issues as I find them

Adrian

g-goessel, 2015-07-27:
I've never noticed anything about offsets, sorry. 
I think the homing before the print and the print head remaining in position after the print are software issues. I use repetier server on my Raspberry Pi and create gcode with curaengine on repetier host and I don't have those issues. 
At the start of a print my extruders don't go to their bins so I don't have this issue ^^
The head remaining hot worries me. I've never had this issue but I always turn it off after printing. If you ever experience it again we will have to hunt a bug. 
I can't help you for the last bug because I've never printed using the 2 extenders in the same time (yet) 
Nice to see someone else owning a 2.0A :-)

luc-github, 2015-07-28:
There are no samples with custome fw - I guess you used XYZ sample and these won't work with custom FW.

About extruder inverted did you check my previous message ? Extruder 1 (E0/T0) is left , Extruder 2 (E1/T1) is right, you need to adjust S3D naming/setting accordingly

About offsets they need to be calibrated as every 2.0 has small differences but 35mm is weird - be noted offsets must not be set in slicer as they are managed directly by FW, the current offsets are the ones from my 2.0.

Homing is done by custom GCODE, same for end print, end sequence need to be added in end GCODE.
all print commands are managed by GCODE provided by slicer - so all expected behavior must be set in slicer commands.

About hot extruder:
there are 3 safety implemented:
1 - if temperature reach 300 printer stop heat and generate a sensor issue
2 - decouple test : if heating behavior is not correct (16 s heating without temperature elevation when target temperature or temperature over target ) printer stop heating with decouple test error
3 - if printer enter in dead loop and then do not control temperature more than 8s, printer restart and so stop heating

if printer is hot but it is targeted temperature, then there is no abnormal behavior from printer point of view as it is expected. I do not know your GCODE but based on your previous comment - I guess there are some missing/wrong settings.

adrianbear, 2015-08-03:
am starting to get used to using Simplify3d with Repetier on my da vacinci 2.0a, however 1 thing that has been irritating but I could work around was the reversal of the extruders (left labelled as right and vice versa).

I have been able to deal with this, but only just today (ok maybe I've been a bit slow) I realised that when printing with the left extruder, the cooling fan on the right is extruder is running instead.

Can you please let me know if this is an issue with the gCode Simplify3d is sending, or is it likely to be an issue with Repetier firmware I am running?

Thanks

luc-github, 2015-08-03:
this  is FW issue, but this has not been changed for ages
can you :
A - heat from printer right extruder / Extruder 1 and check which fan is running
B - heat from printer left extruder / Extruder 2 and check which fan is running

adrianbear, 2015-08-03:
In both cases detailed below, the fan on the “right extruder / extruder 1” is running and not the other.
I was pretty certain I selected DAVINCI 3 MODEL 1. Is there anyway to identify this via the LCD Menu?

luc-github, 2015-08-03:
go to system configuration in settings it display version fw which include the information

adrianbear, 2015-08-03:
Thanks Luc,
It reads

Da Vinci 2.0
1.15-06-04_1.2A
Repetier 0.92Mod

How does any of this relate to the #defines for davinci = 3 (or 2 as you suspect I incorrectly have) and model=1

Thanks

luc-github, 2015-08-03:
yes you should have 3A not 2A if you set DAVINCI 3

adrianbear, 2015-08-05:
Thanks Luc, I fixed my issue above and re-flashed the firmware. This seems to have resolved my offset and fan problems.

However I know have a new issue that may either be related to the firmware or the Simplify3d software I am using for slicing.

New Issue....
What could have caused the print layer to suddenly decide it needs to be 10 mm off in the X axis after 26 hours of printing beautifully (see attached photos)?

I am printing from Simplify3d 3.0.1.
Have you seen this before? Is it caused by the Repetier firmware or is it an issue with simplify3d software?

[image]
[image]

Thanks

luc-github, 2015-08-05:
it is false signal send to end stop that generate this - have several root causes all HW, it affect stock FW and repetier FW on new generation board - it is likely a wire issue or can be a overheat of motor

check these : http://voltivo.com/forum/davinci-peersupport/830-1-0a-and-repetier-host-92-layer-shifting-support-group
http://www.soliforum.com/topic/11773/zone-on-yaxis-causes-massive-print-failure/
http://www.soliforum.com/topic/11742/sporadic-x-axis-shifting/
http://www.soliforum.com/topic/10905/major-shift-in-printing/
http://www.soliforum.com/topic/11030/10-running-repetier-92-y-shifting-issue/
http://www.soliforum.com/topic/10595/x-axis-harness-defects-check-if-you-are-losing-steps/

geoanimator, 2015-08-05:
Luc is right the wiring on the davinci is not good.

I also found another issue that was unrelated to either.
I was using octoprint on a Rasp PI and after several hours of printing I would get x-shifting, y-shifting or filament would jam. No pattern to it all ie different models, filaments, temp settings etc.

It turned out that the Rasp Pi was running out of memory as I was also doing timelapse of prints. So after a few hours it would start dropping gcode commands to printer. oops.

You just need to isolate as much as possible to get to the root cause.

Good luck in your search.
PS
Great firmware Luc. The Davinci is printing better than ever.

luc-github, 2015-08-05:
Thanks ^_^

luc-github, 2015-11-10:
Can anyone confirm pin 6 is ok for TOP Sensor on 2.0A ?
or it should be 64 like for AiO and some 1.0A ?

Thanks

BlaineHa, 2015-12-06:
I have DaVinci AIO and DV 2.0a w/2 fans, and flashed accordingly. Both work well, but I need to set up limits because of larger/longer motors for extruder (I'd put heat sinks on stepper driver chips and side of motor itself, and E3Dv6 mod (raises extruder motor and cap strikes y axis rod when extreme edge of table is called)
I'm not yet sure how to set this, but I need to look closer at config h? Actually, I'm considering removing table and relocating adjustment screws if possible, so I can shift 0 reference so I can regain the printable area as it looks like we have "plenty" of unused travel in y negative. After correcting limits.
However, my immediate issue/request pertains to use of M280 s1 and s0 for 2.0a
Went I attempt to use this code, the system returns "unknown m-code" and strangely though I called extruder 1 (left), the machine prints with extruder 2 (right). Prior to m280 call, extruder 1 printed on left.
Is m280 implemented yet? Can I somehow help it along with a donation, maybe?
Thanks
Blaine

luc-github, 2015-12-06:
What do you call limits ? = distance allowed to move ? everything is configurable in EEPROM so you do not need to recompile FW

M280 is for DITTO printing - as ditto is special mode you need to enable it in configure.h because it is disabled by default.
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1348
Change   
```
/* Ditto printing allows 2 extruders to do the same action. This effectively allows
to print an object two times at the speed of one. Works only with dual extruder setup.
*/
#define FEATURE_DITTO_PRINTING 0
```
to    
```
#define FEATURE_DITTO_PRINTING 1
```
but I never tested it - so I cannot confirm behaviour on Davinci

luc-github, 2015-12-08:
So, is ditto working as expected ?

BlaineHa, 2015-12-08:
Hi Luc

 

Thanks so much for following up.

 

Yes, ditto works great now. I cant thank you enough.

In addition to enabling in EEPROM, I forgot; you have to use what Repetier thinks is number 1 extruder (left), then #2 follows 1

But original DV software called #1 extruder the right one, I think. So I’d set parts on #2, when I should have set on #1 in RH

 

Both my AIO and 2.0 are running J37 MB; steppers don’t have sockets. I of course set up EEPROM accordingly. I also 2.0 w/ old style MB (socketed steppers) machine that’s in pieces I hope to revive and improve, but these things take time as this is “just a hobby” and I have more plans than time. 

 

I do have a few more questions though; I hope this is the right place to ask.

 

1)      I need to change the “zero offset” (I’m a CNC repair guy, “soft limits” and “offsets” are what we the min max travel envelope, and shift of coordinate system by a global amount, respectively) The edge of the table on screen doesn’t correspond to actual location of print…its starting print just off the right side, and wont print all the way over to the left. I’m thinking origin setting, but haven’t experimented with that function yet. How is origin used? Or is solution the “endstop back on home X”

2)      In Repetier Host, the table is depicted 180’ rotated around z. When I hit front view, I actually see what print looks like from the back. How to correct?

3)      My extruder motors/cables were upgraded to 1.5 amp capable, and heatsinks applied to driver chips on MB (new style J37); do the digital gains/current settings function on DV MB? What would be 1.5a approximately? Problem is stalling, rather than stripping filament out. Either with stock drive gear or MK7 hobbed.

4)      After killing print, I always need Z to drop about 100mm for failed part removal and re glue sick. I tried “bed down” but it has no function; did I need to set it up somewhere?

5)      For “clean nozzle” I need steppers to be disabled after z and x y moves, so I can push around head to push and pull filament (you cant really load by hand from Y + max front position) and clear extruder drive gears. Sure, currently I just hit disable steppers, but if it could be part of routine that would be great.

6)      I can’t quite tell if I am up to date on .92 I see there is .92.6 which I think is configuration tool that was used to port to Rep Firm to DV, and .92.3 which I don’t know what it is. I’m sort of wondering if I’m running a older .92 FW, and I don’t know if revisions are posted and I don’t know it because they are still called .92?

7)      I don’t know if its possible, but originally DV FW would basically run autonomously, not good for on the fly speed/flow/temp adjust, but would not fail if PC locked up or updated or whatever. It loaded crazy fast, and just ran. I’d like that option again, if possible. Currently loading to SD card is very, very slow, and I don’t know why. Baud rate is as standard, and connect works consistently.

8)      I’m sure you are aware, but if I have to power down either of my DV machines, the UPS com port wont reconnect. The only way to do so is to cycle power, disconnect in RH, then I must physically pull USB, wait a few moments, then reconnect physically, then finally connect in RH. Just powering on and off won’t allow a reconnect form RH. My USB ports are hard to get to in this install. I could add a extension and hub so I can get to them easier, but I like short USB runs if possible

9)      I could be wrong, but once DV FW 1.1.5 is loaded on AIO MB, you can no longer flash to .92 Only 1.1.1 and 1.1.3 DV FW allows .92 to be flashed. I had to return machine to get a MB I could flash (replacement had 1.1.1) I might be wrong about this, but I’m 99% certain 1.1.5 was preventing my reflash. Just public service info on this point, I don’t know if you can get around it.

But future DV might lock us out entirely if they really do have a FW that prevents flashing to .92

10)   My machines are running e3d v6, and I can change surface mount mosfet if I blow it. Can I raise max temp to 290 for nylon in EEPROM, or is there some other hardware/safety consideration why you limited to 270c? 

 

 

Thanks for reading all this, and I certainly am willing to try experimental things on my machines

I ordered a new DV PRO, but I’m thinking of returning it, as they will have 3 in 1 coming (Like AIO with laser marker and 1,0 and scanner) and I expect we will lose all that functionality as well. Any chance we can hop for at least laser marker functionality with .92 or maybe a .93? If not, I’m sending back the PRO that’s coming in today. Likely will move to Robo R2 when it comes out.

This way I can spend money for toys and trinkets for my kids at a even faster rate…silly hobby that we have, it seems… 

 

Regards

Blaine Handzus

luc-github, 2015-12-08:
1 - you have XYZ min position in EEPROM do you can modify print area position according your wishes
here is the explaination : https://github.com/luc-github/Repetier-Firmware-0.92/issues/80#issuecomment-131387275
but also be sure your FW settings match your repetier host settings (printer shape tab), if not, it can be the reason of bad origin when start printing

 2 - I do not use repetier host but I see there are settings to change X / Y / Z axis orientation in configuration / printer configuration  and tab printer 

3 - there is no digital control for the current as far I know - I have a 2.0 not 2.0A but I doubt they added a component as the purpose is to do a cost down printer, on 1.0/2.0 there is even no manual way to change the current like on ramps drivers.

4 - the kill print function disable stepper - and when stepper is disabled I reset homing flag for safety so bed down (to position 200 mm )won't work as printer does not "know" the distance from home position as homing flag is reset.
you can work around using the repetier script for kill, adding customs commands : it is on menu Config\printer settings and scripts (on 1.6.0) in previous version scripts were accessible from GCODE editors

5 -For “clean nozzle” [...] so I can push around head to push and pull filament : not sure to understand - there is a loading/unloading function and it is not part of clean nozzle - I do not see the relationship

6 - if you use master branch then you use the latest stable version for DAVINCI based on 0.92.3, I am currently implementing the changes from 0.92.6 in experimental branch (0.92.6-sync) but it is not even in pre-alpha, hope end soon

7 - yes SD transfert is terrible, it read line by line, evaluate the code, send to printer which evaluate line by line again before saving to check integrity - stock FW used another way by sending file without line checking but final crc check = faster.
I do not have computer connected to my printers - so for file transfert I use WIFI sdcard (toshiba) it is very fast - for 2.0A/AiO you need cable extender as wifi card is SD not micro SD

8 - Repetier host usb connection is not great - it is easy to get locked - be sure you set port to auto in settings of repetier host - printer does not need a power cycle - unfortunately the solution you gave is the one I also used before 

9 - I do not have new generation but normally when you use the jumper it erase all - so what fw was before should not impact in theory

10 - 270 is for default davinci extruder, like all settings, it is not supposed to handle more - if you change it for one extruder which can support more, it should be ok, just be careful on carriage / belt which are very close to extruder, the residual heat could affect/damage them, so be sure you add eventually some extra thermal protection to avoid this -  but this is you will see by experimenting your temperature control.
Remember Davinci is not a RAMPS so unless your thermistor behave exactly as stock one  - you need to use custom table or rework your board

11 - with 0.92.6, repetier come with some CNC and Laser features : https://github.com/luc-github/Repetier-Firmware-0.92/blob/0.92.6-sync/src/ArduinoDUE/Repetier/Configuration.h#L845-L899
I plan to build a laser machine and also a milling cnc in coming monthes so will be able to test that :wink: 

Hope is clarify your questions

BlaineHa, 2015-12-08:
Thanks Luc

 

So, If I take delivery of DV “PRO” today (its 499us for existing DV owners) and buy laser for $200us in few weeks, I’ll be able to run tests for you on .92.6

I’m a good technition, but I haven’t done any software porting work. But let me know what you need and I’ll make it happen, hopefully

 

The filament loading in ”clean nozzle” and disable steppers:

I mean I physically push tension release lever, and pull filament out while hot. Some times I’ll clip the end and go in and out with it to push out any garbage. You cant quite do this in the forward most position for clean nozzle, but you do need it forward to use “nozzle poking wire tool” and brass brush

Often it fails in print, and having steppers disabled after clean nozzle zyx moves would save a lot of button pushing. 

 

Regards

 

Blaine Handzus

CNC Machine Solutions LLC

Lake Hopatcong NJ 07849

973-809-2370               CNC-MachineSolutions.com

 

here is the explaination : #80 (comment) <https://github.com/luc-github/Repetier-Firmware-0.92/issues/80#issuecomment-131387275> 

I plan to build a laser machine and also a milling cnc in coming monthes so will be able to test that Image removed by sender. :wink:

luc-github, 2015-12-08:
I am pretty sure repetier won t work as it is on DV pro, they may change pin out, mcu, sensors, etc...

you may end with a dead printer if you cannot do the porting of the fw

BlaineHa, 2015-12-08:
Yes, understood

I’m sending it back…too much to do to learn the porting right now

Looking forward to using laser on maybe ramps based machine, right?

Where can I follow your work on this?

luc-github, 2015-12-08:
yes mega/ramps or Due/fd-ramps as I have some spare , I will create a project on github, as I will use also wifi I think

luc-github, 2016-01-07:
@all 2.0A users - is your top sensor working ?
currently top sensor pin is set to pin 6 but this may be wrong and could be 64 as I got only one feedback for 1.0A.

When active, top sensor should bip only when top cover is open and heating

Thanks for the feedback

g-goessel, 2016-01-24:
Hum no it's definitively not working. Never realized I had a top sensor x) (didn't use stock firmware a lot)

luc-github, 2016-01-24:
Thanks so I will update pin from 6 to 64
Thanks

limited660, 2016-01-27:
This is working now.

3dfixer, 2016-03-09:
got a 2.0a second hand 6 weeks ago. unhappy with zyxware taking so long to slice so got s3d.
unhappy with stock firmware so took plunge the weekend and installed repetier.
after resetting with j37 took a while to get computer to see it as a due not a bossa port so then reflashed
all went ok, printer says version 1.15-10-30_1.3a  i have 2 fans.
top cover sensor works, most annoying beep, took a while to work out what was causing it
so i boot up s3d to do a print and it wont connect to printer. try changing somethings but still no.
load up repetier host and everything works fine, connects, manual jog, temps ect all working. try small test print and that goes ok.

question is how do i get s3d to connect??

also found small bug. when asking it to load left extruder the head just move straight across to the left bin then starts dumping (after heating of course). trouble is its caught on the tip cleaner/scrapper things. it should move forward a bit, then far left and then backwards to avoid this. can this be changed in fw?

back to main issue of s3d connecting. all help grateful

izzy

limited660, 2016-03-09:
Try this...

Tools-->Firmware Communication-->Communication and then in the Flow Control setting, set to Hardware Flow Control.

As for the priming issue, I'll look tomorrow after work to see if I can change it.

3dfixer, 2016-03-09:
that worked for the conection, thanks
now got to sort out rest of s3d configs

deuse1, 2016-04-17:
I installed repetier on my DaVinci 2.0A (jumper J37,  #define DAVINCI 3  and  #define MODEL 1 ).

I'm only using one extruder\hot end(Right one as I'm doing a mod). Is there away to turn off\not use the left one to stop the decoupling error on the left side? Thank you for your time.

luc-github, 2016-04-17:
you have 2 options:
Fix the decouple error by checking connector, disable tempororary decouple test in configuration.h then check whas is wrong,may be heating is very slow and need test period to be longer

or you can permanently disable decouple test

but you said you have decouple error on left even you do not use ?

deuse1, 2016-04-17:
Thanks luc. I get the decouple because I'm only using one(the right one)extruder now.
No I don't use the left one at all now. That's why I'm hoping I can turn it off some how in configuration.h.
Everything else is great.

luc-github, 2016-04-17:
you should not have decouple error because you use one extruder

anyway yes there is flag for decouple test in configuration.h to disable it
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L44

luc-github, 2016-04-17:
set to 0 to disable it

deuse1, 2016-04-17:
Thank you Luc. I will re-flash now and try it. And report back.

deuse1, 2016-04-17:
So no more decoupling. But now it's waiting for the left hot end to heat which Is not there any more.
So it won't go any further. What should I edit now Luc?
Thank you

luc-github, 2016-04-17:
ok so you decouple error is nothing related to your machine neither fw
Issue come from your slicer settings, I am pretty sure you ask your left extruder to heat to 0, so he is waiting for a temperature he will never reach, taising decouple test error, clean your GCODE to remove all reference to left extruder

deuse1, 2016-04-17:
It happens all the time. 
I tried to use the inbuilt manual bed leveller and it's waiting for the left hot end to heat up. Which it can't because it's not there. So I can't level the bed.
I try it with simplify3D and the same thing. So it looks like it's the firmware, it thinks the left hot end is still connected, when it's not.

The left hot end\extruder is disconnected from the motherboard and has been taken off the printer. I only want to use the right extruder\hot end. Maybe this can't be worked around?

thanks for helping Luc.

luc-github, 2016-04-17:
it try to heat only if you send command for, check your slicer settings to not use extruder left

luc-github, 2016-04-17:
it is not FW but in your generated GCODE file - you can share using http://pastebin.com/ - I will confirm

deuse1, 2016-04-17:
I think I haven't explained myself clearly Luc.  On the printers lcd I chose\maintenance then Manual Bed and the picture shows what's happening

[image]

Again this is with the built in bed leveller not slicer.

luc-github, 2016-04-17:
ok yes so it is corner case, so your solution to set NUM_EXTRUDER to 1 will solve the issue but may bring some mess elsewhere

luc-github, 2016-04-17:
you may need to adjust x home position  as no more at 0

deuse1, 2016-04-17:
Thank you so much for your help Luc.  I will flash in a bit.

Maurointhesky, 2016-11-19:
Sorry, I have a 2.0a and just after the flash (followed every single step for the config before flashing) the printer is not recognized anymore by my pc, plug and unplugs itself in loop an winzoz tells me tha can't recognize the device. Is there a way to solve it? thanks

Maurointhesky, 2016-11-19:
**Finally after a bloody afternoon I came out of troubles with my 2.0 Duo**
After so many attempts I finally used a working procedure with a 1.6.12 Arduino IDE and a 1.6.8 Arduino DUE Native USB port (installed by the IDE instruments menu) than copied and overwrited the two files, USB and Variants files, in the system user folder /AppData (usually idden).
Finally after some failed attempts changed 3 different USB port... maybe was the main problem!
Hope this can help!!

luc-github, 2016-11-19:
good to know you succeed
May I know what is the difference between the readme/FAQ procedures and your working procedure ?

Maurointhesky, 2016-11-19:
I've done the same thing in the guide.. just with the last IDE version..
Maybe the USB port change was the secret.

luc-github, 2016-11-19:
Ok, thanks a lot for your feedback

Maurointhesky, 2016-11-19:
At the beginnig i had a lot of times black cubes on the lcd.. and some errors from the arduino flashing session..
Maybe the right way comes from right IDE version + right ArduinoDUE usb + that two stupid files in the right place in the AppData folder :) and the right usb plug sabato,

luc-github, 2016-11-19:
yes black bar means variant.cpp is not well replaced

Maurointhesky, 2016-11-19:
But I had them also with well replaced file.. I'm sure.
Actually everything is working...
Right NOW I'm having the first print (went out crazy for calinration) sabato,

bbozzi12, 2017-01-06:
The flashing took a little longer than expected but, other than that the printer works great. The 2.0 prints fucking beautifully with repetier 0.92. Great work luc!!

luc-github, 2018-06-02:
Closing now as outdated

closed 2018-06-02 by luc-github