luc-github/Repetier-Firmware-4-Davinci #97: gcode to turn Lights on

celevra, 2015-07-22:
Hi,

is there a special code i can send to turn the lights an?
With the regular switch in repetier-host nothing happens, if i go thrugh settings directly on the printer i can turn the lights on an off.

regards

Philipp

luc-github, 2015-07-22:
all codes are here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino

M355 S<0/1> - Turn case light on/off, no S = report status, (no eeprom saving) and reset autoturn off time if activated

luc-github, 2015-07-22:
"With the regular switch in repetier-host nothing happens," there is a switch for light in repetier host ?

celevra, 2015-07-22:
[image]

repetier-server i'm sorry, for repetier-host i've written a simple script with the right gcode in it.

like magic it started working from repetier server after doing it manualy through the gcode...

thanks again for your help!

closed 2015-07-22 by celevra

luc-github, 2015-07-22:
good to know - I do not use repetier server - I should give a try