luc-github/Repetier-Firmware-4-Davinci #99: Add menu Item to low down the bed to max position

luc-github, 2015-07-28:
per @uladz suggestion to prepare bed - will be in maintenance menu

adrianbear, 2015-08-03:
Plus 1 this feature request.
I add gcode at the end of a print to force the bed to lowest position, but if I have to kill a print (90% of the jobs started) I have mess around with connections issues or use the position menu and continually hold down to increase Z position. Would be vastly easier just to have a menu option that said "Prepare bed" and it lowered the bed all the way (or for none da vinci printers, moved the print head as far from the bed as possible in the Z axis)

luc-github, 2015-08-04:
implemented https://github.com/luc-github/Repetier-Firmware-0.92/commit/633a810835743f1f6ad481c61e1106439a4db8ef

closed 2015-08-04 by luc-github