luc-github/Repetier-Firmware-4-Davinci #115: DaVinci AIO - Z will not home and will not go up, only down

The-Experimentalist, 2015-10-14:
Hi

I have installed the Repetier 0.92 firmware on my AIO using Arduino 1.5.8 and this repo. When I go to:

Position > Home All

X and Y seem to work fine but Z just goes down and refuses to go up. Any ideas on what the issue is?

I was trying to 'Home All' so I could calibrate but got stuck straight away due to this. Also I cannot get the lights to come on. I tried formatting the SD card and erasing then uploading the firmware again but with the same result.

Any ideas?

The-Experimentalist, 2015-10-14:
Also the lights do not come on regardless of what I set them to. I am new to Repetier but have some experience with Arduino. I would be very grateful of any guidance. Thanks.

The-Experimentalist, 2015-10-14:
Another point. When I go to manual levelling nothing happens, OK or right arrow have no effect. Am I missing something or do I have an incompatible model?

luc-github, 2015-10-15:
Looks like FW is set Davinci 1.0  (DAVINCI 1)instead of AiO (DAVINCI 4) 
can you share what is the FW version is displaying in Settings/System Version in printer menu ?
no movement manual bed leveling means no values in EEPROM did you do load fail-safe or M502/M502 after flashing as requested ?

The-Experimentalist, 2015-10-15:
Hi

After some thinking I think my board must be different to that of zeus10 and kryts.

I have flashed the printer twice. The first time I followed Jake's guide (http://voltivo.com/forum/davinci-peersupport/501-da-vinci-setup-guide-from-installation-to-wireless-printing) where he ends the firmware programming with "On the printer's screen, go to Settings and then choose Store to EEPROM", which I did. The second time, I took out the SD card and formatted it, flashed the printer and then did 'Load Fail-Safe' > 'Store to EEPROM' as explained here. So the EEPROM should have values.

The display shows:

 Da Vinci 1.0
1.15-09-22_1.1
Repetier 0.92Mod

I think the connections are different as the light does not work and I get a beep every 30 seconds or so which I guess is a door sensor.

Which way is the initial travel of the Z axis when homing? My guess would be up looking for the end stop? This makes me think that Z is moving in the opposite direction to the printers of zeus10 and kryts and that the end stop is on a different input.

I am very happy to get involved and load up any test firmware to check this out. I have also traced the light connection back to the board and can provide any pictures that may help to diagnose the issue. I took high res pictures of the board some time ago and will try and submit a picture of where the light is connected for reference and comparison.

By the way I have also seen others on the Voltivo forum reporting the same issues.

Hope to hear back and looking forward to getting involved in investigating what is going on.

Thanks for all your work on this to date.

luc-github, 2015-10-15:
well no,
 FW show you flashed wrong settings you flashed 1.0 not AiO so it is normal nothing work

you must change configuration.h according AiO
#define DAVINCI 4

when it is correct it show Da Vinci AiO in System - that is why I have added the system version - to check when issue like this happen

The-Experimentalist, 2015-10-15:
I did edit the file but I will check again now and report back. Thanks for your help.

luc-github, 2015-10-15:
be sure value is correct in IDE before flashing

The-Experimentalist, 2015-10-15:
You were right, obviously. Not sure why it did not save the first time but must have been my mistake. Just confirmed it in the IDE as you said and re-flashed now showing as AIO and I can get on and calibrate. Thanks for your help and the firmware, great work. It is nice to be released from the grasps of XYZ finally!

luc-github, 2015-10-15:
It is common error - so I have added this system version to track it quickly ;)

Enjoy your brand new printer :smile:

closed 2015-10-15 by luc-github